iNavFlight/inav #8278: nav_wp_enforce_altitude margin too tight for fixed wings

b14ckyy, 2022-07-31:
## Current Behavior
the target margin for the enforce altitude option on waypoints is too tight. Causing unnecessary loiter turns even on waypoints that have the same altitude. 

## Steps to Reproduce
1. plan mission
2. fly mission in slight winds or turbulences
3. watch altitude and turn behavior at waypoints
4. Plane will switch to "adjust altitude" state on nearly every waypoint

## Expected behavior
forced altitude loiter should only be done during a relatively big deviation. especially on wings

## Suggested solution(s)
Option 1: Disable the altitude enforcement if the last waypoint and current waypoint have a altitude difference below a certain threshold (maybe 10m?)
Option 2: Raise the margin for fixed wings to 10m or make it adjustable
Option 3: calculate the need of altitude adjustment based on the expected maximum climb angle of the airplane (nav climb rate, cruise speed, and distance.

For multirotor no changes might be required. 

## Additional context
Can be seen in real life flight easily but also in the HITL simulation: https://youtu.be/4pVUUmOFEH8 
@breadoven this should be considered to be fixed for 5.1 as the feature is currently unusable. 
---

- FC Board name and vendor: any
- INAV version string: 5.0 Release

breadoven, 2022-07-31:
I thought you had altitude control honed down to cm accuracy ?

However, I think the easiest solution is to make the tolerance a user setting with 0 = disabled otherwise adjustable in meters (cm ?) up to 20 meters perhaps ?

Out of interest how much altitude variation do you get between WP's with the same altitude setting (from flight testing rather than HIL) ?

b14ckyy, 2022-07-31:
In real life if it is sunny and windy, sometimes I can get 2-3m variation from the target altitude. In worst case. That's why I just noticed that recently when I used the feature with my Talon 250G. Very light 247g setup that gets blown around by wind. but yeah a configurable setting would be ideal. For quads the FW default can stay 100cm and for wings I will update the presets with a 500cm tolerance maybe.

closed 2022-08-15 by DzikuVx