iNavFlight/inav #8295: RUSH_BLADE_F7 BMP280 Barometer doesn't work

NikitaKA, 2022-08-06:
## Current Behavior
RUSHFPV RUSH BLADE F722 FLIGHT CONTROLLER FOR DJI DIGITAL FPV SYSTEM supports BMP280 (and it works fine in Betaflight and Emuflight), but not in iNav.

## Steps to Reproduce
[image]

## Suggested solution(s)
Add/fix support for BMP280 in RUSH_BLADE_F7 target.

## Version
INAV/RUSH_BLADE_F7 5.1.0 Aug  4 2022 / 12:57:00 (2590629e)
GCC-10.2.1 20201103 (release)

DavyJohnes, 2022-09-27:
@NikitaKA any updates on this? Did you make it work?

NikitaKA, 2022-09-27:
> @NikitaKA any updates on this? Did you make it work?

Unfortunately not. Swapped to Matek F405SE...

NoviKurniawan, 2023-06-30:
no fix for this bug yet? I tried the latest update (INAV 6.1.1) but still got the same error

LYCH90, 2023-08-24:
This situation is still happended with INAV6.1.1. Please fix it sirs

Firesail99, 2023-08-27:
Flywoo 745 AIO hat auch den Barometer BMP280 verbaut. Dieser wird auch hier INAV5.0 - 6.1.1 nicht  erkannt bzw. lässt sich nicht aktivieren. Gibt es hier eine Lösung? (In Betaflight wird der Barometer sofort erkannt)

LYCH90, 2023-09-08:
Developer INAV  may be dont care?

Firesail99, 2023-09-23:
Ich habe  von Flywoo für den 745 AIO eine eigene spezielle INAV 6.x bekommen. Der Baro funktioniert jetzt. Flywoo sagt sie wären mit INAV Entwicklung in Kontakt.......

UmbrelaCorp, 2023-10-30:
Blade V1.1 f7 digital has hardware
QMP6988 as barometer according to betaflight, it works OK in bf4.4 but been trying for couple days now to get it working on 6.1.1 inav

b14ckyy, 2024-03-21:
Is this still an issue with 7.1?

closed 2024-03-22 by b14ckyy