iNavFlight/inav #8301: Change missions in flight

krl91, 2022-08-07:
## Current Behavior
Actually since iNav 4.10 it is possible to have more than one mission recorded in the FC. 
To change from mission 1 to mission 2 the only way to stop flying, un-arm and use the OSD menu to select the second mission.

## Desired Behavior
* Change missions in flight

## Suggested Solution
* Have the possibility to change in flight the mission number
* Have the possibility to change mission from a button from the TX

## Who does this impact? Who is this for?
All users who wants to have planned missions with multiple possibilities depending on the real situation.
For example, for a fixed wing, with this feature it will possible to manage during the flight the 1. mission itself and 2. prepare for landing 
If there is any mission cancel the user should activate the Mission to prepare for landing, the goal is to be in the good direction, good altitude for prepare for a manual landing. Also it is interesting to prepare several prepare for landing mission, depending to the wind orientation.

## Additional context
in my home there is a long rectangular area that take time fir the fixed wing to be in the correct orientation (like a large road), only 2 possibilities depending of the wind:
[image]

breadoven, 2022-08-15:
I've thought about this but knowing how it's currently implemented it's not so easy to do without completely changing the way it works at the moment. Maybe something to consider for the future though.

breadoven, 2022-08-19:
I managed to get this working in a rough version. Uses a "`missionchange`" mode which when active allows the mission index to be changed using the WP mode switch (cycles though the mission numbers). So you could assign `missionchange `mode to use its own switch or just set it active together with another mode such as Poshold.

closed 2022-09-20 by breadoven

alcmann, 2023-08-03:
Greetings.  Running inav 6.0.0

I can create multiple missions, and load into FC.  On boot I see mission (1/2) loaded of the two missions but still unable to change missions in flight when disarmed.  

I still as indicated above by on ground disarm using OSD menu.  I have tried Breadoven solution of binding both mission change and wpplanner to buttons.  Using different variations of holding pressing and selecting both buttons I am unable to change missions. 

Any clear outline or instructions to follow doing this with anyone who has successfully gotten this to work ? 

If not requesting a feature of an easier interface while in flight to allow mission change to just cycle through missions available with every depress or selection of the switch. 

Thank you for all your help