iNavFlight/inav #8303: Not detecting Gyro, Accel, Baro on Matek F405 based board

chitoku, 2022-08-08:
This is similar to Issue #5835, but is happening on Matek F405 based board.
I built [Tyro 119](https://www.eachine.com/Eachine-Tyro119-250mm-F4-OSD-6-Inch-3-6S-DIY-FPV-Racing-Drone-PNP-w-or-Caddx-Turbo-F2-1200TVL-Camera-p-1700.html), which comes with F405 based FC with MPU6000 and BMP280.

## Current Behavior

- **INAV 5.0**: Accel, Gyro, and Baro **NOT** dtected, all lit RED. Manual selection does not help.
- **BetaFlight 4.3**: Aceel and Gyro detected, lit BLUE. Baro is not detected.

[image]

## Steps to Reproduce
1. Flash `5.0.0 - MatekF405 - 2022-6-25 17:24 (stable)`
2. Connect to FC, select **Mini Quad with 3" - 7" propellers**
3. In the initial "**Setup**" tab window, check if the virtual drone model moves as winggling the FC
4. Check the Gyro, Accel and Baro buttons on the top bar

## Expected behavior
The virtual drone model moves, and Gyro, Accel and Baro buttons lit in blue, like shown in [Joshua Bardwell's tutorial video](https://youtu.be/WYPzt-HcGH8?list=PLwoDb7WF6c8nj7G3o9NJpzVPPTsw2z-Jy&t=122).
[image]

## Suggested solution(s)
I could not fix this and need help.

## Additional context
When flashing BetaFlight 4.3.1 (`MATEKF405STD` selected), and custom defaults applied, Gyro and Accel buttons lit up and the virtual drone model moves as FC wiggled.
[image]

In **Configuration** tab, **Barometer** is actually turned on.
[image]

On BetaFlight CLI

**I2C**
```
# get i2c
mag_i2c_device = 1
Allowed range: 0 - 3

mag_i2c_address = 0
Allowed range: 0 - 119

baro_i2c_device = 1
Allowed range: 0 - 5

baro_i2c_address = 0
Allowed range: 0 - 119

dashboard_i2c_bus = 1
Allowed range: 0 - 3

dashboard_i2c_addr = 60
Allowed range: 8 - 119

gyro_1_i2cBus = 0
Allowed range: 0 - 3

gyro_1_i2c_address = 0
Allowed range: 0 - 119

gyro_2_i2cBus = 0
Allowed range: 0 - 3

gyro_2_i2c_address = 0
Allowed range: 0 - 119

i2c1_pullup = OFF
Allowed values: OFF, ON

i2c1_clockspeed_khz = 800
Allowed range: 100 - 1300

i2c2_pullup = OFF
Allowed values: OFF, ON

i2c2_clockspeed_khz = 800
Allowed range: 100 - 1300

i2c3_pullup = OFF
Allowed values: OFF, ON

i2c3_clockspeed_khz = 800
Allowed range: 100 - 1300
```

**SPI**
```
# get spi
yaw_spin_recovery = AUTO
Allowed values: OFF, ON, AUTO

yaw_spin_threshold = 1950
Allowed range: 500 - 1950

mag_spi_device = 0
Allowed range: 0 - 3

baro_spi_device = 0
Allowed range: 0 - 5

rx_spi_protocol = V202_250K
Allowed values: V202_250K, V202_1M, SYMA_X, SYMA_X5C, CX10, CX10A, H8_3D, INAV, FRSKY_D, FRSKY_X, FLYSKY, FLYSKY_2A, KN, SFHSS, SPEKTRUM, FRSKY_X_LBT, REDPINE, FRSKY_X_V2, FRSKY_X_LBT_V2, EXPRESSLRS

rx_spi_bus = 0
Allowed range: 0 - 3

rx_spi_led_inversion = OFF
Allowed values: OFF, ON

sdcard_spi_bus = 3
Allowed range: 0 - 3

vtx_spi_bus = 0
Allowed range: 0 - 3

max7456_spi_bus = 2
Allowed range: 0 - 3

frsky_spi_autobind = OFF
Allowed values: OFF, ON

frsky_spi_tx_id = 0,0,0
Array length: 3

frsky_spi_offset = 0
Allowed range: -127 - 127

frsky_spi_bind_hop_data = 0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
Array length: 50

frsky_spi_a1_source = VBAT
Allowed values: VBAT, EXTADC, CONST

cc2500_spi_chip_detect = ON
Allowed values: OFF, ON

flash_spi_bus = 0
Allowed range: 0 - 3

gyro_1_spibus = 1
Allowed range: 0 - 3

gyro_2_spibus = 0
Allowed range: 0 - 3

spektrum_spi_protocol = 0
Allowed range: 0 - 255

spektrum_spi_mfg_id = 0,0,0,0
Array length: 4

spektrum_spi_num_channels = 0
Allowed range: 0 - 12

flysky_spi_tx_id = 0
Allowed range: 0 - 4294967295

flysky_spi_rf_channels = 0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0
Array length: 16

```

---

- FC Board name and vendor: [Eachine GPS F4](https://www.eachine.com/Eachine-Tyro119-250mm-F4-OSD-6-Inch-3-6S-DIY-FPV-Racing-Drone-PNP-w-or-Caddx-Turbo-F2-1200TVL-Camera-p-1700.html) by Eachine 
- INAV version string: 
```
# version
# INAV/MATEKF405 5.0.0 Jun 20 2022 / 10:56:14 (d2adf35c)
# GCC-10.2.1 20201103 (release)
```

erstec, 2022-08-08:
@chitoku you using wrong target.
"Matek F405 based" FC is not Matek F405 FC.
You need to ask manufacturer for support, not INAV, as INAV developers don't have this FC and have no clue what it is.

erstec, 2022-08-08:
Additionally: I just looked at BF target (MATEKF405STD) and compared with INAV (MATEKF405), from the first look they are more or less the same, so, at least Gyro/Acc should be detected, as long as OSD, SD Card/Flash and UARTs should work.
Do you selected Full chip erase during flashing your FC with INAV?

chitoku, 2022-08-08:
Hi @erstec , thank you so much for sharing your insights.
Yes, I selected **Full chip erase**.
[image]

It's great news/confirmation for me that Betaflight target definition is more or less the same. 
I will try to see what's happening to the SPI and I2C communication on INAV. 
After all, Joshua Bardwell got to fly this Tyro 119 with INAV 2.4, so I'm hoping there is a way.

chitoku, 2022-08-08:
BTW, when I try to flash the old INAV 2.4.0 using INAV configurator 2.4.1, I cannot connect to the FC after flashing done.
```
2022-08-08 @ 11:42:25 -- No configuration received within 10 seconds, communication failed
2022-08-08 @ 11:42:42 -- Unrecoverable failure of serial connection, disconnecting...
2022-08-08 @ 11:42:42 -- Failed to close serial port
```

chitoku, 2022-08-14:
It seems that Eachine started shipping their FC now with a different IMU. Mine had BM270 as hinted in their update note.
https://www.banggood.com/Eachine-Tyro119-250mm-F4-OSD-6-Inch-3-6S-DIY-FPV-Racing-Drone-PNP-w-or-Caddx-Turbo-F2-1200TVL-Camera-p-1584996.htm

In my local copy, I defined a new target and I made some changes to see if resulted firmware would work on the Eachine FC.
The result was that I got to make Gyro and Accel work along with Magnetometer and the GPS, so push that in my fork.
https://github.com/chitoku/inav/tree/eachine_f405

The BMP280 barometer on my board seems to be defective as it did not work on BetaFlighter either.

DzikuVx, 2022-08-15:
This is the reason it does not work: https://youtu.be/pjTeXzOtEUE

closed 2022-08-15 by erstec