iNavFlight/inav #8306: Angle mode has a fatal bug,It may cause the plane to crash

hjchjcjh, 2022-08-10:
Hello everyone: 
  On this flying wing, I have been running inav4.1 stably for several months, but this time it crashed.
   Because the angle mode has a fatal bug: sometimes the rc command pitch doesn't work, so my wing crashes.
   you can see fly time at 39min16s(dvr and blacklog), at first I thought it was not enough airspeed, so I increased throttle, then RC command: pitch down, but only throttle took effect, pitch did not take effect, so the wings still dived down 30 degrees, until crash
   For details, see dvr or blackbox log

https://user-images.githubusercontent.com/25506063/183927508-c8a41d96-4592-4783-a070-1a0a51dacacf.mp4
[image]

[LOG00007.zip](https://github.com/iNavFlight/inav/files/9300899/LOG00007.zip)
[inav_diff.txt](https://github.com/iNavFlight/inav/files/9300906/inav_diff.txt)
.

stronnag, 2022-08-10:
1. The log is somewhat corrupt, you might consider lowering the logging rate
2. At the time of the FAILSAFE, the log indicates the aircraft was in Manual .... not angle.
```
$ inav_states.rb  LOG00007.TXT 
INAV 4.1.0, states from 2.7.0
LOG00007.TXT: INAV 4.1.0 (611d26f4) OMNIBUSF4PRO
Iteration	Time(s)	Elapsed(s)	State	FltMode	navFlag
...
  2255552	2391.8	(2294.9)	nav_state_idle (1)	ARM|MANUAL	
  2326780	2464.3	(2367.4)	FAILSAFE=>RETURN_TO_HOME (1980,988,1519,1502);(0,1)

```

stronnag, 2022-08-10:
According to the log you're in MANUAL  .....

JulioCesarMatias, 2022-08-11:
Actually, the problem was caused by your radio... It wasn't the iNav system that caused it.

Here the channel is still OK!
[image]

Now the channel value starts to decay
[image]

And here the command hangs at the minimum value.
[image]

I think your radio Fail-Safe is set incorrectly ((wrong channel))

hjchjcjh, 2022-08-11:
> Actually, the problem was caused by your radio... It wasn't the iNav system that caused it.
> I don't agree with it
> Here the channel is still OK! [image]
> that's right, at this time I found that the altitude was relatively low and I was ready to react,.
> Now the channel value starts to decay [image]
> Because at this time, I controlled the radio pitch and tried to pull up the plane, but the inav did not make any response to climb, instead it continued to dive and crash.
> And here the command hangs at the minimum value. [image]
> 
> I think your radio Fail-Safe is set incorrectly ((wrong channel))
    I don't think my radio Fail-Safe is set incorrectly, because the first few times Fail-Safe is ok,
    And between 39 minutes and 16 seconds to 26 seconds (inav didn't respond to my radio pitch, instead it was diving), my plane was always in Angle mode, and rssi still ok. So it's impossible to trigger Fail-Safe

[image]

JulioCesarMatias, 2022-08-11:
entenod... So, the AHRS Drift probably occurred to you

MrD-RC, 2022-08-12:
Quite possibly drift, with the amount of down seen by the FC as higher than the amount of up available. 

[image]

@hjchjcjh why did you not switch to acro when you found you couldn't pitch up enough?

Try the mitigation parameters found in this PR https://github.com/iNavFlight/inav-configurator/pull/1592

hjchjcjh, 2022-08-12:
> Quite possibly drift, with the amount of down seen by the FC as higher than the amount of up available.
> 
> [image]
> 
> @hjchjcjh why did you not switch to acro when you found you couldn't pitch up enough?
> 
> Try the mitigation parameters found in this PR [iNavFlight/inav-configurator#1592](https://github.com/iNavFlight/inav-configurator/pull/1592)

    Great analysis!!!when I looked at the dvr video, it was definitely drift, because the OSD horizon was not correct during the pitch out of control.
[DVR.mp4](https://user-images.githubusercontent.com/25506063/183927508-c8a41d96-4592-4783-a070-1a0a51dacacf.mp4)
    As for why I don't switch to arco mode, because the height is too low, I don't have much time to react, and I used to rely too much on angle mode haha.
    So, as a regular pilot, how can I avoid drift in future flights?

MrD-RC, 2022-08-12:
By checking out the link I posted in my last comment. That will help. They will be the defaults in 5.1.

closed 2022-08-12 by MrD-RC