iNavFlight/inav #8311: OSD incorrect labels and data MATEK F405  INAV 5.0 OSD AT7456E

Bookwormjim, 2022-08-12:
The FPV monitor does not display the OSD horizon, the direction home, the distance to home, the number of satellites, battery voltage and many, many other OSD parameters.
The labels are either missing in the required fields, or other data is available instead. For example, instead of the MCU temperature, the symbol "battery temperature" is displayed.

The configurator displays correctly, the settings are recorded, there are no error messages.

Everything works correctly in Betaflight 10.8.0. I don't tried INAV of another version.

sorry for the automatic translation

FC MATEK F405, drone Darvin 129

# version
# INAV/MATEKF405 5.0.0 Jun 20 2022 / 10:56:14 (d2adf35c)
# GCC-10.2.1 20201103 (release)

MrD-RC, 2022-08-12:
Have you uploaded the fonts?

I'm almost betting no and closing the issue now 🤣 You need to upload the fonts when you change firmware (BetaFlight or inav for example) and with inav, when updating to a new major version.

Bookwormjim, 2022-08-12:
> Have you uploaded the fonts?
> 
> I'm almost betting no and closing the issue now 🤣 You need to upload the fonts when you change firmware (BetaFlight or inav for example) and with inav, when updating to a new major version.

I don't understand your laughter and your frivolous attitude to work.
What update, what are you talking about? I downloaded a new INAV from your github and a new 5.0 firmware, but did not update the old version.
And what does fonts have to do with it, if GPS coordinates show normally?
p.s. I tried changing fonts, it didn't help.
p.p.s In general, this version is a curve at many points.

MrD-RC, 2022-08-12:
Firstly, this isn't work. I don't get paid for it. Secondly, 99.9% of the time, when there is something not displaying correctly on the OSD, it is because the font has not been updated. This can make a lot of things on the OSD look incorrect. Especially if updating from pre-3.0 inav or a different firmware entirely. 

Maybe post an image of your OSD.

You download inav 5.0, but did not update? So when did the OSD start looking incorrect? Something must have changed.

Bookwormjim, 2022-08-12:
> Firstly, this isn't work. I don't get paid for it. Secondly, 99.9% of the time, when there is something not displaying correctly on the OSD, it is because the font has not been updated. This can make a lot of things on the OSD look incorrect. Especially if updating from pre-3.0 inav or a different firmware entirely.
> 
> Maybe post an image of your OSD.
> 
> You download inav 5.0, but did not update? So when did the OSD start looking incorrect? Something must have changed. Or are you using configurator 5.0 with an old version of inav on your flight controller. You should always match major and minor versions of configurator and firmware.
[image]
[image]

OK, thanks 
I buy new drone with pre-installed Betaflight firmware. 
I don't have before installed betaflight or INAV softvare. 
Betaflight firmware work correctly, i use OSD and changed OSD without problem.
Then i download INAV last version 5.0.0, upload to FC NAVI firmware latest version with full erased chip. And OSD work not correctly full time. 
i pick up pictures with configurator and fpv .

MrD-RC, 2022-08-12:
That is the wrong font. You need to go to the Font Manager at the bottom of the OSD page and upload the correct font to the flight controller.

Bookwormjim, 2022-08-14:
> That is the wrong font. You need to go to the Font Manager at the bottom of the OSD page and upload the correct font to the flight controller.

Thanks, it worked!
But it started working on the third attempt when I loaded the "Impact" font. When I loaded the font "Bold", the result did not give.
Thanks again, good luck.

MrD-RC, 2022-08-14:
All the fonts are the same as far as characters go. Perhaps the first 2 uploads failed?

closed 2022-08-14 by MrD-RC

Bookwormjim, 2022-08-14:
No, I haven't seen any error messages. The flight controller accepted the download and rebooted successfully.