iNavFlight/inav #8336: MSP_SET_RAW_GPS is not setting correct values

miguelwon, 2022-08-18:
Hi,
I'm trying to set GPS data using MSP. The problem is that `fix` and `altitude` are not correctly injected. 

Following MSP [protocol](http://www.multiwii.com/wiki/index.php?title=Multiwii_Serial_Protocol), I'm injecting the values: 

```
fix 1
num_sats 5
lat 405663433
lon 81738581
altitude 776
spd_over_grnd 51
```

that when converting according to each data type results in the data array:  

`[1, 5, 201, 238, 45, 24, 85, 59, 223, 4, 8, 3, 51, 0]`

where I'm using the following function to convert the values:

```
def convert(val_list, n=16):
    """Convert to n*bits (8 multiple) list

    Parameters
    ----------
    val_list : list
        List with values to be converted

    n: int, optional
        Number of bits (multiple of 8) (default is 16)

    Returns
    -------
    list
        List where each item is the equivalent byte value
    """
    buffer = []
    for val in val_list:
        for i in range(int(n/8)):
            buffer.append((int(val)>>i*8) & 255)
    return buffer

data_gps = convert([fix], 8)
data_gps += convert([num_sats], 8)
data_gps += convert([lat], 32)
data_gps += convert([lon], 32)
data_gps += convert([altitude], 16)
data_gps += convert([spd_over_grnd], 16)
```

After injected I read the GPS values with `MSP_RAW_GPS` that shows: 

`{'fix': 0, 'numSat': 5, 'lat': 405663433, 'lon': 81738581, 'alt': 7, 'speed': 51, 'ground_course': 0, 'distanceToHome': 0, 'ditectionToHome': 0, 'update': 0, 'chn': [], 'svid': [], 'quality': [], 'cno': [], 'hdop': 0}`

This is confirmed by the reading I see in INAV configurator:

[image]

The fix and altitude values are incorrect. Fix should be 1 and altitude  776 meters. I initial thought that perhaps the problem in the altitude is related to units. But oddly, when tying to inject the altitude in cms (77600) it results in a very wrong value:

```
fix 1
num_sats 4
lat 405663615
lon 81739192
altitude 77500
spd_over_grnd 51
[1, 4, 127, 239, 45, 24, 184, 61, 223, 4, 188, 46, 51, 0]
{'fix': 0, 'numSat': 4, 'lat': 405663615, 'lon': 81739192, 'alt': 119, 'speed': 51, 'ground_course': 0, 'distanceToHome': 0, 'ditectionToHome': 0, 'update': 0, 'chn': [], 'svid': [], 'quality': [], 'cno': [], 'hdop': 0}
```

---

# version
# INAV/UBIONE 5.0.0 Aug 15 2022 / 09:25:39 (3a379a7c)
# GCC-10.2.1 20201103 (release)

stronnag, 2022-08-18:
There is a bug in the SET_RAW_GPS altitude conversion, it is not converted to cm (whereas cm to converted to m outbound).
See #8337

stronnag, 2022-08-18:
It was not setting the fix either. Fixed in #8337

JulioCesarMatias, 2022-08-19:
Thanks for reporting this! ʸᵉˢ ᴾᵒʳᵗᵘᵍᵃˡ ˢᵗᵒˡᵉ ᵃˡˡ ᵗʰᵉ ᵍᵒˡᵈ ᶠʳᵒᵐ ᴮʳᵃᶻᶦˡ

stronnag, 2022-08-19:
Closing, as a fix has been merged.

closed 2022-08-19 by stronnag