iNavFlight/inav #8340: IRC Tramp RF Power Selection Enhancement

jc9aj, 2022-08-20:
## Current Behavior
Current implementation of the Tramp Power Protocol is too rigid in INAV for modern VTX units. There are currently four power tables that are implemented for 5G8:

25, 100, 200, 200, 200
25, 100, 200, 400, 400
25, 100, 200, 400, 600
25, 100, 200, 500, 800

Which table is used for programming the VTX is determined by the maximum power output reported by the VTX. The power table selection can be overridden by the `vtx_max_power_override `CLI function (note that setting this >800mW will still select the 800mW table). Unlike Smart Audio, the Tramp protocol sends the desired power output value in mW to the VTX. Therefore, there are two main issues, plus two other minor issues with the current implementation:

1. INAV is not able to instruct any Tramp VTX to a higher output than 800mW 
2. There are a vast number of Tramp VTX units on sale that have at least one RF power output less than 800mW that is not able to be selected by INAV (See below for a list of a number of VTX units and how they might be affected).
3. The number of power levels supported by each VTX can vary between 4 and 6. By having constant 5 power settings, for some VTX INAV is limiting the selection of the 6th, while VTX units with 3 power outputs cannot act on the 'extra' setting

## Desired Behavior
Implement a more flexible way of modifying the Tramp Power tables in INAV, either via CLI command or via the configurator GUI. 

## Suggested Solution
Ideally this should include a setting for number of Tramp power levels, the Power Values to be sent to the VTX and the Power Labels to be displayed in the OSD. Betaflight and Cleanflight use the following commands to set the Tramp power levels:

```
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 100 200 400 600
```

_Note, it is worth considering a similar approach could be adopted for Smart Audio, that would permit the correct RF power labels to be displayed in the OSD for both interfaces_

## Who does this impact? Who is this for?
Everyone who operates a non Immersion RC VTX with Tramp interface, especially those that wish to have higher powered VTX units

## Additional context
List (not exhaustive) of VTX units that have/are expected to exhibit less than optimal performance with INAV based on the current design. I've documented the output power of each and highlighted which power levels are likely to be an issue (my assessment that vtx_max_power_override is not used and that each VTX is reporting max power correctly).

**Speedybee tx800** - 25mW, 200mW, 400mW, 800mW  (400mW not selectable via Tramp in INAV)
**iFlight BLITZ** - 25mW, 400mW, 800mW, 1600mW  (400mW & 1600mW not selectable via Tramp in INAV)
**iFlight SucceX Force 800mW** - 25mW, 100mW, 400mW, 800mW  (400mW not selectable via Tramp in INAV)
**iFlight SucceX Mini** - 25mW, 100mW, 200mW, 400mW, 500mW  (500mW not selectable via Tramp in INAV)
**iFlight SucceX Micro Force** - 25mW, 100mW, 200mW, 300mW (300mW not selectable via Tramp in INAV)
**HGLRC Zeus 800mW** - 25mW, 100mW, 200mW, 400mW, 800mW  (400mW not selectable via Tramp in INAV)
**HGLRC Zeus Nano** - 25mW, 100mW, 200mW, 350mW (350mW not selectable via Tramp in INAV)
**Foxeer Reaper Nano** - 25mW, 100mW, 200mW, 350mW (350mW not selectable via Tramp in INAV)
**Foxeer Reaper Extreme** - 25mW, 200mW, 500mW, 1500mW, 2500mW (1500mW and 2500mW not selectable via Tramp in INAV)
**Diatone MAMBA Ultra 1000** - 25mW, 200mW, 400mW, 800mW, 1000mW (400mW and 1000mW not selectable via Tramp in INAV)
**Diatone MAMBA TX500** - 25mW, 200mW, 500mW (500mW not selectable via Tramp in INAV)
**GEPRC RAD MINI** - 25mW, 200mW, 500mW, 1000mW (1000mW not selectable via Tramp in INAV)
**GEPRC RAD**  - 25mW, 200mW, 600mW, 1600mW (1600mW not selectable via Tramp in INAV)
**JHEMCU RuiBet** - 25mW, 200mW, 400mW, 800mW, 1600mW (400mW and 1600mW not selectable via Tramp in INAV)
**JHEMCU FE300T** - 25mW, 50mW, 100mW, 200mW, 300mW (50mW and 300mW not selectable via Tramp in INAV)
**JHEMCU VTX30-800** - 25mW, 100mW, 200mW, 400mW, 800mW (500mW not selectable via Tramp in INAV)
**ATOMRC Exceed** - 25mW, 200mW, 500mW (500mW not selectable via Tramp in INAV)
**Eachine Nano Plus** - 25mW, 200mW, 600mW, 800mW (600mW not selectable via Tramp in INAV)
**Flywoo GOKU TX-NANO** - 25mW, 50mW, 100mW, 200mW, 450mW (50mW and 450mW not selectable via Tramp in INAV)
**Flywoo GOKU HM600** - 25mW, 50mW, 100mW, 200mW, 600mW (50mW not selectable via Tramp in INAV)
**Flywoo GOKU HM850** - 25mW, 50mW, 100mW, 200mW, 850mW (50mW & 850mW not selectable via Tramp in INAV)
**PandaRC VT5804 Mini** - 25mW, 50mW, 100mW, 200mW, 400mW (50mW not selectable via Tramp in INAV)
**PandaRC VT5804 Nano** - 25mW, 50mW, 100mW, 200mW, 400mW (50mW not selectable via Tramp in INAV)
**PandaRC VT5804 X1/Q1** - 25mW, 100mW, 200mW, 400mW, 800mW (400mW not selectable via Tramp in INAV)
**PandaRC VT5804 V3** - 25mW, 100mW, 200mW, 400mW, 800mW, 1000mW (400mW and 1000mW not selectable via Tramp in INAV)
**PandaRC VT5804 BAT** - 25mW, 400mW, 800mW, 1500mW, 2500mW (400mW, 1500mW and 2500mW not selectable via Tramp in INAV)
**Matek 5.8G VTX** - 25mW, 200mW, 500mW (500mW not selectable via Tramp in INAV)

Ultraexpert, 2022-08-24:
pretty sure my speedybeevtx has all 4 power levels

tzheng777, 2022-09-25:
I believe I'm seeing this issue on a Speedybee VTX-DVR. 

According to the specs this VTX only has 3 possible power outputs: 25, 200 and 600mW. 

From the configurator and using the automatically chosen VTX table, I am only able to trigger the 25 and 200mW options. I was unable to trigger the 600mW settings, even after trying all allowable values from 1 to 5. I'm guessing this is due to the VTX reporting a different max output to the FC, and thereby choosing the wrong VTX table?

To verify what power setting is actually being used, I check the LED pattern on the VTX board itself. I can also manually enable the 600mW option by using the board mounted button, however this does not seem to update the `vtx_power` setting on the configurator. 

Coming from Betaflight I do find it peculiar that INAV does not allow the user to upload or customise VTX tables. 

In the meantime I will try setting `vtx_max_power_override = 600` to see if this allows me access to the 600mW option.

Dteksolutions, 2023-02-10:
Same issue on an iflight blitz. Only the first 2 work.

Stemplar42, 2023-03-06:
Mismo problema en un GEPRC RAD 1,6 Watts

fermiums22, 2023-03-11:
Up 6.0.0 RC3. Fix please.

MrD-RC, 2023-03-11:
> Up 6.0.0 RC3. Fix please.

6.0 had a featured freeze in January. This will not be in 6.0 if it is developed.

fermiums22, 2023-03-11:
> > Up 6.0.0 RC3. Fix please.
> 
> 6.0 had a featured freeze in January. This will not be in 6.0 if it is developed.
I corrected the table in the tramp.c and build the project. everything works. I hope it will work when it is relil and I won't have to build inav myself.

MrD-RC, 2023-03-11:
The tables will be as they are now. You will need to build a custom version when 6.0 is released. 

Ideally Tramp needs to use power tables now, as there are too many options added by vendors. 

It shouldn't be needed for SA, as that can be queried for the power levels.

Stemplar42, 2023-03-12:
> The tables will be as they are now. You will need to build a custom version when 6.0 is released.
> 
> Ideally Tramp needs to use power tables now, as there are too many options added by vendors.
> 
> It shouldn't be needed for SA, as that can be queried for the power levels.

OK. Thanks for the reply. Can you provide me with a procedure to compile INAV in a custom way and change the power tables for IRC TRAMP? I have a 1.6W GEPRC VTX in which I cannot select that power. Thank you very much Darren.

MrD-RC, 2023-03-12:
@Stemplar42 No. It is not a simple procedure. There are documents in the docs/developers section for setting up development environments and building targets. You would also need to understand coding in C. 

Unfortunately, most people are going to have to wait until a developer decides to implement this change.

Stemplar42, 2023-03-12:
> @Stemplar42 No. It is not a simple procedure. There are documents in the docs/developers section for setting up development environments and building targets. You would also need to understand [...]

Okay.  I understand perfectly.  Although I am a C language programmer, I understand that it is a delicate change that would have to be made.  Thank you so much

sketch4x4, 2023-09-15:
quite a simple change, I've managed to do it on my hglrcf722 board using the zeus 1600mw vtx. you need to add two sections of code to src\main\io\vtx_tramp.c towards the bottom (first bit of code is added at line 575 on my file) ive attached a screen grab for location.

`const uint16_t trampPowerTable_5G8_1600[VTX_TRAMP_5G8_MAX_POWER_COUNT]         = { 25, 400, 800, 1600, 1600};
const char * const trampPowerNames_5G8_1600[VTX_TRAMP_5G8_MAX_POWER_COUNT +1 ] = { "---", "25 ", "400", "800", "1600" };`

and

`if (maxPower >= 1600) {
                // Max power 1600mW: Use 25, 400, 800, 1600 table
                vtxState.metadata.powerTablePtr  = trampPowerTable_5G8_1600;
                vtxState.metadata.powerTableCount = 4;
                
                impl_vtxDevice.capability.powerNames = (char **)trampPowerNames_5G8_1600;
                impl_vtxDevice.capability.powerCount = 4;`

[image]

Mauritz17, 2023-10-17:
I got it with Foxeer reaper Infinity 5W VTX. Some sellers provite some Tables for Betaflight. but when you serch, you could find the list what value mean the real Output Power. you coud set the command "set_vtx_max_power_override = 200" one of the 4 possible Value stables. 
25, 100, 200, 200, 200  "set_vtx_max_power_override = 200"
25, 100, 200, 400, 400  "set_vtx_max_power_override = 400"
25, 100, 200, 400, 600  "set_vtx_max_power_override = 600"
25, 100, 200, 500, 800  "set_vtx_max_power_override = 800"
some VTX youse one of this Value stable to controll the real output power. there is the table of Foxeer Infinity 5W:
("value": 25,
 "label": "50 ")
( "value": 100,
 "label": "500")
("value": 200,
"label": "1W ")
("value": 400,
 "label": "2.5" )
("value": 600,
 "label": "5W ")
default was the 800mW Setting, this Values was Mismatch with the Values to feed my VTX.
I yoused the 600mW setting to get full controll over the VTX.
i can no select via the OSD full output Power of 5W. but the OSD tell me only the values of 600mW. I can selct 600mW in OSD to get 5W Output in real world. the values from OSD is now mismatch with my VTX but it works great.

Stemplar42, 2023-10-17:
> quite a simple change, I've managed to do it on my hglrcf722 board using the zeus 1600mw vtx. you need to add two sections of code to src\main\io\vtx_tramp.c towards the bottom (first bit of code is [...]

Tanks Sketch4x4, I've been researching the source code for a while now, and I was able to come up with the same solution as you. I have it working perfectly now on a GEPRC 1600 mW and on an FC Wing-FC DOF 10.

sketch4x4, 2023-10-17:
> Tanks Sketch4x4, I've been researching the source code for a while now, and I was able to come up with the same solution as you. I have it working perfectly now on a GEPRC 1600 mW and on an FC [...]

No worries, glad I could help. Granted this should be implemented in future firmware but at the rate this high powered vtx's are coming out it's more than likely going to be a DIY to get the exact outputs required. 

Does anyone know if it possible to get actual power outputs on the osd instead of the 1-4 numbers?

Tonynux, 2024-01-08:
I'm facing this issue on a GEPRC RAD MINI 5.8 GHz. I can't select the 1 W power.

artempolikarpov, 2024-01-09:
I have the same issue with iFlight Blitz 2500mw. @DzikuVx  Is there any chance to push this fix in nearest patch?

ivanent, 2024-02-12:
Has there been any movement on this or is an easy workaround known ? I just built a quad with inav 7 and a foxeer reaper extreme and can't figure out how to use it ,is it possible in inav yet to set the vtx to the the 2.5w output its supposed to do

Mauritz17, 2024-02-13:
> Has there been any movement on this or is an easy workaround known ? I just built a quad with inav 7 and a foxeer reaper extreme and can't figure out how to use it...

Foxeer knows that there are problems with inav, that's why they have their 
Tramp protocol so that the values do not correspond to the real values. That's why I give you the command you should insert in the Inav CLI and test if it works.
Provided you have already unlocked the VTX. 
In the manual you have written that the buttons on the VTX are disabled as long as the Tramp protocol port is connected. So disconnect this contact while you unlock it,
Foxeer has also published a frequency table on the website which command of the Tramp protocol corresponds to the real output power.
I explained it earlier in this chat.
Now the table:

Translated with DeepL.com (free version)
"powerlevels_list": [
            {
                "value": 25,
                "label": "25 "
            },
            {
                "value": 100,
                "label": "200"
            },
            {
                "value": 200,
                "label": "500"
            },
            {
                "value": 400,
                "label": "1.5"
            },
            {
                "value": 600,
                "label": "2.5"}

Now the command in the CLI:
"set_vtx_max_power_override = 200"