iNavFlight/inav #8348: MSP connection successfully closed

audris123, 2022-08-21:
## Current Behavior
I use Matek H743 wlite board and inav 5.0 with linux
I can't connect board to inav but I can flash firmware

2022-08-22 @ 01:26:27 -- MSP connection successfully opened with ID: 1
2022-08-22 @ 01:26:37 -- MSP connection successfully closed

## Steps to Reproduce

## Expected behavior

## Suggested solution(s)

## Additional context

- FC Board name and vendor: Matek H743 wlite
- INAV version string: inav 5.0
- PC : Linux Ubuntu 20.04

audris123, 2022-09-01:
I don't know how, but I solved the problem.

closed 2022-09-01 by audris123

codegrue, 2022-09-02:
I have this same issue. Still not resolved. Thinks I have tried:

- Ran zadig driver tool. tried all 4 drivers
- Ran ImpulseRC Driver Fix
- Flashed back to v4, same issue

codegrue, 2022-09-02:
Solved (Kind of...)

Issue was caused by having an SD Card in the FC:

https://github.com/iNavFlight/inav/issues/7759

040572, 2023-05-17:
yes my issue was same and it solved by when i removed sd card

040572, 2023-05-17:
I dont know what the reason behind this ,and one issue I faced that twice i upload firmware in speedebee f405 its stop responding unable to load fw after 2 attempt. pls anybody know the problem why its happened

vasilisaaue, 2024-02-22:
> I dont know what the reason behind this ,and one issue I faced that twice i upload firmware in speedebee f405 its stop responding unable to load fw after 2 attempt. pls anybody know the problem why its happened

You resolved?

weblint, 2025-12-31:
actually i have the same problem. After flashing SpeedyBee F405 V4 with iNAV 9xxx (unstable relases or with official 8.0.1) i can't connect to flight control.
Log:
2025-12-31 @ 15:47:06 -- Connected to: COM3 @ 115200 baud

2025-12-31 @ 15:47:06 -- MSP connection successfully opened with ID: 3

2025-12-31 @ 15:47:06 -- Error: Opening COM3: Access denied

2025-12-31 @ 15:47:06 -- MSP connection successfully closed