iNavFlight/inav #8355: SKYSTAR F722 HD PRO2 - accelerometer not working

mauroS123, 2022-08-24:
## accelerometer not working on INAV 5.1 -  Same board tested with betaflight and all working fine.

## Steps to Reproduce

1.
2.
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor: skystar f722 hd pro 2
- INAV version string: 5.1

JulioCesarMatias, 2022-08-24:
Are you sure it's PRO2? maybe it's the PRO3

JulioCesarMatias, 2022-08-24:
ok, we need to add the MPU6000 drivers for this target

mauroS123, 2022-08-24:
It is PRO2 and have BMI270 gyro. 
If I set to "AUTO", on first time the gyro works, but after a reboot it no more works again

JulioCesarMatias, 2022-08-24:
hmmm... I had seen on the web that the IMU was an MPU-6000, but that's ok...

mauroS123, 2022-08-24:
No, it's BMI270.  If I set "auto" it found BMI270 and working. If I disconnect USB and reconnect remain on BMI270 but not works. Then I set again "auto", reboot and it works again one time.

mauroS123, 2022-08-24:
this is the pro3 and also have BMI270     http://skystars-rc.com/Products/ESC/2022/0716/103.html

9886834239, 2022-09-09:
> ok, we need to add the MPU-6000 drivers for this target

Yes, Skystars F722 HD Pro2 uses MPU-6000. Same is the issue with my board, can't detect the gyro and accelerometer after flashing iNav.

mauroS123, 2022-09-09:
The board have bmi270.  If you manual set it works until first reboot

9886834239, 2022-09-09:
I could see mpu6000 in my Fc, please consider this.

mauroS123, 2022-09-09:
My board v2 have bmi 270 and works only if I manually set that. If I set mpu6000 dont works.

JulioCesarMatias, 2022-09-09:
@9886834239  I can include the drives in the target

@mauroS123  I don't know how to solve your problem

mauroS123, 2022-09-16:
I would want to retry to flash INAV. Can you confirm that the target have now the driver ?

9886834239, 2022-09-21:
@JulioCesarMatias  
Hi,
 Have you uploaded the drivers for MPU6000?
Could you please tell us when we can expect this update?

JulioCesarMatias, 2022-09-21:
[inav_6.0.0_SKYSTARSF722HD.zip](https://github.com/iNavFlight/inav/files/9620237/inav_6.0.0_SKYSTARSF722HD.zip)

mauroS123, 2022-09-22:
Thanks. Now working.

JulioCesarMatias, 2022-09-22:
> Thanks. Now working. INAV have found and set automaticaly BMI270 gyro.

strange the BMI-270 now works fine for you... I just added the MPU-6000 drivers

mauroS123, 2022-09-22:
Works perfects. Only a little bug during calibration sequence. After the calibration on the second position (upside down) I need to recalibrate the first one that loses its value.

mauroS123, 2022-09-22:
I think now it working due a bug fix inside inav 6.0 release and not due mpu6000 driver.

MrD-RC, 2022-09-22:
> Works perfects. Only a little bug during calibration sequence. After the calibration on the second position (upside down) I need to recalibrate the first one that loses its value.

It looks like the IMU is on the underside if that flight controller. So the first position should have the FC upside down. This would have the IMU right side up.

mauroS123, 2022-09-22:
Yes imu is on the underside of the board

closed 2022-09-22 by JulioCesarMatias

9886834239, 2022-09-23:
Thank you, now I could able to run my fc with MPU6000 drivers.