iNavFlight/inav #8357: FOXEERF722V3 Target

oliver602, 2022-08-25:
## Current Behavior
Can't flash the Foxeer F722 V3 from INAV configurator.

## Suggested Solution
Create a target

## Who does this impact? Who is this for?
People with this flight controller who want to use INAV

## Additional context
Betaflight target:
https://github.com/betaflight/unified-targets/blob/master/configs/default/FOXE-FOXEERF722V3.config

JulioCesarMatias, 2022-08-25:
can you test this build? >> 
[inav_6.0.0_FOXEERF722V3.zip](https://github.com/iNavFlight/inav/files/9428727/inav_6.0.0_FOXEERF722V3.zip)

For this, iNav Configurator 6.0 will be needed, you can download it here >> http://seyrsnys.myzen.co.uk/inav-configurator-next/

oliver602, 2022-08-26:
The accelerometer is not automatically detected. If I manually select the BMI270 it says unavailable. The LED blinks fast. Let me know what other information I can provide.

```
Entering CLI Mode, type 'exit' to return, or 'help'

# version
# INAV/FOXEERF722V3 6.0.0 Aug 25 2022 / 20:13:26 (a38bce9b)
# GCC-10.2.1 20201103 (release)

# status
System Uptime: 23 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 0.00V (1S battery - NOT PRESENT)
CPU Clock=216MHz
STM32 system clocks:
  SYSCLK = 216 MHz
  HCLK   = 216 MHz
  PCLK1  = 54 MHz
  PCLK2  = 108 MHz
Sensor status: GYRO=UNAVAILABLE, ACC=NONE, MAG=NONE, BARO=NONE, RANGEFINDER=NONE, OPFLOW=NONE, GPS=NONE, IMU2=NONE
Stack size: 6144, Stack address: 0x20010000, Heap available: 1840
I2C Errors: 5, config size: 8974, max available config: 16384
ADC channel usage:
   BATTERY : configured = ADC 1, used = none
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = none
  AIRSPEED : configured = none, used = none
System load: 11, cycle time: 510, PID rate: 1960, RX rate: 9, System rate: 9
Arming disabled flags: ANGLE CAL HWFAIL RX CLI SETTINGFAIL DSHOTBEEPER
### ERROR: Invalid setting: nav_fw_launch_abort_deadband
VTX: not detected

# tasks
Task list         rate/hz  max/us  avg/us maxload avgload     total/ms
 0 -       SYSTEM       9      24       2    0.5%    0.5%     10152
 1 -          PID    1972      79      54   16.0%   11.1%      4431
 2 -         GYRO    9615      11       1   11.0%    1.4%       225
 3 -           RX       9      45      36    0.5%    0.5%        15
 4 -       SERIAL      99   42276       3  419.0%    0.5%        66
 6 -  TEMPERATURE      99      11       1    0.6%    0.5%         6
 7 -       BEEPER      99      20       7    0.6%    0.5%        31
18 -          CMS      49       3       1    0.5%    0.5%         2
21 -      VTXCTRL       4       1       1    0.5%    0.5%         0
22 -  PROGRAMMING       9      16      10    0.5%    0.5%         4
24 -          AUX      99      10       4    0.5%    0.5%        13
25 - SPORT MASTER     493      10       1    0.9%    0.5%        21
Task check function             6       2                         0
Total (excluding SERIAL)                    32.1%   17.0%

# resource
IO:
----------------------
A00: FREE 
A01: FREE 
A02: FREE 
A03: FREE 
A04: BEEPER OUT
A05: FREE 
A06: FREE 
A07: FREE 
A08: MOTOR2 OUT
A09: MOTOR1 OUT
A10: FREE 
A11: USB IN
A12: USB OUT
A13: FREE 
A14: FREE 
A15: FREE 
B00: FREE 
B01: FREE 
B02: FREE 
B03: FREE 
B04: FREE 
B05: FREE 
B06: FREE 
B07: SERIAL1 UART RX
B08: I2C1 SCL
B09: I2C1 SDA
B10: FREE 
B11: FREE 
B12: FLASH CS
B13: SPI2 SCK
B14: SPI2 MISO
B15: SPI2 MOSI
C00: FREE 
C01: FREE 
C02: FREE 
C03: SPI_PREINIT CS
C04: FREE 
C05: FREE 
C06: FREE 
C07: FREE 
C08: MOTOR4 OUT
C09: MOTOR3 OUT
C10: FREE 
C11: FREE 
C12: FREE 
C13: FREE 
C14: FREE 
C15: LED1 OUT
D02: FREE 

# 
```

JulioCesarMatias, 2022-08-26:
test this other build >> 
[inav_6.0.0_FOXEERF722V3.zip](https://github.com/iNavFlight/inav/files/9429170/inav_6.0.0_FOXEERF722V3.zip)

oliver602, 2022-08-26:
Still the same as above

JulioCesarMatias, 2022-08-26:
> Still the same as above

ok, thanks for testing! Unfortunately my help ends here, I don't know what else to do without having the FC in hand. Sorry

JulioCesarMatias, 2022-08-26:
I have use last attempt >> 
[inav_6.0.0_FOXEERF722V3.zip](https://github.com/iNavFlight/inav/files/9436156/inav_6.0.0_FOXEERF722V3.zip)

oliver602, 2022-08-26:
That worked! the BMI270 is detected. UART6 is missing from the configurator. I'll continue to test more.
```
version
# INAV/FOXEERF722V3 6.0.0 Aug 25 2022 / 20:13:26 (a38bce9b)
# GCC-10.2.1 20201103 (release)

# status
System Uptime: 21 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 0.00V (1S battery - NOT PRESENT)
CPU Clock=216MHz, GYRO=BMI270, ACC=BMI270
STM32 system clocks:
  SYSCLK = 216 MHz
  HCLK   = 216 MHz
  PCLK1  = 54 MHz
  PCLK2  = 108 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=NONE, RANGEFINDER=NONE, OPFLOW=NONE, GPS=NONE, IMU2=NONE
Stack size: 6144, Stack address: 0x20010000, Heap available: 1820
I2C Errors: 5, config size: 8974, max available config: 16384
ADC channel usage:
   BATTERY : configured = ADC 1, used = none
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = none
  AIRSPEED : configured = none, used = none
System load: 6, cycle time: 516, PID rate: 1937, RX rate: 9, System rate: 9
Arming disabled flags: RX CLI SETTINGFAIL
### ERROR: Invalid setting: nav_fw_launch_abort_deadband
VTX: not detected

# tasks
Task list         rate/hz  max/us  avg/us maxload avgload     total/ms
 0 -       SYSTEM       9      24       2    0.5%    0.5%      5054
 1 -          PID    1980     131     107   26.4%   21.6%      5180
 2 -         GYRO    2754      40      26   11.5%    7.6%      1788
 3 -           RX       9      46      38    0.5%    0.5%         9
 4 -       SERIAL      99   42346       3  419.7%    0.5%        60
 6 -  TEMPERATURE      98      18       2    0.6%    0.5%         5
 7 -       BEEPER     100      18       4    0.6%    0.5%        11
18 -          CMS      49       3       1    0.5%    0.5%         1
21 -      VTXCTRL       4       6       1    0.5%    0.5%         0
22 -  PROGRAMMING       9      17      10    0.5%    0.5%         2
24 -          AUX      99       8       4    0.5%    0.5%         8
25 - SPORT MASTER     492       6       1    0.7%    0.5%        12
Task check function             5       1                         0
Total (excluding SERIAL)                    42.8%   33.7%

# resource
IO:
----------------------
A00: FREE 
A01: FREE 
A02: FREE 
A03: FREE 
A04: BEEPER OUT
A05: SPI1 SCK
A06: SPI1 MISO
A07: SPI1 MOSI
A08: MOTOR2 OUT
A09: MOTOR1 OUT
A10: FREE 
A11: USB IN
A12: USB OUT
A13: FREE 
A14: FREE 
A15: FREE 
B00: FREE 
B01: FREE 
B02: MPU CS
B03: FREE 
B04: FREE 
B05: FREE 
B06: FREE 
B07: SERIAL1 UART RX
B08: I2C1 SCL
B09: I2C1 SDA
B10: FREE 
B11: FREE 
B12: FLASH CS
B13: SPI2 SCK
B14: SPI2 MISO
B15: SPI2 MOSI
C00: FREE 
C01: FREE 
C02: FREE 
C03: SPI_PREINIT CS
C04: MPU EXTI
C05: FREE 
C06: FREE 
C07: FREE 
C08: MOTOR4 OUT
C09: MOTOR3 OUT
C10: FREE 
C11: FREE 
C12: FREE 
C13: FREE 
C14: FREE 
C15: LED1 OUT
D02: FREE 

# 
```

JulioCesarMatias, 2022-08-26:
I can compile it for you with UART6, but the S5 and S6 motor/servo outputs will no longer work...

oliver602, 2022-08-26:
No its good for me, thanks for your help.

JulioCesarMatias, 2022-09-04:
Is the IMU orientation correct?

oliver602, 2022-09-04:
Yes, I didn't have to make any changes.

JulioCesarMatias, 2022-09-06:
I just researched today, and saw that this version is not official from Foxeer, I didn't know that... so, by the rules, I don't think we should add support for an unofficial version.

closed 2022-09-06 by JulioCesarMatias

oliver602, 2022-09-06:
This is the flight controller. https://www.foxeer.com/foxeer-f722-v3-flight-controller-x8-dji-type-c-g-430

luigifpv, 2023-06-19:
Would it be possible to compile it with UART6 for me?