iNavFlight/inav #8367: The CJMCU-055 board can not be detected over the I2C in INAV

asgarPL, 2022-08-30:
## Current Behavior
The CJMCU-055 board can not be detected by INAV as a IMU2 device over I2C connected to the
CL1/DA1 pads (near the DFU button) of the MATEK F765 FC. The IMU2 status icon is red and the IMU2 status is
NONE in the CLI.

## Steps to Reproduce

1. Connect CJMCU-055 board to the CL1/DA1 pads (near the DFU button) of the MATEK F765 FC.
[F765-WING_Wiring](http://www.mateksys.com/wp-content/uploads/2019/08/F765-WING_Wiring.jpg)
2. ENABLE BNO055 imu2 with the "set imu2_hardware=BNO055" CLI command.
3. Check IMU2 status icon.

## Expected behavior
The CJMCU-055 imu2 board successfully detected (no red IMU2 status icon, nor the "imu2: NONE" status in the CLI).

## Additional context
I have basic MATEK F765 setup with the MATEK GPS module connected according the MATEK INAV wiring picture,
without any additional devices like DigitalAirSpeed, OLED, Bluetooth and etc.
The CJMCU-055 board seams to be alive, since it can be detected under the 0x28 address in my testing ARDUINO setup,
thus board pads soldering configuration is not an issue here.
The correctness of the SDL/SCL connections was verified, in a way that:
- the CJMCU-055 SDL line is electrically connected only with the SDL pad of the QMC5883 magnetometer
- the CJMCU-055 SCL line is electrically connected only with the SCL pad of the QMC5883 magnetometer
- the SDL and SCL pads are not shorted against themselves
- the CJMCU-055 board is powered with the +5V.

I'm not sure whether all I2C1 bus is impacted, but when the CJMCU-055 board is enabled and connected,
then I2C problems are reported only for the MATEK QMC5883 magnetometer and its status icon goes red.

I was trying all available I2C bus speed options, without any success.
I was trying adding also additional pullup resistors with the values 2.2k and 4.7k, also without any success.

I can not get working the CJMCU-055 board in the UART configuration either, however I prefer I2C, due to
5V voltage being applied to the communication interface configuration pads of the BNO055 chip,
which operates rather in 3.3V domain.

I have seen some planned tasks about dropping the BNO055 over I2C, so are there any well known I2C issues for the STM chip
and particular BNO055 clone? Well, it was straight easy for me to communicate with it under ARDUINO eco-system.
Would it be hard to enable such I2C communication by myself, in context of officially dropping this I2C configuration?
I was looking to the code of the BNO055 library, used in inav. There is some logging with the HAL_ERROR/HAL_INFO statements.
Can it be somehow easily captured and analysed?
The CJMCU-055 I2C problems can be caused by some well known MATEK F765 hardware design faults? Can it be somehow mitigated?

I would be very grateful for helping me figuring out the I2C problem. I've managed already to mitigate the AHD problem significally,
with the 2.6 INAV ACC/GYRO sensors fusion improvments, however in my case this is still not bullet proof approach. Thx in advance. 

# version
# INAV/MATEKF765 4.1.0 Feb 11 2022 / 11:11:49 (611d26f4)
# GCC-10.2.1 20201103 (release)

- FC Board name and vendor:
(EOL)MATEK F765 FC (INAAV tag:M765, version: 0)
- INAV version string:
INAV, version: 4.1.0

The config file:
[INAV_I2C_CJMCU-055_config.txt](https://github.com/iNavFlight/inav/files/9455438/INAV_I2C_CJMCU-055_config.txt)

JulioCesarMatias, 2022-08-31:
the answer to your problem is simple... MatekF765 doesn't have the drivers for the BMI-055 enabled on the backend. To solve your problem, you need to modify the iNav files, and make a compilation for your target. Here we have files available teaching how to configure the development environment to compile the algorithm >> https://github.com/iNavFlight/inav/tree/master/docs/development

JulioCesarMatias, 2022-08-31:
Well... I think this can be closed.

closed 2022-08-31 by JulioCesarMatias