iNavFlight/inav #8369: Switching into or out of pitmode doesn't work with Eachine Nano v3 VTX + softserial+ INAV 5.1

cmanley, 2022-09-01:
## Current Behavior
If I disable pitmode by switching to a power level on the VTX using the button, then I am able to switch between power levels 1 (25mw) to 4 (400mw) from INAV. What INAV is unable to do however is switch out of pitmode when arming or into pitmode when disarming.
Unfortunately I'm stuck after many hours of trying and unable to find much documentation in the wiki on VTX communication and possible issues and solutions.
This may be related to https://github.com/iNavFlight/inav/issues/5844
and/or https://github.com/iNavFlight/inav/issues/5896

## Steps to Reproduce
1. Connect VTX RX to softserial 1 TX.
2. Set vtx_power = 4 in INAV.
3. Set vtx_low_power_disarm to ON in INAV.
4. If the VTX is already in pitmode, then arming won't make it exit pitmode.
5. If the VTX is not in pitmode (after setting a power level using VTX button), then INAV can change the VTX power level to 4 on arm, and back to 1 on disarm, but it won't put the VTX into pitmode.

## Expected behavior
INAV can control pitmode state of VTX.

## Suggested solution(s)
Perhaps more documentation if some issues/solutions are known.

## Additional context
```# 
serial 0 2 115200 115200 0 115200
serial 30 4096 115200 115200 0 19200

vtx_halfduplex = ON
Allowed values: OFF, ON

# I've tried both options here:
vtx_smartaudio_early_akk_workaround = OFF
Allowed values: OFF, ON

# I've tried both options here:
vtx_smartaudio_alternate_softserial_method = ON
Allowed values: OFF, ON

# I've tried both options here:
vtx_softserial_shortstop = ON
Allowed values: OFF, ON

# I've tried both options here:
vtx_smartaudio_stopbits = 1
Allowed range: 1 - 2

vtx_band = 4
Allowed range: 0 - 5

vtx_channel = 4
Allowed range: 1 - 8

vtx_power = 4
Allowed range: 1 - 5

vtx_low_power_disarm = ON
Allowed values: OFF, ON, UNTIL_FIRST_ARM

vtx_pit_mode_chan = 4
Allowed range: 1 - 8

vtx_max_power_override = 0
Allowed range: 0 - 10000

vtx_frequency_group = FREQUENCYGROUP_5G8
Allowed values: FREQUENCYGROUP_5G8, FREQUENCYGROUP_2G4, FREQUENCYGROUP_1G3
```
- FC Board name and vendor: Matek F411 WTE (it has 2 UARTs and 2 softserial ports, the UARTS are used for GPS and CRSF and softserial 1 tx is used for IRC Tramp)
VTX: [Eachine Nano v3](https://www.eachine.com/Eachine-Nano-V3-VTX-5_8GHz-40CH-25-or-100-or-200-or-400mW-Switchable-FPV-Transmitter-Built-in-Microphone-Support-OSD-or-Pitmode-or-IRC-Tramp-For-RC-Racing-Drone-p-2114.html)
I haven't tried a previous INAV version.
- INAV version string: 5.1

b14ckyy, 2024-03-20:
PIT mode is in general not supported by INAV as many many VTX cause issues with remote controlled PIT mode.

closed 2024-03-20 by b14ckyy

anti-vaxxer, 2024-05-07:
I have HGLRC Zeus nano VTX which uses IRC Tramp and I can change band,channel & power level, but the pit mode setting in OSD menu doesn't do anything.
Pit mode with my VTX works in Betaflight and Ardupilot.

I want to be able to use pit mode because otherwise my VTX becomes too hot while disarmed.

anti-vaxxer, 2024-05-12:
can you please open this issue? I want IRC Tramp pit mode to work in INAV

MrD-RC, 2024-05-12:
Your VTX should not get hot if you set it to 25mw. Which is currently possible. 

Pitmode has never been supported in INAV. Often, it causes more problems than it solves.

cmanley, 2024-05-12:
It would be nice if that detail was included in the documentation ( https://github.com/iNavFlight/inav/blob/master/docs/VTx.md ) to prevent similar tickets and questions in the future. Or am I reading the wrong documentation?

anti-vaxxer, 2025-02-28:
I understood from b14ckyy and MrD-RC's comments above and [there](https://github.com/iNavFlight/inav/issues/8365#issuecomment-2012796704) that INAV doesn't support pitmode,
but ABLomas [said](https://github.com/iNavFlight/inav/pull/10261#issuecomment-2288377140) pitmode works with INAV on his SmartAudio VTX.
So evidently INAV supports pitmode with SmartAudio.

I found a [comment](https://github.com/iNavFlight/inav/blob/38dfe9a4edb37cba839737d0fb0de0d624c512c5/src/main/io/vtx_tramp.c#L319) in INAV tramp code _''only disabling pitmode supported''_, so looks like pitmode for IRCTramp has not been implemented.
Does anyone know how to implement it? because I don't know how.