iNavFlight/inav #8373: HGLRCF722 Mini won't detect HGLRC M80 Pro Compass (QMC5883)

tonykakuuu, 2022-09-04:
## Current Behavior

GPS works flawlessly, compass/magnetometer is not recognized. 

## Steps to Reproduce

1. Solder [M80 Pro to HGLRCF722](https://user-images.githubusercontent.com/47401248/188315954-1e132a72-e9f6-48e0-8e33-e10d58a6d13b.jpg)
2. Plug in battery and connect to INAV configurator
3. Select QMC5883 (or AUTO) as the magnetometer
4. Get I2C errors + MAG=UNAVAILABLE under CLI -> status

## Expected behavior
Get no I2C errors and have magnetometer work. 

## Suggested solution(s)
Unknown. Issue #8206 may or may not be related, as the line 
`#define BARO_I2C_BUS            BUS_I2C1`
in the target.h may also create a conflict with
`#define MAG_I2C_BUS             BUS_I2C1  
`
I did however flash a custom build of INAV without the first line and it changed nothing. I can't find good documentation for this revision of the board and I'm unsure if maybe the I2C1_SCL and I2C1_SDA pads have a different identifier. 

## Additional context
There are at least two revisions of this board, maybe more when judging by other forum posts. [Here are two diagrams I found, old one on the right, left is the one I have.](https://user-images.githubusercontent.com/47401248/188316270-57b6238c-2de7-40c0-a175-371967edb8c1.jpg)

---

- FC Board name and vendor: HGLRCF722
- INAV version string: INAV/HGLRCF722 5.0.0 Sep  4 2022 / 09:49:14 (3a379a7c) (But the result is the same on official 5.0.0 and 5.0.1)

JulioCesarMatias, 2022-09-05:
well... I don't have FC in hand to test/reproduce the problem... But once, a while ago, a friend of mine had the same problem, but it was on a Matek H743, we spent a little time until I found the problem... and the problem was that his FC only recognized the Mag with the I2C refresh rate at 200Khz. You could try this and see if anything will change...

And about Baro and Compass using the same BUS (BUS_I2C1), it's not the problem, no conflict is generated here, forget about it...

JulioCesarMatias, 2022-09-05:
Is your flight controller barometer detected?

tonykakuuu, 2022-09-06:
Baro is working on 5.1.0 with battery plugged in. 

`
Entering CLI Mode, type 'exit' to return, or 'help'

# status
System Uptime: 21 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 16.38V (4S battery - OK)
CPU Clock=216MHz, GYRO=BMI270, ACC=BMI270, BARO=DPS310
STM32 system clocks:
  SYSCLK = 216 MHz
  HCLK   = 216 MHz
  PCLK1  = 54 MHz
  PCLK2  = 108 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=UNAVAILABLE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK, IMU2=NONE
Stack size: 6144, Stack address: 0x20010000, Heap available: 1924
I2C Errors: 60, config size: 9680, max available config: 16384
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = none, used = none
System load: 11, cycle time: 502, PID rate: 1992, RX rate: 49, System rate: 9
Arming disabled flags: HWFAIL RX CLI
VTX: band: R, chan: 8, power: 1 (25 mW), freq: 5917 MHz`

tonykakuuu, 2022-09-06:
Baro is working with battery plugged in on 5.1.0. Changing the I2C speed did not change the result for the magnetometer.

VardenTheOne, 2022-09-06:
I have [Skystars F722 HD Pro3](https://www.rotorama.cz/cms/assets/docs/949e0fab7f2d88fc0847ae1605036a66/11101-0/mmexport1655980256421.jpg) and `Readytosky Ublox NEO M8N GPS`. Behaviour is the same as described above. I checked compass with Arduino I2C scanner and it replies, but I did not try to get any data from it.

Also how does FAKE Magnetometer should work in Inav? It does not generate any additional I2C commands and errors (as I expected). But it shows compass icon in red. I was sure that it will show some emulated working compass, but it does not work. Also it is not available in sensors tab.

tonykakuuu, 2022-09-06:
I just realized that Youtuber Painless360 used this exact FC-GPS-Mag combination when [demonstrating the Rekon7](https://www.youtube.com/watch?v=WMKwy232NmY). For him, the magnetometer seemed to work instantly, though this was on 3.0.1. I'll test this version as soon as I can but for now this makes me believe my compass might simply be dead.

tonykakuuu, 2022-09-18:
HGLRC was very cooperative in their troubleshooting and the German reseller issued a replacement device. I can now confirm that my unit simply had a defective magnetometer. Sorry for all the trouble, next time I'll try to debug more thoroughly (I don't have any I2C debug tools unfortunately and all my other FCs don't have INAV targets :/)

closed 2022-09-18 by tonykakuuu