iNavFlight/inav #8381: problem with baro INAV 5.0.0 - target MAMBAF405US_I2C

Torchi, 2022-09-11:
Same problem has been probably already issued, I attach here my description and data.

Model already flown 3-4 times. I had already noticed a minor problem with altitude. This time was a severe problem.
Took off in HOLD and sky rocketed (see movie attached).
Switched to ACRO / ANGLE in order to descend and land.  Altitude has been all the time obviously wrong (see movie attached).

I attach flight blackbox and diff.

https://user-images.githubusercontent.com/18173871/189526440-8cb8cc17-ab13-4355-a17f-84578d34bec3.mp4

https://user-images.githubusercontent.com/18173871/189526443-ff25e1b2-d1de-4298-a4a3-81e8b5693dff.mp4

[INAV_cli_Super_8_20220908_163058.txt](https://github.com/iNavFlight/inav/files/9542449/INAV_cli_Super_8_20220908_163058.txt)
[blackbox_log_2022-09-10_112811.TXT](https://github.com/iNavFlight/inav/files/9542450/blackbox_log_2022-09-10_112811.TXT)

stronnag, 2022-09-11:
Not a happy baro.
[image]

breadoven, 2022-09-11:
Is the Baro chip exposed to direct sunlight by any chance ?

Torchi, 2022-09-12:
Yes, it can be exposed to sunlight.
Here is a picture of quad.   A circle shows where FCB is located: it is under a carbon fiber roof  (that is narrow and with small openings).

[image]

This is the side of the FCB facing up (and approximatively rotated the same).
In the circle I have outlined what "I believe" is the baro chip.

[image]

I guess this is the reason of the baro malfunctioning ... ?!

breadoven, 2022-09-12:
I've noticed the altitude varies wildly if the chip has direct sunlight on it. The BMP280 chip is light sensitive apparently, gets in through the small hole in the case. I've seen errors of over 80m. Not as much as in your case but easy enough to rule this out by covering the chip with some open cell foam which is something you should do anyway to damp out pressure changes from prop wash etc.

Torchi, 2022-09-12:
Ok, I am convinced  (also calmed because i have seen that chip is used in many FCB types and the problem is frightening when it manifests). I shall do, with this model, as you suggest.

obiwikii, 2022-09-16:
perfect now for me

closed 2024-03-20 by b14ckyy