iNavFlight/inav #8390: Center the center marker (spread over 2 or 4 chars) and the AHI on the 30 char wide grid.

cmanley, 2022-09-12:
## Current Behavior
There is no center character in a 30 character grid, but INAV has been placing the center marker either left or right (currently) of center. This is visually confusing to some, and I was one of them trying to figure out why my cameras needed to be facing slightly left in order to get the OSD centered when in reality the center marker was off.
With PAL there is also no vertical center with 16 rows, but that is much less important than horizontal centering.
The AHI also has one dot too many on one side (probably so that it looks ok in combination with the out-of-center center marker).

## Desired Behavior
A horizontally centered AHI and centered center marker so that hundreds/thousands of INAV users can finally fly and align their cameras straight.
If it can be vertically centered too (for PAL at least), then that'll be nice to have.

## Suggested Solution
As for the AHI: add another dot on the left side.
As for the center marker: share 2 or 4 characters to get it dead center.

## Who does this impact? Who is this for?
Every INAV pilot that uses AHI and/or center marker to fly with or uses it to align their cameras with (typically on fixed wings) as I did a few times until I figured out that it is too coincidental that all my camera image sensors or lenses seemed offset to the same side by the same angle.

## Additional context
[This issue was mentioned before](https://github.com/iNavFlight/inav/issues/4911)
I can't judge the amount of available font characters that can be used (or sacrificed) for the center marker.

Jetrell, 2022-09-14:
This would require too many characters to be changed. And end up being a waste of free characters, for very little gain.
All these would require changing.
* 8 Crosshairs in total. 
* Horizontal and Vertical AHI markers.
* Home location symbol (when using homing radar)
* Homing arrow set

And considerable code changes would be required in both the firmware and Configurator.

MrD-RC, 2022-09-14:
I agree with @Jetrell. This would be far too much of a character sacrifice for very little benefit.

closed 2022-09-14 by MrD-RC

MrD-RC, 2022-09-14:
It is really not needed. No other firmware does it. It is simply a limitation of the character based OSD.

It is documented https://github.com/iNavFlight/inav/wiki/OSD-Hud-and-ESP32-radars#accuracy-and-limitations

cmanley, 2022-09-16:
Alright. Thanks for taking the time to respond and the link. I must've missed that when perusing the wiki.

cmanley, 2022-10-28:
Someone has done this they way I imagined for wtfosd. For reference:
https://github.com/Knifa/material-osd/releases/tag/v0.0.3