iNavFlight/inav #8391: FW emergency braking by switch

zakopanypouszy, 2022-09-13:
Hello, I had an idea that could save our equipment in situations when it might be too late to pull it up and we know that we will crash. Is it possible to program the braking function? So that by flipping the switch, the engine starts to rotate in the opposite direction and thus reduces the speed of the ship as much as possible.

Jetrell, 2022-09-14:
This can be accomplished via the BLheli32 reversible motor function, and the iNav logic conditions.
But you should be aware, reverse thrust will also reduce the control authority and stabilization, by lowering the models airspeed. 

In the end, it could reduce any chance of you recovering the model from a bad situation... You need to be 100% sure you're going to crash before activating it.

zakopanypouszy, 2022-09-14:
I once saw on youtube (unfortunately I can't find it again) where someone had it done in fw and used it for braking during landing, when flying low above the ground (about 1.2 meters) he was in the "landing" place he was simply reversing the thrust and wing after it just fell to the ground where the engine turned.

I think that in emergency situations such a possibility would be helpful, everyone who flies fw and has already had some crashes can definitely judge when it is too late to steer and the only thing that can help are the brakes.

If Inav supports turtle mode in multirotr, why wouldn't it be possible to somehow assign switch to reverse thrust?

MrD-RC, 2022-09-14:
I've seen that video. It's a bad idea. On that particular plane, he should have used crow braking.

By using reversible ESCs, you're effectively halving the resolution of the throttle. You'd also need to remap the throttle stick to only use the upper half of the channel output, or have it centre sprung.

b14ckyy, 2022-09-21:
To add to Darren's concerns. 
Yes for analog protocols you lose half the resolution. But with DShot a reverse command by the flight controller is doable as it is in turtle mode. 

BUT!

if you are about to crash a plane like in a stall spin or similar, it just needs 1 or 2 seconds at most to recover the plane if you have trained it. If you are so low that you can't recover you only have less than 2 seconds to decide if you want to go to reverse and you have to think if the point when to reverse as it would be useless if the plane is in a flat spin already. So you have to analyze the kind of control loss, decide if you can recover or not, then move your finger to the switch and flick it. All that in less than 2 seconds? If you can do that properly in a panic moment, then you are a machine and not human anymore. 

also reverse thrust on a pusher wing will make it absolutely unpredictable how it will behave with that reverse airflow over the fuselage and inner wing. It could also flip around and suck itself into the ground at high speed instead. 

The next problem is that for reverse thrust, you have to stop the motors completely and then reverse direction. This can be done fast on small quad motors and props but if you spin a 8" prop or even bigger, a sudden stop and reverse will either take at least a second or it will instantly kill your ESC and/or Motor. 

Overall there are so many problems involved that it is just not worth the effort for that one in a million chance where it might reduce damage in the plane. 

Just cut the throttle. If the plane is nose down you can quickly recover due to correct airflow and if it is in a spin, the impact speed will be low anyway to cause little damage.

zakopanypouszy, 2022-09-23:
Thank you very much for all the answers, my doubts have been dispelled.

closed 2022-09-23 by zakopanypouszy