iNavFlight/inav #8398: TMOTOR F7 Pro - Gyro Problem

Alexandre-Miguel, 2022-09-15:
In Inav 5.0 or 5.1 when using Target TMOTOR F7 V2 after flashing the drone gets UPSIDE Down, i go to the Setup i can put ALL the Options CW 180º, CW 270º or CW 270 Flip and the drone is always in the same position, doesn´t change. In Betaflight works perfectly!
Only way for now is to put the Board in Normal Position.....

Using 5.1.0 Configurator

JulioCesarMatias, 2022-09-15:
Explain things better... Questions:

What parameters are you changing? You quoted "CW 180º, CW 270º or CW 270 Flip", these options are part of the Magnetometer alignment, not the Gyro.

You described "Gyro problem" in the title, but I'm pretty sure it's a problem with the Accelerometer. Can you confirm this?

Do you know how to use the CLI?

If your answer to the last question is YES... The parameters to correct the Accelerometer alignment are: align_board_roll, align_board_pitch and align_board_yaw. The minimum and maximum values ​​are -1800 to +1800 (-180 degrees to +180 degrees).

Alexandre-Miguel, 2022-09-16:
Hello,

Yes the problem is with the Accelerometer, when i try to calibrate the drone is always UPSIDE down in the Setup and then when i go to calibration it is also upside down. 
So i go to Configuration Tab and then in the Board Sensor Alignment i can choose any value and save and reboot and the drone keeps the same, it does note change anything.... I do the same in Betaflight or Emuflight and works ok!
I solved by rotating the board and now it's ok but in the configurator it does not work, i will try in the CLI but only Monday, i'm out in work.

JulioCesarMatias, 2022-09-16:
in the first step of accelerometer calibration, if you calibrate with the IMU upside down first, I think your problem will be solved

Alexandre-Miguel, 2022-09-16:
I will try also and leave here feedback!

closed 2023-10-27 by DzikuVx