iNavFlight/inav #8402: Autolaunch is not working

dirkmaat, 2022-09-17:
I can fly with my plane in Manual mode horizon mode Gps mode. When i put Permanently autolaunch on. Then gives the system Not ok en cannot arm. This is my video https://www.youtube.com/watch?v=9ouc4EnMkog
What am i doing wrong?

breadoven, 2022-09-17:
Go into Configurator CLI tab and type "status". This will tell you why it isn't arming. Auto launch related Arming issue is likely due to the roll/pitch stick not being centred so check roll and pitch values are 1500 in the Configurator Receiver tab. Any other value will cause problems.

dirkmaat, 2022-09-17:
thank you. I have adjusted the value in my transmitter under outputs so that in inav it is 1500. Now I can turn on Autolaunch permanently and also put it on arm. Can test it well outside.

closed 2022-09-18 by breadoven