iNavFlight/inav #8408: Broken USB on MATEK 722 STD : any alternative way to connect by UART ? - Lot of confusion about

alccrl, 2022-09-20:
### Discussed in https://github.com/iNavFlight/inav/discussions/8400

<div type='discussions-op-text'>

<sup>Originally posted by **alccrl** September 17, 2022</sup>
Hi.

On Internet there is much confusion about solutions to have a second connection to PC , other then USB MSP/VCP.
Online Videos seem to be not updated with last Inav versions or fc hw, and most refer Betaflight, not Inav.

On Matek F722 STD FC I tried FTDI, direct USB2Serial, DFU on UART1, but no way to communicate between FC and inav-configurator.
Serial ports (usb2serial or ftdi) are correctly recognized by PC, ST programmer utility (set to UART) complains no connection when FC is booted in DFU mode. 

It is not even clear how to feed vcc to the system, 3.3v or 5v.
Direct usb2serial protocol and hw sufficies three wires (GND, TX, RX) if FC and PC are already powered, whilst FTDI may convert USB from PC to 5v or 3.3v and is put in the middle between pc and fc.
In the Mateksys faqs there is a note suggesting connections but their solution does not work, and it is quite an obsolete post. 

What has to be done in the inav configurator for enabling secondary connection to pc ?
Is it necessary to use a free uart (I have UART5 unused) or just solder tx1 rx1 (Uart1) port ?

Does anybody have any idea ?

Since DFU mode does not work on TX1/RX1, I guess that port 1 cannot be used for this purpose on this FC, but this is against declaration on mateksys site that assert  uart1 can be used for USB connection alternative.

So,
combinations are so many, either sw or hw, and I would like to keep my fc, not to change it because of a stupid usb dying connector ...!

My case:
fc : matek f722 std
fw : inav 5.1
FTDI FT232RL USB to TTL Serial Converter Adapter Module 5V and 3.3V
Serial2USB commercial cable with 9 pin connector
pc : laptop with both Win10 and Linux
USB cable with microUSB fc side
USB cable with miniUSB connector to be used on FTDI side.
FTDI board connected to UART1 on fc side and miniUSB connector pc side

Thanks in advance,

Alfredo</div>

stronnag, 2022-09-20:
Closing as there is a discussion with solution,

closed 2022-09-20 by stronnag

alccrl, 2022-09-20:
How to reproduce (inav 5.1 for matek 722 std):

1)solder tx1 gnd 5v rx1 or tx# gnd 5v rx# and connect to FTDI or direct serial2usb connector (no vcc in this case)
2)configure pc drivers for ftdi or usb2serial
3)start configurator
4)it says connection open -> connection succesfully closed (no connection)
5) Try also poweron fc in boot mode -> configurator does not connect either (for flashing)
6)also stProgramming in uart mode (as suggested in Matek faqs) )says "no connection"

A link to MatekSys FAQs (not working (obsolete ?) FC USB break off:
http://www.mateksys.com/?p=5712#tab-id-14

alccrl, 2022-09-20:
Dear Stronnag,
please write down or reproduce correct method WITH MATEK F722 STD ....before closing ...

alccrl, 2022-09-20:
As a rule of thumb, closure means to reproduce problem in same way it appears and demonstrate fix works.
Your configuration is not what I am describing ...

alccrl, 2022-09-20:
Thank anyway for Your support.

alccrl, 2022-09-20:
It takes time to answer ... but also to describe problems ... takes time ....!!!

stronnag, 2022-09-20:
There is no probem
* Flashing over serial works 
* MSP over serial UART works 

Issues are not a support channel.