iNavFlight/inav #8409: arm and disarm causes MATEKH743 to lose connection with radio in INAV 5.1.0

vlamgat007, 2022-09-20:
## Current Behavior
I am in the process of upgrading a Matek H743-WING V2 Flight Controller + XF Nano Diversity setup from INAV 3.0 to INAV 5.1 and am running into an issue.  When I arm and disarm the system, I lose all radio control to the board (in the Radio tab the bars stop moving).  I need to reboot the board to gain control.  It does not seem like any of the other switches cause the same behavior.  Firmware for the XF components (RX + VTX) have also been updated.

(FYI....I have successfully completed a similar upgrade for a different plane with a similar setup using Matek F405-WSE Controller + XF Nano and did not experience the same problem)

# Steps to Reproduce
1. Switch on Radio
2. Power up the plane (plug in battery)
3. Connect to board with PC via USB
4. Connect to board using INAV 5.1.0
5. Verify that Receiver is connected (all control surfaces moving and bars moving in receiver tab)
6. Wait for the system to be ready to arm
7. When ready to arm, flip the arm switch
8. Disarm
9. At this point the board stops responding to the radio (bars not moving in the Radio tab and control surfaces on the plane not moving)
10. Reboot the board to regain control

## Additional context: dump all
<https://pastebin.com/uVBzcK75>

---

# version
# INAV/MATEKH743 5.1.0 Aug 19 2022 / 12:34:02 (76f22b25)
# GCC-10.2.1 20201103 (release)

- FC Board name and vendor:  Matek H743-WING V2 Flight Controller
- INAV version string: 5.1.0

b14ckyy, 2022-09-21:
what crossfire FW version do you use? There are some older versions with odd behavior like this.

b14ckyy, 2022-09-21:
hmmm 6.19 should be stable. did you do a factory reset of your crossfire receiver and module after the last update? If you update from 6.14 or earlier you have to do a factory reset. Maybe try that and then see if the issue still appears. 

if yes, I suggest to reset the FC to defautls and just set up the basics and see if this solves the issue. Save a DIFF ALL before that.

vlamgat007, 2022-09-21:
I did do a factory reset of the TX  but not sure about the RX.  I will test this later today and post my results.
Thanks!

b14ckyy, 2022-09-21:
Rx you should reset first over the crossfire settings LUA. because you will lose binding and doing it last will be more difficult

vlamgat007, 2022-09-21:
Thanks, will do.

vlamgat007, 2022-09-22:
Ok, I think I have narrowed down the issue.  The RX and TX reset did not fix the issue.  As suggested I then reset the defaults on the board and started from scratch. I set up only one mode for arming on Channel 5.
Every time I made a config change and saved it then tested (are + disarm) and verified if I still had radio connection.  I continued till I hit the "**Continuously trim servos on Fixed Wing**" setting.  As soon as I saved this and went through the test cycle, it froze up, and when I removed the setting the issue disappeared.  I tested this multiple times to confirm.

b14ckyy, 2022-09-22:
this is really strange. can you please post a DIFF of your current config right at this point where enabling or disabling this setting causes the lockup?

breadoven, 2022-09-22:
Might have something to do with it ?

https://github.com/iNavFlight/inav/blob/69b18f267872b10ceb3327932348a5d0eb6c091d/src/main/fc/config.c#L330

breadoven, 2022-09-22:
It looks like it suspends the Rx signal for a fixed 1.5s period (`SKIP_RC_ON_SUSPEND_PERIOD` in `rx.c`) before resuming the Rx signal regardless of whether or not eeprom has been saved ... which doesn't make sense. I'm guessing this isn't long enough for 5.1 which takes longer to save eeprom. Perhaps 1.5s is right on the limit for some boards and not others, a timing issue. Although it would make more sense not to have a fixed time period but simply reset the suspend time when eeprom write was finished.

breadoven, 2022-09-22:
Having said that surely if it's writing to eeprom it shouldn't be able to do anything else should it, like try to process Rx signals ?

MrD-RC, 2022-09-22:
> 
> @vlamgat007 do you use a switch to arm and disarm or do you disarm with a stick command? 
> 

There is no stick command to disarm anymore. I think it was even gone in 3.0.

vlamgat007, 2022-09-22:
Diff all at point of failure:  https://pastebin.com/8YEMD5Bg

Please note the Failsafe triggered but I have had multiple times when there was no failsafe but I had no control.

vlamgat007, 2022-09-22:
> > @vlamgat007 do you use a switch to arm and disarm or do you disarm with a stick command?
> 
> There is no stick command to disarm anymore. I think it was even gone in 3.0.

I use a switch on channel 5 to arm and disarm.

vlamgat007, 2022-09-22:
diff all with no failsafe:  https://pastebin.com/EDA0WcY8

breadoven, 2022-09-22:
Failsafe is suspended during eeprom save so it doesn't trigger when the Rx is suspended. The fact that Failsafe behaves erratically during this problem might indicate there is an issue with Rx suspend during eeprom save. Be useful to know why the Rx is suspended during eeprom save, not immediately obvious.

Also wondering if this has anything to do with https://github.com/iNavFlight/inav/issues/7128. The only time I had a Config wipe was on a plane that was connected to Configurator and powered from the battery when it happened. The Rx only works on that plane when on battery power not USB.

vlamgat007, 2022-09-22:
I have tried with battery connected and without (usb only) and had the same result.

breadoven, 2022-09-23:
@vlamgat007 Can you try the attached firmware. It's the current master with Rx suspend time increased to 3 seconds. You'll need to use the latest 6.0 Configurator which can be found at http://seyrsnys.myzen.co.uk/inav-configurator-next/.

[inav_6.0.0_MATEKH743.hex.zip](https://github.com/iNavFlight/inav/files/9632264/inav_6.0.0_MATEKH743.hex.zip)

0crap, 2022-09-23:
@vlamgat007 If you want to stick with 5.1 you might also want to try my [5.1 build](https://github.com/0crap/inav/releases/tag/5.1.0) without auto ContinuousServoAutotrim saving.

vlamgat007, 2022-09-23:
Thank you @breadoven and @0crap!  I like having ContinuousServoAutotrim so I will try v6.0.0 first.
I am tied up with work so I will get to this later today.
Thanks!

0crap, 2022-09-23:
@vlamgat007 Try whatever you like, but please be assured that in my 5.1 build ContinuousServoAutotrim is fully functional.
Only it will not auto save (which seems to be your issue) the new values after you land and disarm. You can still save the new values if you wish, by going into the OSD and choose save and reboot.

vlamgat007, 2022-09-23:
@0crap, thanks for the clarification!

vlamgat007, 2022-09-23:
Success!  I loaded 6.0 and flashed with the 6.0.0 firmware provided by @breadoven.  I updated using the last Diff All that I posted and then tested the arm and disarm.

There is a bit of a longer delay before you get control back after flipping the disarm switch but that seems like a minor issue.

I completed the rest of the setup and everything seems to be working as expected.

BIG THANKS to everyone for your support on this!!

Let me know if you have any further special instructions.  I take it that I need to keep my Matek F405-WSE Controller on 5.1.0?

breadoven, 2022-09-24:
@vlamgat007  Just remember the 6.0 firmware is dev standard so could have issues, not that I've had any problems. Other FC boards are fine on 5.1.0 if they work OK.

So it appears the default Rx suspend time setting during eeprom write is probably causing problems but it's not clear why it only seems to happen with Auto Trim on disarm but not with other eeprom writes such as sensor detection and Accelerometer calibration on boot up. I assume nothing odd happened previously with 5.1.0 during accelerometer calibration on boot, completed as normal with the beeper confirmation at the end ?

b14ckyy, 2022-09-24:
@breadoven maybe because when disarming, the actual eeprom write is done first before the actual disarm happens. So all radio inputs are still expected to be valid during that time. All other eeprom saves happen only in already disarmed state.

vlamgat007, 2022-09-24:
I just came back from flying the plane after the changes.

Bad news is that it froze up when i disarmed after I landed. Good news is that this happened after it was safely on the ground.

breadoven, 2022-09-24:
> @breadoven maybe because when disarming, the actual eeprom write is done first before the actual disarm happens. So all radio inputs are still expected to be valid during that time. All other eeprom saves happen only in already disarmed state.

It waits until the Arming flag is disarmed before saving. Maybe the problem is Stats also get saved at the same time the Arming flag changes state. 2 eeprom writes in succession. Maybe this needs changing so you only write once on disarm.

breadoven, 2022-09-24:
Ah, missed the comment that the problem is still there. Probably worth trying @0crap's 5.1 version with the ContinuousServoAutotrim eeprom auto save removed to see if that fixes things.

vlamgat007, 2022-09-26:
I have loaded the firmware from @0crap.  It has not frozen up yet, unfortunately I ran out of time to do a test flight.

vlamgat007, 2022-09-29:
Flew and did not hang up on me.  Used the OSD to save.  Strange enough my other wing using the 405-wse controller did hang up after landing and disarming :-|

breadoven, 2022-09-30:
Would appear this eeprom write on disarm is somehow randomly causing problems.

@vlamgat007  Do you have `stats `set to ON ? Could try turning it OFF to see if that makes a difference. Test on the H743 with current 5.1.0 release that hangs every time you disarm. Stats writes to eeprom on disarm at the same time as the Continuous Autotrim but they're written separately. Hard to see how that could be an issue but maybe the timing causes problems.

Also noticed normal servo Autotrim (not continuous) does a `saveConfigAndNotify()` on disarm rather than just `writeEEPROM()` used for the Continuous Autotrim. Surely the same eeprom save method should be used for both.

MrD-RC, 2022-09-30:
@breadoven I agree. I think anything that saves on disarm should call saveConfigAndNotify(). Maybe that could just set a flag to do the actual saving and notifying in the next loop, so save is only done once.

b14ckyy, 2022-09-30:
While things get changes, I guess it might be a good idea to finally also disable autosave for Continous Servo trim. There is no point in saving it every time with deviations of 10us between flights. 
With the longer and longer save times I also see that as a safety issue when the disarm takes 1.5s  or even longer after the disarm switch is applied. This can make the difference between a flesh wound and losing a hand. Just keep it for the classic autotrim and stats as for stats there is no other way and Autotrim is  intentionally used on a single flight.

breadoven, 2022-09-30:
Saving happens after disarming so it doesn't increase disarm time. You lose the RC link during saving but the motors should have a stopped by then.

b14ckyy, 2022-09-30:
But the Motors are actually still running. At least on a wing. Not sure about copters as dshot wilkl reset if the signal interrupts briefly during safe.

vlamgat007, 2022-10-01:
Stats = OFF on both planes

breadoven, 2022-10-01:
> But the Motors are actually still running. At least on a wing. Not sure about copters as dshot wilkl reset if the signal interrupts briefly during safe.

You're right. I have a delay on disarming for another reason and don't use the servo trim stuff so wouldn't have noticed any additional disarm delays caused by eeprom saves (although I do have Stats enabled).

Seems even though the motors are set to the disarmed state in the mixer code it isn't actually written to the motors until after servoAutoTrim is processed during which the eeprom write will occur:
https://github.com/iNavFlight/inav/blob/c6706adc91a8f6ae0a37163188342240378b2d39/src/main/fc/fc_core.c#L927

This doesn't look like the best way of doing things. Certainly shouldn't have multiple eeprom writes on disarm and the order of events isn't ideal especially if it delays disarming.

breadoven, 2022-10-01:
> Stats = OFF on both planes

So only 1 eeprom write causes this problem which simplifies things. Only question is why and also why aren't others having the same problem.

Makes you wonder what happens if you turn Stats ON ?

vlamgat007, 2022-10-01:
> Makes you wonder what happens if you turn Stats ON ?

I will test.

MrD-RC, 2022-10-01:
@breadoven I've consolidated the save calls in to a single action. Being on the next loop, this should also allow other tasks to complete; such as disarming. Of course, that will depend on the scheduler too.

https://github.com/iNavFlight/inav/pull/8439

breadoven, 2022-10-06:
> > Makes you wonder what happens if you turn Stats ON ?
> 
> I will test.

Did you get a result from this test in the end ?

vlamgat007, 2022-10-06:
> Did you get a result from this test in the end ?

I am sorry I have not!  I will see if I can bench test tonight and fly over the weekend.

vlamgat007, 2022-10-07:
> > > Makes you wonder what happens if you turn Stats ON ?

@breadoven I set stats = ON and it is freezing up again.  This is with the firmware provided by @0crap on the matek h743-wing v2 board.

breadoven, 2022-10-07:
> > > > Makes you wonder what happens if you turn Stats ON ?
> 
> @breadoven I set stats = ON and it is freezing up again. This is with the firmware provided by @0crap on the matek h743-wing v2 board.

OK, well that clarifies things. Just need to work out what's wrong with the eeprom save now.

zvikaf, 2022-10-07:
Have the same with H743 Wing V3 ...  #8450

0crap, 2022-10-08:
> > > > Makes you wonder what happens if you turn Stats ON ?
> 
> @breadoven I set stats = ON and it is freezing up again. This is with the firmware provided by @0crap on the matek h743-wing v2 board.

Weird stuff.
I bench tested with the `stats = ON` setting on my build and no issues.
That said, it's not using the diff all from vlamgat007, just my own fully configured diff all setup.
After a `arm` and `disarm` the FC stays responsive.
Because autolaunch for fixed wing is always on, the arm command gives the "raise throttle" indication in the OSD.
At that point I disarm and do this over and over, no issues.

breadoven, 2022-10-09:
> > > > > Makes you wonder what happens if you turn Stats ON ?
> > 
> > 
> > @breadoven I set stats = ON and it is freezing up again. This is with the firmware provided by @0crap on the matek h743-wing v2 board.
> 
> Weird stuff. I bench tested with the `stats = ON` setting on my build and no issues. (MATEKH743 V2) That said, it's not using the diff all from vlamgat007, just my own fully configured diff all setup. After a `arm` and `disarm` the FC stays responsive. Because autolaunch for fixed wing is always on, the arm command gives the "raise throttle" indication in the OSD. At that point I disarm and do this over and over, no issues. (Stats screen shows up on OSD.)

@0crap Are you using the same receiver type, i.e. crossfire ?

0crap, 2022-10-09:
> @0crap Are you using the same receiver type, i.e. crossfire ?

I use a ExpressLRS RX (EP1 on v3.0). Which does use the CRSF protocol between RX and FC.

MrD-RC, 2022-10-09:
> 
> https://github.com/iNavFlight/inav/actions/runs/3169611952
> 
> 
> 
> Is this version OK to test with @MrD-RC ?

Yes, all good.

EhAye, 2022-10-14:
i'm having a similar problem. H743-WING v2, and tried two different ELRS RX

i can trigger the failsafe by moving yaw to the left or right and holding it there. (0 throttle, pitch&roll middle). This seems to trigger the failsafe a lot more reliably than arm/disarm with a switch... which only causes a failsafe sometimes.

After a failsafe, i tried to reboot the RX and TX (leaving the FC running) by using the elrs lua script. They bind and TX gets full telemetry including GPS telemetry, but failsafe RX flag remains & channels remain unresponsive.

stats=on or off doesn't seem to make any difference, though my vtx isn't powered

MrD-RC, 2022-10-14:
That sounds like an ELRS issue to me. It is the ELRS receiver that tells INAV it is in failsafe, not the other way around. INAV does not command failsafe, it only does what it is told.

breadoven, 2022-10-14:
> i can trigger the failsafe by moving yaw to the left or right and holding it there. (0 throttle, pitch&roll middle). This seems to trigger the failsafe a lot more reliably than arm/disarm with a switch... which only causes a failsafe sometimes.

INAV 5.1.0 ? And is this failsafe caused by the yaw stick movement when armed or disarmed or both ?

EhAye, 2022-10-14:
> INAV 5.1.0 ? And is this failsafe caused by the yaw stick movement when armed or disarmed or both ?

when i arm with a TX switch, the yaw movement does not cause failsafe. It only happens when ch5 switch is disarmed.
yes inav 5.1.0

b14ckyy, 2022-10-14:
The only stick command that has a centered pitch/roll and a full yaw right input is the Arming Safety bypass but that should not trigger anything else than that. 
isn't one of the yaw stick movements also used for HDZero or Runcam Cameras to enter the camera menu? Is this maybe a ELRS feature that is triggered and reporting a FS to prevent arming while in the menu?

breadoven, 2022-10-14:
> > INAV 5.1.0 ? And is this failsafe caused by the yaw stick movement when armed or disarmed or both ?
> 
> when i arm with a TX switch, the yaw movement does not cause failsafe. It only happens when ch5 switch is disarmed. yes inav 5.1.0

I'm still not sure what you're saying here. It sounded like a failsafe can be triggered just by moving the yaw stick left or right without touching the arm switch. Are you in fact saying that the failsafe is only triggered when using the arming switch to disarm but the failsafe is more likely to occur with the yaw stick held hi or low when disarming ?

EhAye, 2022-10-14:
> isn't one of the yaw stick movements also used for HDZero or Runcam Cameras to enter the camera menu? Is this maybe a ELRS feature that is triggered and reporting a FS to prevent arming while in the menu?

It also failsafes on Throttle up, Yaw left, Pitch up Roll Right (both up and to the far corners) I  can't see any commands listed in the documents associated with this position, but it causes a failsafe as well.

The Calibrate Compass stick command causes a failsafe as well (Throttle up, Yaw Right, Pitch down).... most positions don't cause it. But these do.

EhAye, 2022-10-14:
> I'm still not sure what you're saying here.

sorry. I get failsafes when arming with channel5 switch sometimes.

But while disarmed, if i move the yaw stick to the left and hold it there for about 2 seconds, i get a failsafe. There's a few other stick commands that will cause failsafe as well, like the Calibrate Compass command

breadoven, 2022-10-14:
> That sounds like an ELRS issue to me. It is the ELRS receiver that tells INAV it is in failsafe, not the other way around. INAV does not command failsafe, it only does what it is told.

If it's related to https://github.com/iNavFlight/inav/commit/3a13edfdad83fb3eaeada195654fbadbe5acb495 then it could be an INAV problem although it's odd it only seems to affect these H743 boards.

EhAye, 2022-10-14:
> > That sounds like an ELRS issue to me. It is the ELRS receiver that tells INAV it is in failsafe, not the other way around. INAV does not command failsafe, it only does what it is told.
> 
> If it's related to [3a13edf](https://github.com/iNavFlight/inav/commit/3a13edfdad83fb3eaeada195654fbadbe5acb495) then it could be an INAV problem although it's odd it only seems to affect these H743 boards.

While its in the RX failsafe state, i put both the RX and TX into wifi mode and hit the "Save & Reboot" button, to reboot them while keeping the FC powered

They re-establish link and resume advanced telemetry normally. But the FC still reports RX failsafe and shows no channel movements on the flaps or receiver tab 

Could inav just be getting 'stuck' in this state, and refusing to reset the flag?

breadoven, 2022-10-14:
> > I'm still not sure what you're saying here.
> 
> sorry. I get failsafes when DISARMING* with channel5 switch sometimes.
> 
> But while disarmed, if i move the yaw stick to the left and hold it there for about 2 seconds, i get a failsafe every time. There's a few other stick commands that will cause failsafe as well, like the Calibrate Compass command

I assume it doesn't failsafe when moving the yaw stick if it's never been armed ?

EhAye, 2022-10-14:
> I assume it doesn't failsafe when moving the yaw stick if it's never been armed ?

freshly rebooting everything, i can simply move the yaw to the left, hold for 2 seconds, and it failsafes. Not touching anything else

[70 more replies]

PR #8948 (2023-04-05): Revert "Switch H7 to -Ofast optimizations"
Probably 
fixes #8409
fixes #8905 

Could resolve #8899 for H743WLITE
closed 2023-04-21 by DzikuVx