iNavFlight/inav #8411: Unable to drop throttle below set nav_fw_launch_idle_thr = 1500 after successful autolaunch

preisedj, 2022-09-21:
# version
# INAV/OMNIBUSF4V3 5.1.0 Aug 19 2022 / 12:45:07 (76f22b25)
# GCC-10.2.1 20201103 (release)

I have set nav_fw_launch_idle_thr = 1500 to spin up the motors before launch.  The launch behavior is as expected but after launch, the throttle will not go below the set 1500.   I launch in Angle Mode and have the motor stop on low throttle set to ON.  Settings are default in the Output tab.
This behavior is repeatable. 
The only way I was able to land my craft (i.e. reduce/cut throttle) was to disarm on my approach and glide in. 
 
[INAV_5.1.0_cli_Dart250_20220921_122617.txt](https://github.com/iNavFlight/inav/files/9618535/INAV_5.1.0_cli_Dart250_20220921_122617.txt)

b14ckyy, 2022-09-21:
Wasn't it you who posted the same question on FB? 

as I already assumed in the comments, and I was right, you have enabled althold permanently when armed. Why? 
If althold is activated together with ANGLE or HORIZON mode, INAV takes over throttle control. Althold always has automatic throttle control. 

Disable althold in your modes and the issue is solved
[image]

preisedj, 2022-09-21:
No Nav Alt was NOT activated.  But I figured out what caused it.
UPDATE: So I figured out what is going on.  Instead of enabling Autolaunch permanently in the Configuration tab, I set it in the Mode tab to be ON when the arm switch is turned ON.  Still, I would think that is an unexpected behavior. Any stick movement should stop the autolaunch sequence, including min throttle.  But in any case, the problem is solved by setting it in the Configuration tab.

closed 2022-09-21 by preisedj