iNavFlight/inav #8413: air mode inav 5.1 not working

davifelix, 2022-09-22:
good afternoon.
i installed inav 5.1 on holybro kakute f4v2 AIO card. unfortunately I can't get the air mode to work. from the "mode" tab it is enabled but then in reality the propellers do not turn with zero throttle. how can i solve?

Voltharion, 2022-09-22:
I don't know if it's related but I'm also having a problem with a mode. I can't enable NAV CRUISE, it keeps saying it's in ACRO whenever I switch to that mode. Manual & horizon are no problem.
Didn't have the issue on 4.1

Jetrell, 2022-09-22:
The `airmode_throttle_threshold` value has to be exceed before Airmode will be enabled and stay active. 
And `throttle_idle` also requires a motor idle value.

davifelix, 2022-09-23:
I honestly don't remember having the same problem on previous versions. Cruise I don't remember trying it..... I hope to get help because without airmode I feel lost.

breadoven, 2022-09-23:
MOTOR_STOP  enabled ?

davifelix, 2022-09-23:
then...... after several tests ..... in no case does it make me activate airmode with a switch, or rather, in the "modes" section loswitch activates the function but in reality nothing happens. if I deactivate "motor stop" the engines turn but do not control the vehicle, they run empty on gas at zero and that's it. the only thing I can do is activate motor stop and pthen...... after several tests ..... in no case does it make me activate airmode with a switch, or rather, in the "modes" section loswitch activates the function but in reality nothing happens. if I deactivate "motor stop" the engines turn but do not control the vehicle, they run empty on gas at zero and that's it. the only thing I can do is activate motor stop and "permanently airmode". I don't like this thing because I'm not used to it but it seems to be the only solution. anyway...... I do not activate airmode with a switch so there must be some problem.

davifelix, 2022-09-23:
sorry I translated very badly .......

davifelix, 2022-09-23:
then ...... after various tests ..... in no case does it make me activate airmode with a switch, or rather, in the "mode" section the switch activates the function but in reality nothing happens.  if I deactivate "stop engine" the engines run but do not control the vehicle, they idle at zero gas and that's it.  the only thing I can do is activate the engine stop and pthen ...... after various tests ..... in no case does it make me activate airmode with a switch, or rather, in the "mode" section the switch is activated the function but nothing actually happens.  if I deactivate "stop engine" the engines run but do not control the vehicle, they idle at zero gas and that's it.  the only thing I can do is activate the engine stop and "permanent air mode".  I don't like this because I'm not used to it but it seems to be the only solution.  anyway ...... I don't activate airmode with a switch so there must be some problem.

Jetrell, 2022-09-24:
@davifelix Are you aware, that if you don't use `Permanently enable airmode`... Then you need to activate `Airmode` in the modes tab, according to the mode switch position you require it to be activate on. Or in conjuration with other flight mode.

[image]

davifelix, 2022-09-24:
I know exactly this thing. but it doesn't work at all. the tab changes
color so from the radio comes an input but the engines remain stationary
and does not control with the gas closed.

Jetrell, 2022-09-24:
> I know exactly this thing. but it doesn't work at all. the tab changes color so from the radio comes an input but the engines remain stationary and does not control with the gas closed. 

Your saying that, Air mode activates in the Configurator mode tab, when you switch to that selected slider position? But it does not keep the motors spinning when the throttle stick is pulled low, once the `airmode_throttle_threshold` value is exceeded the first time?

Do you have `airmode_type = THROTTLE_THRESHOLD` .. You don't have it set to `airmode_type = STICK_CENTER`?

What is your `throttle_idle` set too?

What is your `airmode_throttle_threshold` value set to?  It should ideally be set just below the quads lift off throttle. e.g if your quad hovers at 1300uS, then set `airmode_throttle_threshold` to 1270uS. This is the throttle value Airmode will activate, and stay latched ON, until disarming... If its set too high, you may not be pushing the throttle high enough to activate it.

It must be something in your settings.. I tried it on a quad with 5.1. And it works as expected.

This is the order of things.
1) Set your mode switch to Airmode.
2) With the throttle low, Arm the quad (the motors will not spin)
3) Increase your throttle until it lifts off. And until your `airmode_throttle_threshold` value is exceeded.
4) Land again. Then lower the throttle.. The motors will now keep spinning until you disarm, because airmode is now active.

MrD-RC, 2022-09-24:
> I know exactly this thing. but it doesn't work at all. the tab changes
> color so from the radio comes an input but the engines remain stationary and does not control with the gas closed.
> 
>

Just to clarify how the modes tab in configurator works. 

If the range is lit up light blue. That mode _is_ working. The flight controller is telling configurator that it is in that mode now. 

If the range is lit up dark blue. That mode _is not_ working. But will work when the flight controller is happy that the requirements for that mode are correct.

Jetrell is much better at all this quad stuff than I am. But I do know that in INAV, it is not the same as BetaFlight. For example, you can't just arm, enable airmode, then use the pitch sticks to tilt the quad. For INAV, when AIRMODE is enabled, a threshold has to be passed before it becomes active. I believe on 5.x, that is around 15% throttle. 

So with motor stop off (correct for a quad), when you arm and enable AIRMODE the motors should spin, but your pitch, roll, and yaw inputs won't do anything. If you raise the throttle to above 15%, then lower it, the pitch, roll, and yaw sticks will have an effect.

davifelix, 2022-09-24:
gee..... accustomed to betaflight I did not know all these things. as soon
as I have a minute I do a test and let you know. thank you very much in the
meantime!

davifelix, 2022-09-24:
boys! you are right about everything! practically I had done everything
right and there was no problem. only that I expected to see the propellers
turnyou are right about everything! practically I had done everything right
and there was no problem. only that I expected to see the propellers turn!
🤦‍♂️🤦‍♂️🤦‍♂️🤦‍♂️ grazie mille per il vostro prezioso aiuto. non ne
sarei uscito da solo!

closed 2022-09-27 by breadoven