iNavFlight/inav #8414: VCC Pad and DJI Plug does not work in inav 5.1.0 with the Mamba F722 mini MK4

Ch0ndr0t3c, 2022-09-23:
## Current Behavior

The VCC / 9V pad and the plug for the DJI air unit  on my Mamba F722 Mini MK4  does not have any voltage . It shows 0 volts.
On Betaflight the pad and the plug works fine without any setup.
I flashed the target : MAMBAF722 2022B. ( Betaflight and Inav )  A sticker on my FC says , that the MAMBA F722 2022B is the correct target.

[image]

[image]

## Steps to Reproduce

1. Flashed the FC with iNav 5.1.0 -> No Voltage on VCC Pad and DJI Air Unit Plug
2. Flashed the FC with Betaflight 4.3.1 -> VCC pad and DJI air unit plug gets battery voltage / 9 V without any problem.
3. Reflashed it with iNav 5.1 -> again no voltage on VCC pad and the DJI air unit plug

## Expected behavior

The VCC pad and the DJI plug should receive voltage from the battery . VCC or 9 Volts. it depends on the jumper pad.

## Suggested solution(s)
Does anyone knows, who i can get voltage to the vcc pad and the DJI air unit plug ?

INAV/MAMBAF722_2022B 5.1.0 Aug 26 2022 / 10:02:27 (b63e68a6)
GCC-10.2.1 20201103 (release)

JulioCesarMatias, 2022-09-23:
A user said the same when support was added... See comment >> https://github.com/iNavFlight/inav/pull/8368

Ch0ndr0t3c, 2022-09-23:
> A user said the same when support was added... See comment >> #8368

Thanks

JulioCesarMatias, 2022-09-23:
did you enable the "Pin IO Box" in the settings?

JulioCesarMatias, 2022-09-23:
In the "Modes" tab, go to the bottom of the page where the "USER1" and "USER2" boxes are.

USER1 =  VTX power switcher
USER2 = WiFi Switcher

Ch0ndr0t3c, 2022-09-23:
> did you enable the "Pin IO Box" in the settings?

Sorry for this question. This is my first build with iNav. Where can i find this Pin IO Box in the settings ? Just over the CLI , i guess ?

Ch0ndr0t3c, 2022-09-23:
> In the "Modes" tab, go to the bottom of the page where the "USER1" and "USER2" boxes are.
> 
> USER1 = VTX power switcher USER2 = WiFi Switcher

Ok thanks. I wil have a look, tomorow.

Ch0ndr0t3c, 2022-09-23:
> In the "Modes" tab, go to the bottom of the page where the "USER1" and "USER2" boxes are.
> 
> USER1 = VTX power switcher USER2 = WiFi Switcher

That‘s it.  I added a switch in my TX16 s settings and activated the User1 in inav 

Now i can switch the vtx on and off .

Thank you for your help.

closed 2022-09-23 by Ch0ndr0t3c