iNavFlight/inav #8415: iNav 5.1 - CRUISE mode activating

Voltharion, 2022-09-24:
## Current Behavior
When enabling NAV CRUISE, wing stays in ACRO mode. This didn't happen on 4.1.

## Steps to Reproduce
Enable NAV CRUISE

## Expected behavior
Wing should switch to NAV CRUISE mode - OSD[system message] should display "CRUISE".

## Additional context
CLI dump: https://ghostbin.me/632cae9a43d25

# status (when switch to CRUISE)
System Uptime: 132 seconds
Current Time: 2041-06-28T01:04:00.000+00:00
Voltage: 14.90V (4S battery - OK)
CPU Clock=168MHz, GYRO=MPU6000, ACC=MPU6000, BARO=SPL06
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK, IMU2=NONE
SD card: Startup failed
Stack size: 6144, Stack address: 0x10010000, Heap available: 1728
I2C Errors: 0, config size: 9696, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = none, used = none
System load: 8, cycle time: 1005, PID rate: 995, RX rate: 151, System rate: 9
Arming disabled flags: CLI
VTX: not detected

- FC Board name and vendor: Matek F405-WSE
- INAV version string: INAV/MATEKF405SE 5.1.0 Aug 19 2022 / 12:05:31 (76f22b25)

MrD-RC, 2022-09-24:
Cruise does not activate on the bench, it never has done. 

If the range for cruise is dark blue on the modes page, when the switch is in the cruise position, it will be fine in flight.

Voltharion, 2022-09-24:
Ah okay thanks, the reason I tested it on the bench if because it didn't activate mid-air AFTER having altered some other settings too which I reverted before when testing it on the ground.

My bad, will check again today.

Voltharion, 2022-09-25:
It was indeed my own mistake. Thanks for the info!

closed 2022-09-25 by Voltharion