iNavFlight/inav #8417: Looking for help to bring some nice features from our fork to the main INAV

ricardodeazambuja, 2022-09-25:
Hi. I am one of the developers of [the CogniFly](https://github.com/thecognifly) (an open-source collision resilient quadcopter). We have made some changes to [our INAV fork](https://github.com/thecognifly/inav/tree/CogniFly) that allow us to use a flight controller without barometer and still enable the autonomous flight modes when you have a rangefinder and optical flow. This same branch supports [a Python library](https://github.com/thecognifly/cognifly-python) that makes super easy to control your drone autonomously, from an external computer or PS4 joystick. Additionally, we have [another branch in our INAV fork that allows you to use a motion capture system](https://github.com/thecognifly/inav/tree/Optitrack) as the global position, again, enabling all the autonomous flight modes.
So, I am here because we have been planning to make a pull request long ago, but our lab is always without enough interns or students with time or firmware expertise to do that. I truly think our changes would be very useful for the INAV community, therefore I am submitting this issue hoping to find volunteers to help us bring those changes to the main INAV.

Any comments / suggestions are welcome :)

0crap, 2022-10-02:
The only way this will work is if you make a Pull Request.
At that point the devs can have a look.

stronnag, 2022-10-02:
Firstly, very interesting research project.

INAV is not a over-resourced project, we certainly do not have a team of developers looking for interesting 3rd party developments to pop up.

It is also the case that some of the features you've developed would be of interest to some INAV users; they address some frequently asked questions in our Discord; you may find more volunteers via Discord that in the Github which is read by fewer people.

As has been noted, your team would have to make pull requests to the INAV project. Preferably, small, discrete PRs that can be reviewed in isolation. Given the age of your repository (pre INAV 2.6?), just rebasing may be non-trivial.

ricardodeazambuja, 2022-10-04:
Thanks for your replies, I will have a look at your discord server, but I will leave this open so someone else using github may see it and reply :)

EceChaik, 2023-05-29:
Hi,
I assume your 'Optitrack' branch holds the development you've done to enable external positioning data?
Do you have a Readme file or something similar, outlining how you feed the data to the inav flight controller?
Thank you very much

ricardodeazambuja, 2023-05-29:
@EceChaik, we moved from the that branch because the newest versions of inav now allow use to more easily fake GPS, Baro, etc. using MSP. Check the examples on this branch: https://github.com/thecognifly/YAMSPy/tree/yb/stable

EceChaik, 2023-05-30:
Ah yes I suppose the fake ones can be used for more than SITL. I see that you built your package in python. In our case we'll probably use a wifi mcu so we'll probably need to do the same stuff in C, but hopefully it will work out

ricardodeazambuja, 2023-05-30:
@EceChaik, we are not using inav fake sensors like the SITL stuff (I haven't tried, maybe it would work as well...), we are literally faking the sensors so the flight controller receives the messages as if they were real sensors connected using MSP. Let's say you have a sensor that gives you a global pose (e.g. optitrack) then you convert this global pose into a few MSP messages like GPS, BARO (or RANGEFINDER if you use SURFACE mode) and COMPASS, and send these messages to the flight controller (you probably need to adjust the gains for your situation). Usually we do this using only one UART, but I think it could be better to split into two if sending a sensor message that is high frequency like RANGEFINDER.

EceChaik, 2023-06-22:
@ricardodeazambuja Hi! We're working on this implementation, we've succeeded with sending the vicon position as a gps_msp message so far. Do you have any examples with sending the mocap orientation as a compass_msp message? At first we thought we would group it in the gps message but apparently for multicopters inav is setup to get heading only from compass

ricardodeazambuja, 2023-06-25:
@EceChaik you can check this [example script](https://github.com/thecognifly/YAMSPy/blob/yb/stable/Examples/test_msp2_sensor_compass.py)
And remember the firmware is expecting a real compass (magnetometer) and that needs calibration. It won't work without calibration, but since you are faking the compass you can set the configuration related to the compass calibration directly.  More details about the settings can be found [here](https://github.com/iNavFlight/inav/blob/master/docs/Settings.md). Below is an example of what we were using:
```
set align_mag = CW0
set mag_hardware = MSP
set mag_declination = 0
set magzero_x = 0
set magzero_y = 0
set magzero_z = 0
set maggain_x = 32767
set maggain_y = 32767
set maggain_z = 0
set mag_calibration_time = 30
set align_mag_roll = 0
set align_mag_pitch = 0
set align_mag_yaw = 0
```

I'm closing this issue because it's better to just use the current firmware for this and my initial message doesn't make sense anymore.

closed 2023-06-25 by ricardodeazambuja