iNavFlight/inav #8424: INAV 5.1 hangs, latch-ups on H743-WING V3

rashied, 2022-09-26:
## Current Behavior
Recently purchase an H743-WING V3, flashed it to 5.1, configured it, and then soldered on components:
- RX: Matek ELRS (and then Betaflight ELRS to verify it was not Rx issue) Rx to UART 6
- GPS: Matek M8Q-5883 to UART2 (GPS), DA1/CA1 (compass)
- Connected servos
(Connected ESC/motor with clips to test motor direction, not soldered)

Primary bug: when powered on, FC will hang (latency) in control or latch (no Rx inputs register). This behavior occurs whether powered by battery or by USB-C. When powered by USB-C and connected to INAV, latch happens within ~1-10 seconds and is observable in Receiver window in Configurator. When connected to battery and not INAV, it behaves normally for ~1 minute before hangs occur and then ultimately a latch. 

## Steps to Reproduce
1. Power on controller, FC such that Tx/Rx connect
2. If USB connected (with or without battery), connect to INAV
3. Go to Receiver tab
4. Move controls and observe change in input on Receiver tab
5. Wait for latchup (~1-10s)

## Expected behavior
No hangs or latchups.

## Suggested solution(s)
I'm unclear if this is a code issue as other H743-WING V3 work fine (MrD's video, [response in Discord](https://discord.com/channels/791437907478577172/881440703811559435/1023496657981149254)). 

## Additional context
- Video, note HW and MSP latencies: https://youtu.be/5KB2OGFaBRw
- Flashed Ardupilot (I have no idea how to configure it), but I moved the sticks for ~5-10m without any hang or latch-ups. INAV Configurator doesn't want to reflash back to 5.1. Will post dump when I figure that out.
- Several other people in Discord commented they were having similar problems on H743. Also, [this bug report](https://github.com/iNavFlight/inav/issues/8409) may be related??  

- FC Board name and vendor: H743-WING V3
- INAV version string: latest target for H743-WING V3 (9/24/2022)

b14ckyy, 2022-09-30:
Do you have installed an SD card for blackbox? If yes please remove it and see if it still happens.

rashied, 2022-10-01:
No SD card in any of these! 

Matek suggested it could be me triggering the stick commands. I did another set of three test where I deflect sticks in a way that shouldn't trigger any commands. It failed all of them, but didn't record the first. In the second one below, you'll see that it latches just after powering on & connecting after a few moments.

-     https://youtu.be/-XuWpO1wDbQ
-     https://youtu.be/hx0zX2nXvdc

Left on, the board keeps beeping at random(?) intervals.

b14ckyy, 2022-10-01:
OK this is not INAV. This is your ELRS losing link. 
INAV even switches into Failsafe mode. check your OSD with link quality and see if that goes down before the dropout happens. 
What ELRS module do you use? ELRS has ridiculous long duty cycles on older versions and even on newer versions if set up too aggressively (One reason why ELRS is not good for other people to fly with). 
Check if your module overheats.

rashied, 2022-10-01:
I've tried two ELRS Rx, [Matek](ELRS-R24-D) & [Betaflight ](https://betafpv.com/products/elrs-nano-receiver) on the most recent stable version (as of a few weeks ago) of ELRS. Both were working fine on F411-WTE and Foxeer Reaper AOI, respectively.  Also, I didn't have any Rx loss issues on Ardupilot + H743 Wing V3. I tested that for a solid 10 min. 

I haven't even hooked up my VTX - will do it and see what the OSD says and post a video later.

Another thing I haven't tried is using another UART. Any reason that would be different?

b14ckyy, 2022-10-01:
I see that with all the cross checks it does not look like an ELRS issue at first glance but from the behavior, first sensor loss, then RC control loss and failsafe indicated, it is at least no lockup of INAV itself. 

Yes another UART could be an option as well. Some UARTs have differences internally (I am not that deep into this to tell you what exactly) and can cause issues with certain devices. So worth a try. 

We have seen other strange behavior of ELRS in the past as well. Like after a failsafe, the signal came back, Pilot had control for a few minutes and then out of nowhere all inputs froze. No failsafe was triggered as the log show, just all channels froze in place. We have seen this 3 times now in the INAV Fixed Wing Group with exact the same pattern, but it was all 2.x firmware as far as I can remember. 

If the UART does not solve the issue I suggest to backup your config with DIFF ALL, then wipe the board with another flash and set up your RC link and telemetry only. Then see if it still happens. If not then piece by piece load back your config and see when it happens.

EhAye, 2022-10-06:
H743-wing V2 (inav5), and Matek R24-D (elrs 3.0)

I have the same problem. flaps will randomly freeze up to a full second before resuming. After a while, all channels stop responding entirely.

EdgeTX widgets report RSSI and LQ are functioning perfectly fine. inav configurator shows no movement on the receiver tab.

Also noticed telemetry slowly drops out. The first 9 telemetry items update rapidly and are always there. But GPS telemetry stops being received, and eventually yaw/roll/battery telemetry drops out too.

zvikaf, 2022-10-06:
having similar issue with TBS system, H743-wing +INAV 5.1 + TBS micro,
Connecting battery and USB to the computer, arming the system, when disarming the radio link stops, going into FS, rebooting the transmitter does not regain control.
This radio halt, repeats on doing the same procedure.  
Doing the same without the USB does not halt the radio ....
The same FW was functioning well on INAV 4.1 and F722

b14ckyy, 2022-10-06:
@zvikaf this is not similar, this is a completely different symptom. There where crossfire versions doing crap like that. Check your version and Update to 6.17 or 6.19

zvikaf, 2022-10-06:
@b14ckyy  thanks for the quick and elaborated response.  the TBS is updated to the latest version (6.19) , suspect that if it was CRSF issue, it would have surfaced on arming/disarming without USB connection as well. 
This is not random issue as this is repetitive.

b14ckyy, 2022-10-06:
@zvikaf I suggest that you open a separate issue for that though. We can collect information there until a dev can look at it. Also post a full DIFF and all the relevant info. If there is an issue, it is likely different from this ticket here.

zvikaf, 2022-10-06:
@b14ckyy  Thanks again.. will do :-)

OptimusTi, 2022-10-06:
I have a V2 board that I can check with.

zvikaf, 2022-10-06:
@OptimusTi  thanks, this will be great, wonder if it is consistent.

b14ckyy, 2022-10-07:
@zvikaf Sorry I had a brain fart yesterday and did not think of that one here: https://github.com/iNavFlight/inav/issues/8409
I think this is your actual issue. Devs are working on it.

rashied, 2022-10-08:
Hi folks, no video and still using 4.1.0 for now. I moved UARTs around and I still have hangs but no latching. RSSI never drops below 99 on VTX.

b14ckyy, 2022-10-08:
@rashied I see no hardware fault on the FC here. AS said before INAV is not freezing. It's the RC link that starts to become unstable. Cycle time, CPU load, MSP load, all stable and still updating. And INAV clearly shows a failsafe signal coming from the Receiver to the FC. 

Can't tell why this only happens on the H743-Wing for you both but maybe open a issue ticket in the ELRS Github. 
Maybe downgrade to 2.5.1 or older and see if it happens there as well.

rashied, 2022-10-09:
Yeah, I don't think it's the HW given that it was functioning on Ardupilot but I don't understand how it's the RC link. I'll just get a new FC when I have more time for FPV. Unfortunately, a lot seems out of stock and I'm looking for more UARTSs than the Matek F411-WTE. 

Just so I understand what you mean about the RC link, can you explain a little more? I'll add an issue on the ELRS git if I can get on the same page as you.

What do you mean that RC link is becoming unstable? I don't have any drop in RSSI in the OSD and the ELRS Rx LED stays lit solid, so Tx/Rx seem to maintain pairing. 

"Cycle time, CPU load, MSP load, all stable and still updating."

I don't know if this was the case, but I'd have to rewatch the videos.

"And INAV clearly shows a failsafe signal coming from the Receiver to the FC." 

I don't know what you're referring to. Is this the beep?

Thanks again, @b14ckyy , I really appreciate it.

b14ckyy, 2022-10-10:
"And INAV clearly shows a failsafe signal coming from the Receiver to the FC."

in the configurator below the battery Icon at the top, you have a small parachute symbol. This lights up red. 
This means that INAV either got a RX-Loss Data package reported by the receiver or INAV does not get any receiver signals at all. So the communitcation between receiver and flight controller is breaking down. But INAV is still working fine on its end as the PID loop and MSP communication is still active. 

Someone would need to debug the Serial communication on the INAV side to tell you exactly what's going on.

gcmcnutt, 2022-10-12:
@b14ckyy I have a similar issue with a new H743-Wing V3
- new with 5.1 FW installed
- default config
- nothing connected
- inserted several SD Cards to see if they would recognize
- in **all** cases, the controller doesn't fully start

What would be the best way to debug this?  I've got STLink and Jtag debuggers.  Not sure how to best hook it to this Matek -- do you have a reference?

MrD-RC, 2022-10-12:
@gcmcnutt that is a different issue. It is because of the SD Card. There is an issue on SD Card lockup already.

gcmcnutt, 2022-10-13:
@MrD-RC ahh, ok, any suggestions on how I can debug this?  Or is there an issue filed already?

OptimusTi, 2022-10-14:
> @OptimusTi  thanks, this will be great, wonder if it is consistent.

Sorry I haven't been able to look into this too busy

zvikaf, 2022-10-15:
Looks like the H743 is problematic, might be some bugs in the kernel :-(

EhAye, 2022-10-18:
> might be some bugs in the kernel :-(

Is that fixable? or is that a phonecall to Matek with hopes they send me something else?

gcmcnutt, 2022-10-19:
I'll give it an attempt to root cause this -- I could use some pointers on dev/debugger hookups for this device...

zvikaf, 2022-10-19:
WOW ... so indeed the kernel is problematic ... and the solution need to be done by ST :-(

EhAye, 2022-10-19:
> so indeed the kernel is problematic

I just saw this on Ardupilot's problems page... it has an H7 section:
https://ardupilot.org/copter/docs/common-when-problems-arise.html#common-when-problems-arise

> AutoPilots utilizing the H7 series of processors can, on rare occasions, get into a state where they will no longer complete initialization
...
> It is believed that this may be a memory corruption problem which can be caused by interrupting a flash memory write (as when changing parameters). Unfortunately, due to the processor’s architecture, there is no way in the firmware to correct this automatically

b14ckyy, 2022-10-19:
> I just saw this on Ardupilot's problems page... it has an H7 section: https://ardupilot.org/copter/docs/common-when-problems-arise.html#common-when-problems-arise

This is a completely different topic. INAV does not write single parameters like Ardu. This was never a problem on INAV. 

> WOW ... so indeed the kernel is problematic ... and the solution need to be done by ST :-(

Why do you think that? as @gcmcnutt said this looks promising by just lowering the SIDO clock. 

I think you guys mix up very different issues right now that have nothing to do with the OP.

MartinHugh, 2022-10-19:
In case it is relevant to this bug report :`

I experienced a failsafe on two occasions yesterday as soon as I disarmed.

H743 + iNav5.1 + ELRS 3.0 (EP1)
Continuously trim servos : Disabled.
stats = OFF

gcmcnutt, 2022-10-20:
Yeah, I too have seen  lockup that even for the saved config.  On more than one occasion I have had a lockup after a disarm and 'save updated pids' action...

Anyway, I agree, there seem to be a couple of faults being conflated here.  For now I will look at the blackbox issue, e.g. sdcard i/o reliability first -- and guessing this is something about the sdcard bus or transfer speed (I don't know enough about this code to understand the differences). If there is a better issue to connect this work to let me know.

Actions:
- trying to find the inav equivalent code for the 25mhz -> 12.5 mhz difference (maybe a cap on the sd transfer clock)
- i am able to upload locally compiled firmware so that is a good first step
- also trying to figure out how/which debugger to use.

(sorry ramping up here)

gcmcnutt, 2022-10-21:
Another data point:

- I measured the clock and data signals at the sdcard during init -- and yes a "HC I" card was showing > 25mhz. (if the system is following the spec it'd attempt to set clock to 50mhz)
- I found a very old card (a model "HC") -- and the system boots up just fine -- and card is recognized.

So, maybe some evidence we need to limit the clock as was done on the other flight controller s/w.

MartinHugh, 2022-10-21:
> Another data point:
>
> * I measured the clock and data signals at the sdcard during init -- and yes a "HC I" card was showing > 25mhz. (if the system is following the spec it'd attempt to set clock to [...]

Good data, but this was not a factor in my failsafes (see about 4 posts up) as I have no SD card fitted.  Unless not having an SD card fitted itself causes issues.

gcmcnutt, 2022-10-22:
@MrD-RC thanks -- i'll copy over the updated information.

Ktr128, 2022-11-11:
I have the same problem.
Matek H743-WING V3, INAV 5.1.
The elevon servos hung during autolaunch and the flying wing crashed. Although before that it took off 3 times perfectly.
After that, I caught the elevons hovering while testing in my hand.
Turned on the ANGLE mode and moved the flying wing in pitch.
After about 30 seconds, the elevons hung for 1-2 seconds, after which they started working again.
Has anyone solved this problem?

b14ckyy, 2022-11-11:
@Ktr128 please read the whole post and not only the title. What you describe is different from the issue reported here as it only happens when Disarmed in this case. 

To me this sounds like a Regulator brownout you had. Please open a post in the discussions sections as you might just need support on your setup. Don't forget to add any details possible about your setup. What plane, servos, FC etc etc.

Ktr128, 2022-11-11:
@b14ckyy   thank you for the quick reply.
I have read all messages #8424.
I didn't find in the @rashied  description what happens after disarm. Maybe I'm stupid )))
In particular, the situation is the same for @EhAye 
https://github.com/iNavFlight/inav/issues/8424#issuecomment-1269662600

Did you mean #8409 ?

If you recommend creating a new question with a detailed description, I will do it.
Thanks in advance!

maxgunn19, 2022-12-29:
I am having the same problem. 

I can start up but I cannot arm so my Motors won't spin (they spin in the outputs tab though). I can move the servos around and around for about 15 seconds without problems and then everything stops. A few seconds later the failsafe will be triggered and according to CLI I lose RX. Also in CLI my battery is reported as taking up 196% of my maxload and I cannot seem to resolve that. Any ideas would be helpful. 

Matek H743 Wing V3 with ELRS RX & TX using INAV 5.1.0

Has anyone found a solution to this?

maxgunn19, 2022-12-31:
Unfortunately, I've given up on INAV for now with this setup. I changed over to ardupilot and haven't seen any problems yet. Best of luck!

breadoven, 2022-12-31:
> I am having the same problem.
>
> I can start up but I cannot arm so my Motors won't spin (they spin in the outputs tab though). I can move the servos around and around for about 15 seconds without [...]

Do you know why it wouldn't arm, any status message in the CLI ? Moving the sticks around when disarmed is likely to trigger a stick command that will cause the problem mentioned in https://github.com/iNavFlight/inav/issues/8409, (although that doesn't fit with the arming issue you mention).

closed 2023-11-17 by DzikuVx