iNavFlight/inav #8433: Add support for DJI wtfos MSP-OSD full screen

jeffhendrix, 2022-09-30:
The new wtfos project ([https://github.com/fpv-wtf/wtfos](https://github.com/fpv-wtf/wtfos))
allows the drawing of all OSD elements on the DJI FPV goggles. 
iNav currently uses the HDZERO mode to take advantage of this feature, but it's limited to 50x18 and the goggles are capable of 60x22.

Changes are required to iNav and the configurator, but should be fairly simple (leveraging the HDZERO code).

jeffhendrix, 2022-09-30:
Can someone assign this to me, I have the code almost finished.

erstec, 2022-09-30:
> Can someone assign this to me, I have the code almost finished.

It is not needed to be assigned. You are welcome to open pull request with your code here for INAV and in https://github.com/iNavFlight/inav-configurator/pulls for INAV Configurator.

But if you prefer, I can assing this issue to you )

jeffhendrix, 2022-09-30:
> It is not needed to be assigned. You are welcome to open pull request with your code here for INAV and in https://github.com/iNavFlight/inav-configurator/pulls for INAV Configurator.
>
> But if you [...]
I just didn't want someone else to duplicate the work
I hope to have a PR soon (I might play with it for a day or 2 first)

erstec, 2022-09-30:
> Can someone assign this to me, I have the code almost finished.
>
>
> It is not needed to be assigned. You are welcome to open pull request with your code here for INAV and in [...]

Anyone can do same development as your one.
As @stronnag already explained you in Discord: _if someone else comes along with a better implementation, we'll take that. It's not exclusivity, it merit that counts_

stronnag, 2022-09-30:
But just the fact that you raised the issue and said you're working on it is enough. 
We're not a megacorp, we don't do "waterfall", we don't have a list of staff to assign work.

We self-manage and welcome contributions, so thanks for volunteering.

PR #8434 (2022-10-18): Added support for DJI wtfos MSP-OSD full screen 60x22 OSD
Changes to allow DJI goggles that have the wtfos installed to use the full screen OSD 60x22
New peripheral was added to the Ports tab "DJI WTF" to enable new mode.
The MSP_OPTIONS subcommand will then send a 2 to the goggles to enable the full screen mode.
OSD tab added the video format option DJIWTF and the preview is enlarged to allow placing elements in the full 60x22 grid.
Pull Request for inav config to support changes is also up.
Closes #8433
closed 2022-10-18 by DzikuVx