iNavFlight/inav #8440: Matek H743 WLITE problem with permanent MANUAL mode

jmajew, 2022-10-01:
## Current Behavior
During configuration of INAV 5.1.0 on my Matek H743 WLITE board for fixed-wing I have noticed problem with MANUAL mode. When I assigned MANUAL to some range of a channel it becomes permanently on - no mater what value the channel is (ACRO is never activated). After some testing I have managed to narrow down to the fact that the problems occurs only when compass is active:

View of the problem in the configurator - ACRO should be on and MANUAL off.
[image]

The same configuration, only the compass was disabled - it is as I would expect it to be.
[image]

I have also checked if the problem is related to the configurator by checking the current mode on OSD - unfortunately the ACRO mode was never activated which would indicate the problem exists inside the firmware. 

Here is diff for the case when issue exist:
[diff_problem.txt](https://github.com/iNavFlight/inav/files/9690642/diff_problem.txt)

## Steps to Reproduce

1. Flash Matek H743 WLITE using INAV 5.1.0 
2. Set platform_type = AIRPLANE
3. Setup RC link and sensors (including compass and GPS)
4. Assign a range of a channel to activate MANUAL mode
5. Change value of the channel to activate ACRO
6. MANUAL mode would be still on

# version
- FC Board name and vendor: Matek H743 WLITE made by Mateksys
- INAV version string: 5.1.0 Aug 19 2022 / 12:34:02 (76f22b25)

MrD-RC, 2022-10-01:
Is this the release version of Configurator? How did you set up the FC? Was it a dump, diff, or fresh setup?

You shouldn't use compass for fixed wing. Not that it excuses the issue. But when you actually go to fly, it's best to disable it.

jmajew, 2022-10-01:
I have used release version 5.1.0 for both firmware and configurator.

When I first found the issue I have been configuring the FC with everything (wizard, mixer, servos, programming logic, etc.) but then I reset the FC and manually applied minimum changes in order to trigger the problem (to exclude as many variables as possible). The file [diff_problem.txt](https://github.com/iNavFlight/inav/files/9690642/diff_problem.txt) is a diff of that configuration.

> You shouldn't use compass for fixed wing. 

I know but since the compass was integrated with GPS I just enabled it to check if it is alive.

jmajew, 2022-10-02:
I checked today how it is with the latest development 6.0.0 firmware. It seems disabling MANUAL mode is ON no matter if compass is disabled or not...

[image]

jmajew, 2022-10-02:
Well, I thing I found what is the reason of permanent MANUAL ... 

According to `fc_core.c` line 740, the MANUAL mode is enabled according to some rules and when sensors are calibrating 
(function `areSensorsCalibrating()`). However, the `areSensorsCalibrating()` seems to return true unless sensor ARE calibrated. 
I don't know if it is planned behaviour or it is exclusive to MATEKH743 FC but after calibrating accelerometer and compass all works as it should. 

If it is as planned then the issue should be closed, although some warning about this would be nice :) - I usually calibrate compass (on my quads) at the end of configuration process when I am in the field where: a) there is space to move, b) there is no interference from all home devices.

MrD-RC, 2022-10-02:
Forcing manual even calibrating sounds good to me.

closed 2022-10-02 by MrD-RC