iNavFlight/inav #8441: Failure to disarm multicopter

Torchi, 2022-10-03:
Hello. This problem is about the failure to disarm by switch, once landed manually the multicopter in HOLD mode.  
I also had one similar event time ago with an other model and different INAV version.  I did not save any data at the time.
Version here is INAV 5.1, FCB is SKYSTARS F7HD Pro3. 
I attach config Diff file, Blackbox and a short video of the event (pls ignore the figures 
superimposed to the video).
As can be seen in the video, I landed peacefully in HOLD after a perfect flight in ACRO or HOLD.  Landed, I tried several times to stop the engines by the switch dedicated to disarm.  While doing this, I got a bit closer,  to see and be sure that the propellers were still spinning.
After several seconds of confusion, I did a terrible mistake: I switched the mode from HOLD to ACRO...  
Throttle did, by its own, a sudden jump but fortunately at the end no damages as I managed to land (again in HOLD)
Besides my mistake, the problem is:      why it could not be disarmed ???

[INAV_5.1.0_cli_Super_8_20220929_131810.txt](https://github.com/iNavFlight/inav/files/9694745/INAV_5.1.0_cli_Super_8_20220929_131810.txt)

[blackbox_log_2022-10-02_144711.TXT](https://github.com/iNavFlight/inav/files/9694750/blackbox_log_2022-10-02_144711.TXT)

https://user-images.githubusercontent.com/18173871/193505603-0c825e7e-1242-442c-8541-fd828e5394fc.mp4

breadoven, 2022-10-03:
The only obvious things that will cause the Disarm switch to be ignored are problems with the receiver signal or being in failsafe neither of which seemed to have happened in this case based on the flight log (last log in the attached file ?).

I take it the disarm switch worked normally on the second landing, didn't need multiple attempts again ? What switch is used exactly, a basic mechanical toggle or a momentary set up to act like a toggle using OpenTx programming ?

Torchi, 2022-10-03:
That switch is mechanically 3 stable positions. I have assigned it for arm / disarm.

breadoven, 2022-10-04:
OK, so it's not obvious what the problem might be. The log doesn't really help much other than to show that INAV didn't receive a command to disarm. So could be an RC issue or maybe an unreliable Tx switch ?

stronnag, 2022-10-04:
> OK, so it's not obvious what the problem might be. The log doesn't really help much other than to show that INAV didn't receive a command to disarm. So could be an RC issue or maybe an unreliable Tx switch ?

Each log entry *does* shows that INAN recorded "disarm by switch":

```
$ bbsummary ~/dl/blackbox_log_2022-10-02_144711.TXT 
...
Log      : blackbox_log_2022-10-02_144711.TXT / 12
Flight   : Super 8 on 2022-10-02 07:54:26
Firmware : INAV 5.1.0 (GITDIR-N) SKYSTARSF722HD of Sep 26 2022 11:42:40
Size     : 1.29 MB
Altitude : 13.3 m at 00:19
Speed    : 3.1 m/s at 04:58
Range    : 37 m at 01:52
Current  : 25.6 A at 00:18
Distance : 152 m
Duration : 05:55
Disarm   : Switch

Log      : blackbox_log_2022-10-02_144711.TXT / 13
Flight   : Super 8 on 2022-10-02 08:55:14
Firmware : INAV 5.1.0 (GITDIR-N) SKYSTARSF722HD of Sep 26 2022 11:42:40
Size     : 1.02 MB
Altitude : 122.1 m at 01:07
Speed    : 27.7 m/s at 01:32
Range    : 828 m at 01:04
Current  : 94.9 A at 03:53
Distance : 2059 m
Duration : 04:23
Disarm   : Switch
```

Torchi, 2022-10-04:
I  am out and cannot do much. Anyway first point: the second time drone  touched ground, (after passing through acro and again hold mode) the disarm worked ok. Is this  probably the reason why flight seems normally terminated by manual disarm ?  Second: I am not able to interpret It fully but can radio telemetry log help to diagnose what failed ? I shall attach It someway, later.

breadoven, 2022-10-04:
Yes I meant there is nothing obvious to indicate any disarm command was received at the time the disarm switch was tried (around 3:35) but failed to work, e.g. `flightModeFlags `stayed at `ARM `(but then I assume this just shows the state of the `ARMING_FLAG `anyway so wouldn't be helpful). There is nothing in the log showing Aux channel values so no way of knowing if the Arming channel changed to the disarm position.

Torchi, 2022-10-04:
Used switch for ARM is SC (see Taranis manual extract, attached).
From the log attached (see the last flight) seems to me that SC stayed ON all the time while I was willing to switch it OFF and went OFF at the very end of the flight (second touch-down).
I believe that switches are like RSSI i.e. telemetered by the receiver itself without FCB intervention (right ?).
This would mean that the problem is somewhere in the system (myself - transmitter - receiver) ... If so, INAV **would be excluded**... I understand if you will not exclude me (LOL). I instead exclude myself because it is second time (I remember) that this happens and in similar situations (HOLD =  switch SB).  For me the problem is in Taranis (OPEN TX sw probably). It could be an idea for future to associate the arming "also" to another emergency switch.  Just to experiment.
[image]
[Super_8-2022-10-02.csv.zip](https://github.com/iNavFlight/inav/files/9709176/Super_8-2022-10-02.csv.zip)

closed 2023-10-27 by DzikuVx