iNavFlight/inav #8450: INAV 5.1 + H743-wing + DJI Air unit + TBS CRSF micro, RC link hangs

zvikaf, 2022-10-06:
System: 
Aircraft: FMS Pilatus PC-21
Firmware: INAV 5.1 
Flight controller: H743-Wing V3
GPS: Beitan BN-220
Digital airspeed sensor
TBS Nano FW v 6.19 

## Current Behavior
TBS CRSF stop communicating after disarming the system when connected to USB and  configurator, 
enters FS mode due to radio lost. disconnecting the USB does not regain communication, 
booting the transmitter does not regain as well.

Works fine and does not happened if not connected to configurator.

## Steps to Reproduce
1. connect battery, wait for boot, check radio command and telemetry, check DJI googles notification.  
2. connect USB to computer and connect configurator
3. wait for GPS lock
4. Arm system
5. check radio functionality (move sticks, check surfaces)
6. disarm
7. check radio functionality (move sticks, check surfaces)

## Expected behavior
system should continue to disarm mode with full radio link

## Suggested solution(s)
check the code

## Additional context
Diff file :  https://pastebin.com/iByJKMHJ

---

- FC Board name and vendor:   INAV/MATEKH743 5.1.0 Aug 19 2022 / 12:34:02 (76f22b25)
- INAV version string: GCC-10.2.1 20201103 (release)

breadoven, 2022-10-06:
Sounds similar to https://github.com/iNavFlight/inav/issues/8409.

Can you try it with the Servo Auto Trim feature turned OFF and see if it still happens ?

zvikaf, 2022-10-06:
@breadoven  thanks, 
it really sounds close, will try without the auto trim tomorrow (sounds real awkward) :-)

zvikaf, 2022-10-07:
@breadoven  Bingo...  checked now disabling auto trim, and the RC link was fine ... waited a minute to see if the link revives, did not,  left the autotrim off...  Thanks :-)