iNavFlight/inav #8453: Add Y3 support to the fixed wing, get the vertical takeoff feature for a fixed wing.

gghyoo, 2022-10-09:
## Current Behavior

## Desired Behavior

I want to add a function for my airplane which has three motors(in a Y Sharp), two motors can change the direction from horizontal to vertical which mounted in the two sides of wing , one motor mounted in the tail at vertical direction. so I can control the plane Vertical Takeoff and change to a normal airplane in the air. 

## Suggested Solution

## Who does this impact? Who is this for?

## Additional context

there is a demo for my request:
[https://www.youtube.com/watch?v=NJMIgvoN6aU](url)

thehammstr, 2022-10-11:
I’ve been thinking about ways to do this for a while. The challenge here is that instead of a hard logical switch between flight modes like you'd have for a tailsitter, you want a gradual interpolation of different control mixes based on the nominal tilt of he front motors.

It’s not exactly pretty, but I think this can be accomplished with a combination of global variables and logical rules. Your setup may differ by a few pluses and minuses, but I think something like this should work. There are also probably some optimizations that could be made, like caching the sin computation.

Two front rotors that rotate, one fixed rear. 

Mix strategy: Multiply the stabilized angle inputs by sin or cos of the nominal motor tilt so that they gradually shift in and out between cruise and hover.

In hover, only control yaw with differential tilt of the two front motors. You can also optionally control yaw in cruise with differential thrust, but you could also just leave it as a normal twin engine fixed wing.

Motor tilt: 
servo 1000 =    0 deg = cruise
servo 2000 = 90 deg = vertical

Set this up on a slider or slow servo from the transmitter, let’s say Channel 5.

Create a global variable that maps this channel from 0 to 90
logic 0 1 -1 15 1 5 0 1000 0 // map channel 5 to 0-1000
logic 1 1 -1 37 4 0 0 90 0 // map  result of LC0 to 0-90. This is the tilt angle.
logic 2 1 -1 33 4 1 2 26 0 // compute sin(tilt)*stabilized roll
logic 3 1 -1 18 4 2 0 1 0 // Store result of  LC2 in GVAR 1
logic 4 1 -1 33 4 1 2 27 0 // compute sin(tilt)*stabilized pitch
logic 5 1 -1 18 4 4 0 2 0 // Store result of LC4 in GVAR 2
logic 6 1 -1 33 4 1 1 3 0 // Compute sin(tilt) * throttle
logic 7 1 -1 18 4 6 0 3 0 // Store result of LC4 in GVAR 3
logic 8 1 -1 33 4 1 2 28 0 // compute sin(tilt)*stabilized yaw
logic 9 1 -1 18 8 4 0 4 0 // Store result of LC4 in GVAR 4

Motor & servo mixes:

LeftFrontMotor = 
throttle 
+ hover_roll_weight*stabilized_roll*sin(tilt)  
+ hover_pitch_weight*stabilized_pitch*sin(tilt) 
+ (optional)hover_yaw_weight*stabilized_yaw*cos(tilt)
OR
throttle
+ hover_roll_weight * GVAR1
+ hover_pitch_weight * GVAR2 

LeftFrontMotor = 
throttle 
- hover_roll_weight*stabilized_roll*sin(tilt)  
+ hover_pitch_weight*stabilized_pitch*sin(tilt) 
+ (optional)hover_yaw_weight*stabilized_yaw*cos(tilt)
OR
throttle
- hover_roll_weight * GVAR1
+ hover_pitch_weight * GVAR2 

RearMotor = 
throttle*sin(tilt)
- hover_pitch_weight*stabilized_pitch*sin(tilt) 
OR
GVAR3
-hover_pitch_weight * GVAR2

LeftTiltServo = 
RC_CHANNEL_5(slow)
+ hover_yaw_weight * stabilized_yaw*sin(tilt)
OR
RC_CHANNEL_5
+ GVAR4

LeftTiltServo = 
RC_CHANNEL_5(slow)
- hover_yaw_weight * stabilized_yaw*sin(tilt)
OR
RC_CHANNEL_5
- GVAR4

I am only starting to dig into the inav code, so if anyone more experienced can spot errors or improvements to be made, please let me know. I haven't tested this out, so I don't know if, for instance, global variables get updated often enough

Pairan, 2022-10-11:
Finally one goes for it!

gghyoo, 2022-10-11:
great， I will try that in my plane.

gghyoo, 2022-10-11:
> I’ve been thinking about ways to do this for a while. The challenge here is that instead of a hard logical switch between flight modes like you'd have for a tailsitter, you want a gradual interpolation of different control mixes based on the nominal tilt of he front motors.
> 
> It’s not exactly pretty, but I think this can be accomplished with a combination of global variables and logical rules. Your setup may differ by a few pluses and minuses, but I think something like this should work. There are also probably some optimizations that could be made, like caching the sin computation.
> 
> Two front rotors that rotate, one fixed rear.
> 
> Mix strategy: Multiply the stabilized angle inputs by sin or cos of the nominal motor tilt so that they gradually shift in and out between cruise and hover.
> 
> In hover, only control yaw with differential tilt of the two front motors. You can also optionally control yaw in cruise with differential thrust, but you could also just leave it as a normal twin engine fixed wing.
> 
> ```
> Motor tilt: 
> servo 1000 =    0 deg = cruise
> servo 2000 = 90 deg = vertical
> 
> Set this up on a slider or slow servo from the transmitter, let’s say Channel 5.
> 
> Create a global variable that maps this channel from 0 to 90
> logic 0 1 -1 15 1 5 0 1000 0 // map channel 5 to 0-1000
> logic 1 1 -1 37 4 0 0 90 0 // map  result of LC0 to 0-90. This is the tilt angle.
> logic 2 1 -1 33 4 1 2 26 0 // compute sin(tilt)*stabilized roll
> logic 3 1 -1 18 4 2 0 1 0 // Store result of  LC2 in GVAR 1
> logic 4 1 -1 33 4 1 2 27 0 // compute sin(tilt)*stabilized pitch
> logic 5 1 -1 18 4 4 0 2 0 // Store result of LC4 in GVAR 2
> logic 6 1 -1 33 4 1 1 3 0 // Compute sin(tilt) * throttle
> logic 7 1 -1 18 4 6 0 3 0 // Store result of LC4 in GVAR 3
> logic 8 1 -1 33 4 1 2 28 0 // compute sin(tilt)*stabilized yaw
> logic 9 1 -1 18 8 4 0 4 0 // Store result of LC4 in GVAR 4
> 
> Motor & servo mixes:
> 
> LeftFrontMotor = 
> throttle 
> + hover_roll_weight * stabilized_roll * sin(tilt)  
> + hover_pitch_weight * stabilized_pitch * sin(tilt) 
> + (optional)hover_yaw_weight * stabilized_yaw * cos(tilt)
> =
> throttle
> + hover_roll_weight * GVAR1
> + hover_pitch_weight * GVAR2 
> 
> LeftFrontMotor = 
> throttle 
> - hover_roll_weight * stabilized_roll * sin(tilt)  
> + hover_pitch_weight * stabilized_pitch * sin(tilt) 
> + (optional)hover_yaw_weight * stabilized_yaw * cos(tilt)
> =
> throttle
> - hover_roll_weight * GVAR1
> + hover_pitch_weight * GVAR2 
> 
> RearMotor = 
> throttle*sin(tilt)
> - hover_pitch_weight*stabilized_pitch*sin(tilt) 
> =
> GVAR3
> -hover_pitch_weight * GVAR2
> 
> LeftTiltServo = 
> RC_CHANNEL_5(slow)
> + hover_yaw_weight * stabilized_yaw*sin(tilt)
> =
> RC_CHANNEL_5
> + GVAR4
> 
> LeftTiltServo = 
> RC_CHANNEL_5(slow)
> - hover_yaw_weight * stabilized_yaw*sin(tilt)
> =
> RC_CHANNEL_5
> - GVAR4
> ```
> 
> I am only starting to dig into the inav code, so if anyone more experienced can spot errors or improvements to be made, please let me know. I haven't tested this out, so I don't know if, for instance, global variables get updated often enough

Sorry, I don't know how to config the Mixer to set the motor output such as RearMotor = GVAR3- hover_pitch_weight * GVAR2

Set it in the Logical Conditions?

thehammstr, 2022-10-11:
@gghyoo Ah, it looks like the documentation in Mixer.md is out of date, but the code at https://github.com/iNavFlight/inav/blob/master/src/main/flight/servos.h#L57 indicates that you can use GV0-7 as servo inputs.
```
smix <n> <servo_index> <input_id> <weight> <speed> <logic_condition_id>
```
so <input_id> = 30 would use GV0 as an input for that mix rule.

Again, I haven't tested this and I don't have any planes around right now because I'm traveling, but it looks like that should work. I'm basing this all off of the documentation in https://github.com/iNavFlight/inav/blob/master/docs/Programming%20Framework.md. Good luck!

gghyoo, 2022-10-12:
Thanks a lot, I was talking about the motor throttle config, not the server mix config. it seams that, the mmix command can't use the GVARs. I find the Override Throttle operation in the Logical Conditions config, here I can combine throttle and global variables.

thehammstr, 2022-10-12:
That's a fair point. I think if your speed controllers accept PPM input you could just use some of the servo output channels to control them, and use IS_ARMED logic condition to make them safe. You may have a bit higher latency, but these vtol platforms don't usually spend too much time in hover or require high maneuverability. If you can hover for takeoff and landing, it should be good enough. I used to build quadrotors ten years ago back before things like DSHOT were available, just using servo pulses to control motors, and while it wouldn't be up to modern standards of latency and responsiveness, it worked fine for cruising around.

gghyoo, 2022-10-14:
OK，I will try other channel to control the ESC

ddano007, 2023-07-10:
Hi, gghyoo! I`m looking for INAV support for my VTOL & F405WMN & TX16S. What`s the status of Your modification?
[image]

DzikuVx, 2023-10-27:
VTOL is coming to INAV 7

closed 2023-10-27 by DzikuVx