iNavFlight/inav #8454: INAV 5.1.0 did not return to home with normal accuracy

Torchi, 2022-10-09:
Drone was sent to an automatic mission (attached zip of xml)

The automatic mission goes to a place 600m away and comes back, repeats once the going and coming, finally lands home.

This automatic mission was flown one flight .  Then battery changed.  Flown again in a second flight.

While first flight RTH  returned "exactly" (i.e. within a meter or two) to home, second flight returned with maybe 20 meters offset (therefore outside runfield...).

Everything was same in the two missions besides texact take off locations (both were in the runfield but a bit different) and manual take offs.  

The recording of the two missions is attached (BB).

Attached also pictures of the two missions and details showing home area.

[600m.mission.zip](https://github.com/iNavFlight/inav/files/9741257/600m.mission.zip)

[blackbox_log_2022-10-08_114352.TXT](https://github.com/iNavFlight/inav/files/9741258/blackbox_log_2022-10-08_114352.TXT)

[image]

[image]

[image]

[image]

stronnag, 2022-10-09:
In fact, the second attempt terminates precisely at WP8. 

[image]

stronnag, 2022-10-09:
It is also the case that at BBL iteration `292032` ( 294.6s), the HOME position is changed in the BBL. 

Do you have a `reset home` switch ? Did you accidentally switch it?

Torchi, 2022-10-09:
I have a switch that makes the drone return home.  I have not touched switches at that time (as far as I am aware of). 
But even in that accidental case, that "return home" switch  is what you mean for 'reset home switch' ?

stronnag, 2022-10-09:
> But even in that accidental case, that "return home" switch is what you mean for 'reset home switch' ?

No. There is also a "HOME RESET" mode.

breadoven, 2022-10-09:
It shouldn't be possible to reset Home in WP or RTH modes even if attempted. It's prevented according to the code. The target position changed as soon as WP8 was reached to within around 2 meters of the WP8 position. Went straight to RTH landing so it must have thought it was Home. Does it show a definite Home Reset flag @stronnag in your tools, can't anything like that in Blackbox Explorer or Decoder ?

stronnag, 2022-10-09:
> Does it show a definite Home Reset flag @stronnag in your tools, can't see anything like that in Blackbox Explorer or Decoder ?

All my stuff depends on `blackbox_decode`. I'm also baffled as to why the home changed (which is clear in the BBL).

breadoven, 2022-10-09:
@Torchi What is your setting for nav_min_rth_distance ? Seems to be 5000 in the dump from https://github.com/iNavFlight/inav/issues/8441 which is much higher than the default of 500. This will cause RTH to land if RTH is initiated when within 50 m of home.

stronnag, 2022-10-09:
We have a winner!

* For flight 1, Home (44.801275 8.762048)    is 53m from WP8 (RTH)
* For flight 2, Home (44.801205 8.762211) is 40m from WP8 (RTH)

Expected behaviour.

Torchi, 2022-10-09:
Congratulations !
I think that if I did it on purpose (to put WP8 in such a critical  position wrt the landing strip) I would never succeed ...

The 5000, non-default, value for nav_min_rth_distance comes probably from some old situation that I do not even remember of.
Actually I do not see now a value for me to use a figure different from  500 so, from now on, I shall revert to  that default.
Thank you.

stronnag, 2022-10-09:
I think we can now close this. Interesting investigation.

closed 2022-10-09 by stronnag