iNavFlight/inav #8497: INAV GPS messes up simple Angle mode

PetruSoroaga, 2022-10-24:
## Current Behavior
In INAV 5.1, if I enable GPS for navigation, then simple Angle mode (no other NAV modifiers) does not work: I can't take off, motor spin up only up to about 25% power, no matter throttle position.
See the screenshot attached.
More info is explained here: https://www.rcgroups.com/fpv-equipment-862/

## Steps to Reproduce

1. Setup flight modes, one channel for Arm, one channel for Angle/Alt Hold, Pos Hold, RTL
2. With GPS for navigation disabled, Arm and fly in Angle mode. All is fine.
3. Enable option: Use GPS for navigation.
4. When arming and trying to fly in Angle mode or Alt hold, the quad does not take off. Motors spin up only up to about 25% throttle.

## Expected behavior
Simple Angle mode should work no matter if GPS is present or if a 3d Fix is obtained. (by the way, GPS and magnetometer are configured correctly, no errors, a 3D fix is obtained and shown on the map).

 FC Board name and vendor: Matek F405 mini TE: http://www.mateksys.com/?portfolio=f405-minite
- INAV version string: INAV firmware they provide on their site: http://www.mateksys.com/?p=5159#tab-id-2

I attached the full parameters list from CLI.
Thank you!

[image]
[image]
[INAV_5.1.0_cli_20221024_184610.txt](https://github.com/iNavFlight/inav/files/9853700/INAV_5.1.0_cli_20221024_184610.txt)

Jetrell, 2022-10-24:
Turn off Althold for take-off. You can't expect the quad to lift off and hold altitude at the same time.
Althold is a flight modifier, that requires a GPS fix to operate.

PetruSoroaga, 2022-10-25:
AltHold was not engaged on takeoff.
And it should not matter: as I said, I had GPS fix, and AltHold should work on take off too, that's how altitude hold works: with throttle you control the vertical speed.

PetruSoroaga, 2022-10-25:
> Turn off Althold for take-off. You can't expect the quad to lift off and hold altitude at the same time. Althold is a flight modifier, that requires a GPS fix to operate.

@Jetrell Please re-read the description and my follow up, I think you missunderstood the issue or I did not explained it well:
The issue is not that quad can't take off in NAV_x modes without a GPS fix (which would be normal and expected).
The issue IS that the quad can't take off in [B]non  NAV_x[/B] modes when i DO GET a GPS fix :)
That is not expected.

Jetrell, 2022-10-25:
>Kindly please reopen the issue as this was not the issue and it does not solve the issue

Don't worry. The issue isn't closed.
I understand what your saying.
And to be honest, it seems rather unusual !

If I didn't own one of these convoluted flight controllers, I'd say its a setup problem.. But I do own one of them. And I've had my fair share of setup issue with the Matek F405TE as well..  Even simple things. like getting the beeper to work on any occasion, except initialization.

I don't have a GPS connected to mine, so I can't confirm your issue either. 
Nothing comes to mind, off hand.

breadoven, 2022-10-25:
Better to provide a DIFF of the settings. Makes it easier to see what's been changed from defaults.

PetruSoroaga, 2022-10-28:
Flashed INAV 4.1, same params and calibration, no issues.
So definitively there is a bug on INAV 5.1 related to this issue.

Jetrell, 2022-10-31:
> Flashed INAV 4.1, same params and calibration, no issues. So definitively there is a bug on INAV 5.1 related to this issue.

If you still haven't had any luck working this out.. I thought I'd mention that I tried my Matek F405TE with a GPS fix, in `Horizon mode`, which is based on `Angle mode,` with the 6.0 pre-release... And the quad had no issues taking off.

PetruSoroaga, 2022-11-03:
Not managed to find a fix (most likely a bug in the code), same behavior: works fine in INAV 4.1, broken in INAV5.1 (as in description).

stronnag, 2022-11-03:
* Please provide a CLI diff from both 4.1 and 5.1
* Please provide a 5.1 blackbox log of the issue (and a 4.1 log showing the behaviour there)

PetruSoroaga, 2022-11-06:
I have flashed 4.1, I don't have 5.1 or the logs.
It reproduces always on that flight controller, when I did flashed 5.1, default settings. Should be easy to reproduce.

stronnag, 2022-11-06:
OK, if you don't want us to help, I'l close the issue

closed 2022-11-06 by stronnag

PetruSoroaga, 2022-11-09:
Strange way of dealing with issues you have there.
Just because I re-flashed the FC with a working version, does not mean the issue is solved and should be closed.
The issue still exists.
If YOU don't want to fix it, then yes, close it. What do I care...

stronnag, 2022-11-09:
We are more than happy to investigate this issue if you provide the data to let us do that.

PetruSoroaga, 2022-11-10:
You have the exact hardware configuration in the description and a full parameter list dump.
Is that hard to flash that FC with 5.1 and just import all params and debug and see what the issue is?
It reproduced 100% each time I tried 5.1. It was ok each time I flashed 4.1.
As I said, if you are too lazy to setup a FC and debug and fix (just take any 5-7 inch quad, flash 5.1 for that FC, do a full param import, adjust your radio params, calibrate and try to fly in Angle Mode with and without a GPS fix. That's all), that's fine. Just close the issue with Won't Fix and we all move on with our lives.
Closing it as "Complete" because no one wants to investigate and fix this issue it's just bad practice. But what do I know, it's not my project.

MrD-RC, 2022-11-10:
@PetruSoroaga Please do not be rude to the developers. **You** were asked for a blackbox recording for an issue that **you** are having.

@Jetrell has tested this with the same FC and a GPS, in Horizon mode; which when the gimbals are less that 70% __is__ Angle mode. He could not reproduce the problem. This is why it would be helpful to have a blackbox recording with **your** quad, which does exhibit this issue. 

If developers can't reproduce the problem, and you are unwilling to help. What else can be done. Your rudeness is unnecessary.

breadoven, 2022-11-10:
@PetruSoroaga Do you not think others would be reporting the same problem if such a fundamental bug existed ? Many others fly with a working GPS and only Angle mode enabled without issues. This is the main reason why this is considered unlikely to be a bug but instead more likely caused by a setting error or some other issue related to your setup.

Developers can't help if they have nothing useful to work with such as a basic settings DIFF and flight logs. Without this information they are unlikely to spend time searching for a needle in a haystack "bug" knowing that it probably doesn't exist in the first place.