iNavFlight/inav #8502: Switch from 4S to 6S battery makes quad react unsteady when throttle up at Take-off

ArnoTlse, 2022-10-25:
## Current Behavior
Hi,

Not sure If it is a bug, maybe more a "User bug" :-) .
Here is the context for which I need some support.

I have a quadcopter running well with the following configuration : 
- 5inch quad
- motors RCINPOWER MOTEUR WASP MAJOR 22.6-6.6 - 2020KV
- Lumenier H743 FC with Mamba ESC F55 BL32
- 4S battery (LIPO & LION)
- Rev : INAV 5.1.0 (76f22b25) MATEKH743

As my quad set is compatible to a 6S Battery configuration, I wanted to test to fly with a 6S battery.
Last week-end, I flew 2 Packs in 4S LIPO 1500mah without any pb. And then, I switch to the 6S 1400mah LIPO for a try. Normally, I was not expected any issue as the weight of the 6S battery is just 20g more than the 4S I use. 
When I armed the quad, with low throttle, motors spined normally but I first noticed that the Amp consumption given by the OSD was twice the value than in 4S configuration (6S = 5A, 4s=2.5A). Maybe I'm wrong, but I was expected the other way since I have higher voltage with 6S. This is confirmed by the BlackBox also.
Then when I slightly pushed the throttle, the quad started to bump and pushed itself in the air erraticaly. I have tested it 3 times with same behavior. Hopefully, by desarming quickly, no damage so far.

Here is a screenshot of the blacbox when quad is armed on ground with low throtle just before take-off in 4S config. Current = 2,5A, Voltage = 4S 
[image]

Here is a screenchot in same condition except that it is in 6S battery config. Current = 5A, Voltage = 6S.
[image]

We can see the Motor Commands that go crazy.
[image]

Thus, I try to understand why my current consumption gives me a higher number (twice) in 6S than in 4S config ? And what I potentially miss when using 6S battery on my configuration ?

Regards

Jetrell, 2022-10-25:
If your motors `Kv` and propeller size is suited to a 4 cell setup. Then it will certainly draw more current when a higher voltage is applied to that combination.. The only time a 6 cell setup becomes more efficient, is when the motor `Kv` is lowered to suit that specific battery potential.
Its about exchanging the higher Voltage for lower Current, to provide the same Power output.
But if you stay with your present motors and props. It would be best to add a throttle limiter.

Its a common occurrence when making the motors rev harder with 6S, that the IMU will experience higher levels of vibration.
So its even more important to be sure your motors and props are well balanced.

closed 2022-10-26 by MrD-RC

ArnoTlse, 2022-10-26:
Hi,

Thanks. 
To make a long story short, previously, on my quad I had 2400kv 4s motors running on 4s lipo. Since a crash, I have replaced motors by lower kv 2020kv 6s motors in order to switch to 6s with 5. 1 x 4.3 x3 props. So normally, my current setup should be 6s. It's just that while waiting for a 6s battery, I flew it in 4s lipo.

So I understand that on the same config (2020 kv motors) , running in 4s, at armed idle throttle (15% as Inav default value), 4s rig should draw less current than 6s rig as rpm at 15% is higher on 6s lipo than 4s lipo on same motor. So I shouldn't be worried by the osd value given. 

According vibration, I will recheck my gear. Motor are recent without crash before testing 6s lipo and fly well with 4s battery. 
If I look at the rpm range at take off (throttle 10% = 5000 rpm, 30%=14500rpm) in 6s config, it seems to be within rpm range in 4s config when I fly casual (10%=3500, 50%=15000). And if I look at the frequency domain in blackbox in 4s flights with some roll, flips, and push out, I'm not seeing significant harmonics. So, even if 6s battery is requiring more rpm than the 4s battery on the same motor+prop setup, it seems to be within a known range where I haven't encountered issue due to vibration. So at least take off in 6s should go well, which is not the case. 
I'll recheck motor and props balance anyway. 

Regards