iNavFlight/inav #8503: gps wrong positioning

ozkancikikci, 2022-10-25:
I am using fixed wing, matek f411 wte -beitan 880q

rth not working properly, why does it go to wrong location approx. 200 m

MrD-RC, 2022-10-25:
Have you turned off extra arming safety?

How many 3D satellite locks do you have when you arm?

ozkancikikci, 2022-10-25:
> Have you turned off extra arming safety?
> 
> How many 3D satellite locks do you have when you arm?

I didn't turn off security,about 11-13 satellites

Does the compass affect?

Jetrell, 2022-10-26:
Do you have `Safehome` enabled?
It will return the craft to a different home location.
Do you have the home position distance and system messages displayed on your OSD?. This may help determine whats occurring.

breadoven, 2022-10-26:
A DIFF file of settings would be useful.

ozkancikikci, 2022-10-26:
> A DIFF file of settings would be useful.

diff

# version
# INAV/MATEKF411TE 4.1.0 Mar 29 2022 / 12:27:37 (eaa27a45)
# GCC-10.2.1 20201103 (release)

# start the command batch
batch start

# resources

# mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000

# servo mix
smix reset

smix 0 1 1 100 0 -1
smix 1 2 0 100 0 -1
smix 2 3 0 100 0 -1

# servo
servo 1 1000 2000 1500 -100
servo 2 1000 2000 1500 -100
servo 3 1000 2000 1500 -100

# safehome

# logic
logic 0 1 -1 4 1 10 0 0 0
logic 1 1 -1 5 1 10 0 0 0
logic 2 1 -1 6 1 10 0 0 0
logic 3 1 0 25 0 1 0 0 0
logic 4 1 1 25 0 2 0 0 0
logic 5 1 2 25 0 3 0 0 0

# gvar

# pid

# feature
feature MOTOR_STOP
feature GPS
feature LED_STRIP
feature PWM_OUTPUT_ENABLE

# beeper

# blackbox
blackbox -NAV_ACC
blackbox NAV_POS
blackbox NAV_PID
blackbox MAG
blackbox ACC
blackbox ATTI
blackbox RC_DATA
blackbox RC_COMMAND
blackbox MOTORS

# map

# serial
serial 0 2 115200 115200 0 115200
serial 30 0 115200 57600 0 115200
serial 31 2048 115200 115200 0 115200

# led
led 0 15,11::GI:0

# color
color 1 170,75,255
color 6 359,0,255

# mode_color
mode_color 6 7 5

# aux
aux 0 0 0 1300 2100
aux 1 1 1 1325 1675
aux 2 12 1 900 1150
aux 3 10 2 1700 2100
aux 4 21 3 1800 2100
aux 5 37 3 1275 1675
aux 6 36 4 1700 2100

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout
osd_layout 0 0 21 1 V
osd_layout 0 1 24 7 V
osd_layout 0 2 0 0 V
osd_layout 0 3 8 6 V
osd_layout 0 4 8 6 V
osd_layout 0 9 1 2 H
osd_layout 0 10 2 12 H
osd_layout 0 11 2 3 H
osd_layout 0 12 1 4 H
osd_layout 0 14 24 1 V
osd_layout 0 15 1 3 V
osd_layout 0 22 14 11 V
osd_layout 0 23 1 12 V
osd_layout 0 24 12 2 V
osd_layout 0 32 24 8 V
osd_layout 0 34 10 1 V
osd_layout 0 105 1 10 V
osd_layout 0 112 24 10 V

# master
set gyro_main_lpf_hz = 25
set dynamic_gyro_notch_q = 250
set dynamic_gyro_notch_min_hz = 30
set acc_hardware = BMI270
set acczero_x = 44
set acczero_y = -28
set acczero_z = 22
set accgain_x = 4105
set accgain_y = 4100
set accgain_z = 4077
set align_mag = CW270FLIP
set mag_hardware = QMC5883
set magzero_x = -96
set magzero_y = -127
set magzero_z = -728
set maggain_x = 911
set maggain_y = 1003
set maggain_z = 1042
set baro_hardware = SPL06
set motor_pwm_rate = 16000
set motor_pwm_protocol = DSHOT600
set failsafe_procedure = RTH
set failsafe_min_distance_procedure = RTH
set platform_type = AIRPLANE
set has_flaps = ON
set model_preview_type = 30
set small_angle = 180
set imu_acc_ignore_rate = 9
set imu_acc_ignore_slope = 5
set applied_defaults = 3
set gps_provider = UBLOX7
set gps_sbas_mode = AUTO
set airmode_type = STICK_CENTER_ONCE
set nav_wp_radius = 5000
set nav_rth_allow_landing = NEVER
set nav_rth_alt_mode = AT_LEAST_LINEAR_DESCENT
set nav_rth_altitude = 4000
set nav_fw_bank_angle = 45
set nav_fw_climb_angle = 30
set nav_fw_dive_angle = 30
set nav_fw_loiter_radius = 5000
set nav_fw_control_smoothness = 2
set nav_fw_launch_accel = 1500
set nav_fw_launch_motor_delay = 200
set nav_fw_launch_end_time = 2000
set osd_crosshairs_style = AIRCRAFT
set vtx_power = 2

# profile
profile 1

set fw_p_pitch = 15
set fw_i_pitch = 5
set fw_d_pitch = 5
set fw_ff_pitch = 80
set fw_p_roll = 15
set fw_i_roll = 3
set fw_d_roll = 7
set fw_p_yaw = 20
set fw_i_yaw = 0
set fw_ff_yaw = 100
set max_angle_inclination_rll = 450
set dterm_lpf_hz = 10
set fw_turn_assist_pitch_gain =  0.500
set nav_fw_pos_z_p = 15
set nav_fw_pos_z_d = 5
set nav_fw_pos_xy_p = 60
set d_boost_min =  1.000
set d_boost_max =  1.000
set rc_expo = 30
set rc_yaw_expo = 30
set roll_rate = 18
set pitch_rate = 9
set yaw_rate = 3

# battery_profile
battery_profile 1

set battery_capacity = 4000
set battery_capacity_warning = 3200
set battery_capacity_critical = 3600
set throttle_idle =  5.000
set nav_fw_cruise_thr = 1500
set nav_fw_max_thr = 1850

# end the command batch
batch end

#

breadoven, 2022-10-26:
Well `nav_wp_radius `is way to big at 5000cm. Default is 100cm. This should be kept close to this value. Having said that it doesn't really explain why it would be 200m away from the Home position. Try changing `nav_wp_radius `all the same to the default of 100 and see what happens.

stronnag, 2022-10-26:
And please provide a blackbox log

breadoven, 2022-10-27:
> @breadoven it's a fixed wing. `nav_wp_radius` is at 5000cm default from the fixed wing preset to avoid "last meter" course corrections in windy conditions and compensate for the big turn radius of a fixed wing. Might reduce it to 2000cm now that we have better AHRS and turn smoothing though.

That makes no sense though. This setting determines when a waypoint has been reached which means it will consider a waypoint has been reached when in fact it's still 50m from it which is a big error. It also doesn't just rely on this to decide it's reached a waypoint it also checks to see if it has past (missed) a waypoint by a certain angular bearing which is often the check that decides a waypoint has been reached rather than the distance.

Does this 50m setting actually work ? It's wouldn't affect loiter waypoints such as RTH but normal Waypoint Mission waypoints would be consistently reached too early surely ?

b14ckyy, 2022-10-27:
A fixed wing has not the navigation precision to hit a WP positon within 1m if there is any crosswind happening. This often leads to strong and rapid course corrections during the last second of approach. The 50m radius avods that. Keep in mind that most planes fly at 20-30m/s so even if the waypoint is considered as "hit" 50m early and he plane starts the turn, it takes a while for the course to change and the waypoint is still passed at a distance of 10-20m at most. this also depends on the angle the next waypoint is at. For a while we had the Preset at 15m radius but with strong crosswinds and gusts we still had situations where the plane did rapid course corrections before hitting or even missing the WP, especially bigger or faster crafts. 

50m was a good value that worked for any plane so far and if more precision is needed the pilkot can still reduce the WP radius. It was important that the out of box nmavigation works reliable on any plane.

breadoven, 2022-10-27:
OK, I've never used pre-sets for fixed wing so missed that. I guess reaching a waypoint early is actually just a crude version of turn smoothing. Only thing I'd add is the waypoint tracking accuracy option eliminates the rapid course corrections you get when you close in on a waypoint because it homes in on the waypoint course line rather than homing in on the single waypoint position. This avoids the sudden rapid increase in heading correction you get when closing in a single point in space.

Anyway as far as this issue is concerned it's clearly not related.

b14ckyy, 2022-10-27:
I think when turn smoothing is used then indeed this WP radius could be smaller. you know what? Maybe that was the reason why sometimes the smoothing worked and sometimes it did not for me in my testing because it could have hit the WP before the smoothing algorythm kicked in. 
But this does not relate here I will check that and come back to you if I can confirm that.

Mozzytec, 2022-11-04:
Hi,
FIrst post :-)
Don't know if it is linked or not but got the same behaviour with F405 and same GPS beitan 880. Really nightmare...
Several crashes... due to GPS behavior and always RTH completly out of range. (far and far and far from home)
I took gps coordinate (via blackbox) and downloaded on google earth and I can see that from circle to circle around "home position" it deviates and deviates...
After turning around, changing lot's of parameters in INAV 5.0, I decided to change GPS and went to a Matek GPS. 
Now everything is running really fine with same Inav configuration. (Loiter radius 35m and all other paramter set by default).

So question what aboiut if you change your GPS?

Hope it helps.
Cheers.

closed 2022-11-04 by stronnag

ozkancikikci, 2022-11-06:
When I turned off the compass, my problem was fixed.
thnx
regards