iNavFlight/inav #8504: Holybro Kakute H7 mini (V1.3) accelerometer not working

mdhendri, 2022-10-26:
The firmware for INAV 5.1 and 6.0 doesn't seem to support the V1.3 version of the Holybro Kakute H7 mini with the BMI270.  I have tried to select the BMI270 and also the autodetect and it won't detect it.  I does however work in Betaflight.

JulioCesarMatias, 2022-10-27:
can you test this build? >> [inav_6.0.0_KAKUTEH7.zip](https://github.com/iNavFlight/inav/files/9874620/inav_6.0.0_KAKUTEH7.zip)

to use this build you need to download the configurator here >> http://seyrsnys.myzen.co.uk/inav-configurator-next/

DzikuVx, 2022-10-27:
The official support for 1.3 will come quite soon. Hopfully a week or so

BrownC0de, 2022-11-10:
> The official support for 1.3 will come quite soon. Hopfully a week or so
Has the 1.3 update been released yet?

DzikuVx, 2022-11-11:
Yes