iNavFlight/inav #8506: Show Alarms as System Messages

b14ckyy, 2022-10-26:
## Current Behavior
Right now, all alarms only make OSD Items blink. This means we cannot have Alarms for Items that are not on Screen. Also Blinking on MSP-DP based systems (Basically all digital FPV systems now) has issues with very slow blink rates or disappearing elements. 

## Desired Behavior
Show Alarms as System Messages and/or blinking as Options. 

## Suggested Solution
I suggest that any Alarm can be shown as a System message. 
There should be 3 Settings to select from for Alarms: Blinking / System Message / Both
If multiple alarms appear, they should rotate as usual
If possible, general Info messages like Waypoint info or Radar Info should be disabled when an alarm happens to make sure it is not missed. 

## Who does this impact? Who is this for?
Any Pilot in General but mostly Digital Pilots where the OSD Elements have usually a very slow update rate to preserve bandwidth. 
This also allows a cleaner OSD with less elements while still preserve critical warnings like ESC Temp, IMU temp or RSSI/SNR Alarms while only having LQ on screen. 

## Additional context
-

tonyyng, 2023-04-06:
> Also Blinking on MSP-DP based systems (Basically all digital FPV systems now) has issues with very slow blink rates or disappearing elements.

Can you provide more information on the slow blink rates for MSP-DP?  Is a solution for this problem not possible?  I'm encountering this issue with INAV 6.0 and DJIWTF and haven't found it discussed anywhere.

b14ckyy, 2024-03-22:
closing as duplicate of https://github.com/iNavFlight/inav/issues/8121

closed 2024-03-22 by b14ckyy