iNavFlight/inav #8514: Constant MSP Packet errors and very slow MSP data rate as a result

b14ckyy, 2022-10-30:
Somewhere between #2952 and #2951 something changed that causes MSP throwing 2-3 packet errors per second on F411 FCs. 
Only seems to affect F411 targets, specifically the MATEKF411SE. MATEKF405SE shows no problem. 
Packet errors constantly raise and the data rate is extremely low. Applying a DIFF in CLI also aborts at one point and skips a bunch of settings suddenly. 

Saw it the first time on [e088782](https://github.com/iNavFlight/inav/pull/7018/commits/e0887825136dd1fd00df300af255a8441db00b70) but then narrowed it down to somewhere between the above mentioned PRs. #2951 works fine. 

Tasks show no problems:

```

Task list         rate/hz  max/us  avg/us maxload avgload     total/ms
 0 -       SYSTEM       9      12       1    0.5%    0.5%      4393
 1 -          PID     972     135     107   13.6%   10.9%      9231
 2 -         GYRO    3875      30      19   12.1%    7.8%      6465
 3 -           RX       9      56      50    0.5%    0.5%        46
 4 -       SERIAL      99   110671      21 1096.1%    0.7%       489
 5 -      BATTERY      49      23      20    0.6%    0.5%        85
 6 -  TEMPERATURE      99       4       2    0.5%    0.5%        13
 7 -       BEEPER      99      13       6    0.6%    0.5%        56
10 -         BARO      43     235     156    1.5%    1.1%       600
14 -    TELEMETRY     480      22       3    1.5%    0.6%       210
17 -          OSD     244    1941      18   47.8%    0.9%       402
18 -          CMS      49       4       2    0.5%    0.5%         5
21 -      VTXCTRL       4       3       1    0.5%    0.5%         0
22 -  PROGRAMMING       9      25      20    0.5%    0.5%        17
24 -          AUX      99       7       3    0.5%    0.5%        26
25 - SPORT MASTER     480       4       1    0.6%    0.5%        42
Task check function             2       1                         1
Total (excluding SERIAL)                    81.8%   26.3
```

I also tried commit #8493 that does not show MSP Packet errors. Tasks look the same though. 

```
Task list         rate/hz  max/us  avg/us maxload avgload     total/ms
 0 -       SYSTEM       9      11       0    0.5%    0.5%       910
 1 -          PID     974     132     108   13.3%   11.0%      1957
 2 -         GYRO    3891      29      20   11.7%    8.2%      1298
 3 -           RX       9      57      50    0.5%    0.5%         9
 4 -       SERIAL      99   115922      32 1148.1%    0.8%       233
 5 -      BATTERY      49      27      20    0.6%    0.5%        17
 6 -  TEMPERATURE     100       4       2    0.5%    0.5%         2
 7 -       BEEPER      99      12       6    0.6%    0.5%        12
10 -         BARO      43     241     156    1.5%    1.1%       126
14 -    TELEMETRY     482      22       3    1.5%    0.6%        39
17 -          OSD     243    1937      25   47.5%    1.1%        83
18 -          CMS      49       4       2    0.5%    0.5%         1
21 -      VTXCTRL       4       1       1    0.5%    0.5%         0
22 -  PROGRAMMING       9      25      20    0.5%    0.5%         3
24 -          AUX      99      10       3    0.5%    0.5%         5
25 - SPORT MASTER     482       4       1    0.6%    0.5%         8
Task check function             2       1                         0
Total (excluding SERIAL)                    80.8%   27.0%
```

The issue MIGHT also cause issues in flight. I had some strange uncommanded movements of the plane I can't remember that I had this behavior before. I aborted the flight afterwards as my location was over water.

MrD-RC, 2022-10-30:
Are you sure those PRs are right? They're 5 years old.

b14ckyy, 2022-10-31:
@MrD-RC lol seems I accidently copied the Firmware Build Numbers and not the PR numbers.

shota3527, 2022-10-31:
@DzikuVx 
I did some investigation and found this commit 2e5c90fd5fff08081780f36a70848272452eedd is causing the issue
please have a look

shota3527, 2022-10-31:
i have found all sensor icons is dead in configurator, maybe we need to remove the imu2 from the configurator?

b14ckyy, 2022-10-31:
@shota3527 do you mean these? They work for me in master and FP2 as well 
[image]

DzikuVx, 2022-10-31:
@shota3527 you need latest firmware and configurator. FP2 will not work with current master of firmware

b14ckyy, 2022-10-31:
@DzikuVx I guess you mean for the Icon issue? 

But the Task time issue on F411 is unrelated. It happens also with latest configurator master, built 2 days ago.

shota3527, 2022-11-01:
i have tried the latest master c398c526ecc58cd2c6897bc98ebda740c61392c1 with latest configurator master 1a33ca1b
on MATEKF411SE_SS2_CH6. i did not see packet error anymore.
So i think the packet error issue is caused by the 6.0.0fp1 configurator trying to fetch imu2 data caused the error

b14ckyy, 2022-11-01:
Alright, I rebased my fork now with the latest upstream and can confirm that the issue is gone. Thank you both for looking into it! I'll close this now.

closed 2022-11-01 by b14ckyy