iNavFlight/inav #8518: Mixer Motor Command Limit of 85% at 0% throttle

spatzengr, 2022-11-01:
The Mixer is capping the MAX Motor Command to 85% even when MAX_THROTTLE (etc...) is set to 100%.

You can see in the below two examples when MAX_THROTTLE is set to 2000us (100% throttle --- no CAP) that at full throttle on the transmitter, you get 100% Motor Command.  However, when the throttle is dropped to 0% and you input a sharp stick move, it caps the Motor Commands to 85%.  

This is NOT my log file and the filters\tune needs work, but regardless, you can clearly see it.  I have also run tests myself and get the same result on iNAV 5.1.

This reduces the quad power by 25%!!! to implement sharp stick moves.  Of course, even at moderate max rates of 700dps you can easily require more than 85% of your max motor thrust to follow your rates on sharp stick inputs (5" quad even || much worse for larger quads).

EXAMPLE #1:
[image]

EXAMPLE #2:
[image]

To see flat caps on motor commands (just below 85%) when the PID Sum is still rising is a dead giveaway of a Mixer cap.  Coordinating with @DzikuVx , we don't belive there is a variable to adjust the code to prevent this cap; which is no good.

[LOG00088.TXT](https://github.com/iNavFlight/inav/files/9905677/LOG00088.TXT)

robert-b, 2022-12-03:
no comment from the inav devs.
the mixer is highly qustionable as the 'main' concern is clipping to soften the response.
this has a price.

spatzengr, 2022-12-04:
It is definitely a big from when Throttle Cap was implemented.

There is no logical reason to limit Mixer output (hard coded) to an arbitrary 85% of what the PID Controller commands.  Any logical reason would need to.justify 85% vs 70, 60, 90, etc...

robert-b, 2022-12-04:
yes, i agree.

OptimusTi, 2022-12-04:
Have you guys looked at closed issues? For some reason, this sounds familiar.

spatzengr, 2022-12-04:
> Have you guys looked at closed issues? For some reason, this sounds familiar.

I did look and talked to Pawel prior to posting as an issue.  Understanding working hard to get iNAV 6.0 out the door, I figured it was not the top priority.