iNavFlight/inav #8520: inav 6.0 and matek f405-wse

Daramidon, 2022-11-03:
inav 6.0 firmware is not installed on matek f405-wse there is not enough memory in the flight controller.

Daramidon, 2022-11-03:
[image]

Jetrell, 2022-11-03:
I'll ask a silly question. Are you sure its a F405, you have connected?  They have 1024KB of flash.

Here is my attempt to flash the Matek F405-WSE. It was fine.

[image]

MrD-RC, 2022-11-03:
I think you have an F411-WSE. Not F405-WSE. The F405-WSE has 1024KB of flash. The F411-WSE has 512KB flash. I'm pretty certain you have just chosen the wrong target.

Daramidon, 2022-11-03:
ошибся с прошивкой. вопрос решен.

closed 2022-11-03 by Daramidon

Daramidon, 2022-11-03:
Thanks.  I already figured it out.  I really messed up the firmware.  but
it's all right now.