iNavFlight/inav #8527: SPEEDYBEEF405 blackbox logs just stop after a few seconds

MichaelCWarren, 2022-11-07:
## Current Behavior

After arm, the blackbox log starts, but quits after a few seconds.  Anywhere between 1-10 seconds after arming.

I've tried multiple cards, multiple logging rates, etc.

## Steps to Reproduce

1. Enable blackbox logging
2. Arm
3. Fly
4. Disarm

## Expected behavior

A log will show a complete flight.

## Suggested solution(s)
Good question...

## Additional context
I just purchased this FC about a week ago and have been playing with getting my new quad up and running.  I have NEVER gotten a good log file out of this board.  I've tried multiple sdcards, logging rates, loop times, etc.  In addition to the bad log files (attached), I was having trouble with the FC failing to boot when using a certain sdcard similar to #7759.  I was able to get around that by setting the spi bus speed to "standard", as eluded to in the comments.  

Any help would be greatly appreciated.  Thanks!

[LOG00008.TXT](https://github.com/iNavFlight/inav/files/9947644/LOG00008.TXT)
[LOG00012.TXT](https://github.com/iNavFlight/inav/files/9947645/LOG00012.TXT)
[LOG00001.TXT](https://github.com/iNavFlight/inav/files/9947648/LOG00001.TXT)

Here is a diff of my current setup.
https://pastebin.com/zZT8NkkG

- FC Board name and vendor: SPEEDYBEEF405
- INAV version string: 5.1.0+

MichaelCWarren, 2022-11-07:
I'm more than willing to try any debug versions, etc.

MichaelCWarren, 2022-11-07:
I have some more info.  I just completed a test flight with OSD disabled.  Maybe this is a problem on this board when the OSD is enabled?

gcmcnutt, 2022-12-29:
@MichaelCWarren I have everything enabled, OSD, flight control, etc when using an ancient sdcard - do you know if 'standard' bus speed is 12.5 or 25 mhz?  perhaps try forcing to 12.5

Pedrolomel, 2023-01-27:
SpeedyBee F405 v3 with SD card inserted:
https://github.com/iNavFlight/inav/issues/8733#issue-1557271328

MichaelCWarren, 2023-01-27:
> @MichaelCWarren I have everything enabled, OSD, flight control, etc when using an ancient sdcard - do you know if 'standard' bus speed is 12.5 or 25 mhz?  perhaps try forcing to 12.5

From the comments in the code 'standard' is ~10.  Seems like that would correspond to 12.5.

DzikuVx, 2023-11-17:
fixed in 7.0

closed 2023-11-17 by DzikuVx