iNavFlight/inav #8528: iNav Settings get lost after save VTX settings

wavecut, 2022-11-07:
## Current Behavior
When change VTX setting via OSD menu, all iNav settings get lost.

## Steps to Reproduce

1. Install iNav 6.0.0-TP1
2. Change VTX Settings via OSD
3. Save VTX Settings
4. iNav Settings get lost.

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor: JHEMCU H743 AIO
- INAV version string: INAV/JHEH7AIO 6.0.0 Nov  7 2022 / 13:56:10 (2693037c)

breadoven, 2022-11-07:
Does the same thing happen if you use the Save Setting stick command after booting the FC ?

https://github.com/iNavFlight/inav/blob/master/docs/Controls.md

Can you also post a CLI Diff.

wavecut, 2022-11-08:
> Does the same thing happen if you use the Save Setting stick command after booting the FC ?
> 
> https://github.com/iNavFlight/inav/blob/master/docs/Controls.md
> 
> Can you also post a CLI Diff.

The Save Setting stick command works ok, no setting was lost.

Follow is my Diff All:
``
diff

# version
# INAV/JHEH7AIO 6.0.0 Nov  7 2022 / 13:56:10 (2693037c)
# GCC-10.2.1 20201103 (release)

# start the command batch
batch start

# resources

# mixer

mmix reset

mmix 0  1.000 -1.000  1.000 -1.000
mmix 1  1.000 -1.000 -1.000  1.000
mmix 2  1.000  1.000  1.000  1.000
mmix 3  1.000  1.000 -1.000 -1.000

# servo mixer

# servo

# safehome

# feature
feature GPS
feature PWM_OUTPUT_ENABLE

# beeper

# blackbox
blackbox -NAV_ACC
blackbox NAV_POS
blackbox NAV_PID
blackbox MAG
blackbox ACC
blackbox ATTI
blackbox RC_DATA
blackbox RC_COMMAND
blackbox MOTORS
blackbox -GYRO_RAW
blackbox -PEAKS_R
blackbox -PEAKS_P
blackbox -PEAKS_Y

# map

# serial
serial 0 0 115200 57600 0 115200
serial 2 2 115200 115200 0 115200
serial 3 2048 115200 115200 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1875 2100
aux 1 51 3 1875 2100
aux 2 1 1 1350 1650
aux 3 11 1 1825 2100
aux 4 10 2 1825 2100
aux 5 3 2 1350 2100

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout
osd_layout 0 0 19 0 V
osd_layout 0 1 1 0 V
osd_layout 0 3 8 6 V
osd_layout 0 4 8 6 V
osd_layout 0 8 1 2 V
osd_layout 0 9 1 5 H
osd_layout 0 10 4 15 H
osd_layout 0 11 25 14 V
osd_layout 0 12 24 15 V
osd_layout 0 13 25 3 V
osd_layout 0 14 13 0 V
osd_layout 0 15 24 2 V
osd_layout 0 22 15 4 V
osd_layout 0 23 1 15 V
osd_layout 0 26 25 4 V
osd_layout 0 28 1 3 V
osd_layout 0 32 7 0 V
osd_layout 0 34 11 2 V
osd_layout 0 40 1 14 V
osd_layout 0 41 1 8 V
osd_layout 0 42 24 8 V
osd_layout 0 50 13 15 V
osd_layout 0 100 1 6 V
osd_layout 0 103 1 5 V
osd_layout 0 105 1 15 H
osd_layout 0 108 13 14 V
osd_layout 0 110 23 0 V

# logic

# global vars

# programmable pid controllers

# master
set looptime = 500
set gyro_main_lpf_hz = 100
set gyro_main_lpf_type = PT1
set dynamic_gyro_notch_q = 200
set dynamic_gyro_notch_min_hz = 110
set setpoint_kalman_q = 180
set gyro_zero_x = -73
set gyro_zero_y = 3
set gyro_zero_z = -14
set ins_gravity_cmss =  958.122
set acc_hardware = MPU6000
set acczero_x = 32
set acczero_y = 39
set acczero_z = -133
set accgain_x = 4135
set accgain_y = 4109
set accgain_z = 4018
set align_mag = CW270FLIP
set mag_hardware = HMC5883
set magzero_x = -13
set magzero_y = -37
set magzero_z = 32
set maggain_x = 401
set maggain_y = 371
set maggain_z = 374
set align_mag_pitch = 1800
set align_mag_yaw = 900
set baro_hardware = BMP280
set serialrx_inverted = ON
set blackbox_rate_denom = 5
set motor_pwm_protocol = DSHOT600
set failsafe_procedure = RTH
set align_board_yaw = 450
set current_meter_scale = 170
set model_preview_type = 3
set imu_acc_ignore_slope = 10
set applied_defaults = 2
set gps_ublox_use_galileo = ON
set airmode_type = THROTTLE_THRESHOLD
set nav_extra_arming_safety = ALLOW_BYPASS
set osd_video_system = PAL
set osd_crsf_lq_format = TYPE2
set osd_left_sidebar_scroll = ALTITUDE
set osd_right_sidebar_scroll = SPEED
set osd_switch_indicators_align_left = OFF
set i2c_speed = 800KHZ
set name = LR300
set tz_offset = 480
set vtx_power = 4

# profile
profile 2

set mc_p_pitch = 44
set mc_i_pitch = 75
set mc_d_pitch = 25
set mc_i_roll = 60
set mc_p_yaw = 35
set mc_i_yaw = 80
set dterm_lpf_type = PT3
set d_boost_min =  0.800
set d_boost_max =  1.200
set antigravity_gain =  2.000
set antigravity_accelerator =  5.000
set smith_predictor_delay =  1.500
set tpa_rate = 20
set tpa_breakpoint = 1200
set rc_expo = 75
set rc_yaw_expo = 75
set roll_rate = 70
set pitch_rate = 70
set yaw_rate = 60

# battery_profile
battery_profile 1

set throttle_idle =  6.000

# save configuration
save

# end the command batch
batch end

# 

And after Save VTX Settings, all configuration was reseted:
``
diff all

# version
# INAV/JHEH7AIO 6.0.0 Nov  7 2022 / 13:56:10 (2693037c)
# GCC-10.2.1 20201103 (release)

# start the command batch
batch start

# reset configuration to default settings
defaults noreboot

# resources

# mixer

# servo mixer

# servo

# safehome

# feature

# beeper

# blackbox
blackbox -NAV_ACC
blackbox NAV_POS
blackbox NAV_PID
blackbox MAG
blackbox ACC
blackbox ATTI
blackbox RC_DATA
blackbox RC_COMMAND
blackbox MOTORS
blackbox -GYRO_RAW
blackbox -PEAKS_R
blackbox -PEAKS_P
blackbox -PEAKS_Y

# map

# serial

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout

# logic

# global vars

# programmable pid controllers

# master
set gyro_zero_x = -68
set gyro_zero_y = 3
set gyro_zero_z = -13
set acc_hardware = MPU6000
set align_mag = CW270FLIP
set mag_hardware = HMC5883
set baro_hardware = BMP280
set applied_defaults = 1

# profile
profile 1

# profile
profile 2

# profile
profile 3

# battery_profile
battery_profile 1

# battery_profile
battery_profile 2

# battery_profile
battery_profile 3

# restore original profile selection
profile 1
battery_profile 1

# save configuration
save

#

MrD-RC, 2022-11-08:
Does it happen every time, or was it a one-off?

JulioCesarMatias, 2022-11-08:
we need to have a `flush` function that forces the flash write operations to stop when the reboot is called...

JulioCesarMatias, 2022-11-08:
> we need to have a `flush` function that waits for flash write operations to finish before calling reset... This parameter break occurs because when the microcontroller is restarted, the flash is still dealing with calculations for recording... These calculations are broken because the algorithm does not have a function to wait these calculations before restarting. So the blocks are broken, and the algorithm written in the past makes everything "clean" when the flash verification CRC fails

a queue, some whiles and delay can solve this

wavecut, 2022-11-09:
> Does it happen every time, or was it a one-off?

Yes, it happen every time.

wavecut, 2022-11-09:
> > we need to have a `flush` function that waits for flash write operations to finish before calling reset... This parameter break occurs because when the microcontroller is restarted, the flash is still dealing with calculations for recording... These calculations are broken because the algorithm does not have a function to wait these calculations before restarting. So the blocks are broken, and the algorithm written in the past makes everything "clean" when the flash verification CRC fails
> 
> a queue (ring buffer), some whiles and delay can solve this

It's a good news, thanks in advance.

closed 2023-10-27 by DzikuVx

SeriouSerg, 2024-08-30:
And? 
The issue is still there (7.1.2)