iNavFlight/inav #8538: IMU Temperature invalid in OSD - 6.0-FP1

b14ckyy, 2022-11-09:
## Current Behavior
IMU Temperature in OSD shows erratic values by switching between random negative °C and probably correct temperature forth and back. Issue is not seen on older INAV versions on the same FCs

## Steps to Reproduce
1. Flash 6.0 FPV 
2. enable IMU Temperature in OSD
3. Watch temperature (might need to be in flight though)

## Expected behavior
correct and consistent temp readings from IMU

## Suggested solution(s)
fix

## Additional context
Not tested myself yet but seen the same behavior on a Matek F722-Wing, F411-WSE and H743 Wing. Will try later with my own test setups. Issue reported from multiple users so far.

---

- FC Board name and vendor: Multiple
- INAV version string: INAV 6.0 FP1

JulioCesarMatias, 2022-11-09:
This was also mentioned by @Jetrell

JulioCesarMatias, 2022-11-09:
He said that it occurs in Digital... But, and in analog, it also occurs?

b14ckyy, 2022-11-09:
@JulioCesarMatias yes the last appearance I saw was analog.

Jetrell, 2022-11-09:
@b14ckyy I also experienced it with an F405, F722 and F765, in the same manor as you did... While the H743, wouldn't display the temperature, or out of range value at all.. It was blank (3 lines).  This was the case with MSP OSD. Using Avatar and DJI.

b14ckyy, 2022-11-10:
> @b14ckyy I also experienced it with an F405, F722 and F765, in the same manor as you did... While the H743, wouldn't display the temperature, or out of range value at all.. It was blank (3 lines).  This was the case with MSP OSD. Using Avatar and DJI. 

ah H743 might be WAD. there is at least one ICM gyro that has no temp sensor defined. I thing Sampson mentioned that in one of the posts. just not sure if it was disabled in driver or. if the hardware is not capable.

closed 2022-11-27 by MrD-RC