iNavFlight/inav #8540: INAV altitude hold

bohaman1, 2022-11-10:
Hello everyone,

I apologize if there is already another post but I've been search for my problem for at least over 2 weeks now. However, I have no luck and so I hope if anyone here could help point me to the right direction.

I will try to be as descriptive as possible about my build I really hope someone could help me point out what I'm doing wrong.

My current build. I follow to the T by from a Youtuber Mr. FalconFPV.

Below are his build list and that's what I use to follow

T-Motor F7 Pro Flight controller -

T-Motor F55 ProII ESC - 

T-Motor F90 1300kv Motors - 

Gemfan 7037x3 7" props - 

TureRC Singularity VTX antenna -

Rush Solo VTX -

TBS Crossfire Diversity Receiver -

Matek M8Q-5883 GPS -

TBS Immortal T Receiver antenna

It is a 7inch frame designed by him.

After the build, the quad flies very steady in angle mode using the default setting. However, when I switched on altitude hold, the quad behave very strangly. With the default setting the quad just drop. So I thought it need to increase the I. However, as I increase the I, the quad be have even stranger that it continue to vertically oscillating. So I tried to lowering the max_auto_climb  because in the github it say that may be the reason why its oscillating but it still doesn't work. I did have the nav_use_midthr_for_althold=OFF and my hover throttle adjusted correctly but I seems can't to get the quad working in altitude hold.

I went on youtube and make sure to watch the youtuber pawel spychalski for instruction to read the blackbox and his advice to fix the other altitude problem but all of the video are hardware related but not tuning process. I made sure to follow his troubleshoot guide but my gyro reading, baro is very clean, no vibration.

So I went back and go from the beginging. I followed this step by step from this https://github.com/iNavFlight/inav/wiki/Navigation-modes#althold---altitude-hold

These are the step I take from this guide:

First I find out my quad's hover throttle which is around 1300

Then I set nav_use_midthr_for_althold =OFF

set nav_mc_hover_thr = 1300 (my hover throttle)

then I set inav_w_z_baro_p = 0.5

set ALT P(nav_mc_pos_z_p) = 0

Try to fly, my quad oscilating but with an increase in altitude

[img]https://i.imgur.com/8Qp0YBal.jpg[/img]

according to the image above. I set my D in for my altitude auto at 35. Which the guide say at the edge of oscilating.

Then I started to add on my P which is nav_mc_vel_z_p

I changed to 5 and 2 increment at first my quad was oscilating but then as I increased the P it actually stayed stable then I go pass that as the guide said to almost oscilating.

[img]https://i.imgur.com/yeyxsfRl.jpg[/img]

My quad actually stayed stable then I go passed it a little bit to make it oscilating a little bit again.

But when it come to adjusting the I nav_mc_vel_z_i No matter how or what I do my quad just lose altitude

I incrased my nav_mc_vel_z_i by 5 increment the following blackbox image is from 5 up to 250

[img]https://i.imgur.com/60UUNpil.jpg[/img] this is at 5

[img]https://i.imgur.com/hUWtOoPl.jpg[/img] at 2

[img]https://i.imgur.com/aUAtPXcl.jpg[/img] at 10

[img]https://i.imgur.com/77i7hpAl.jpg[/img] at 70

No matter how much I tried my quad just drop in altitude.

I've read the guide and it said that the reason the quad dropped altitude because not enough I but I increased to the max 250 but it still dropped in altitude.

So I thought maybe I need to increase my hover throttle. So I did that and my quad react very violently, jumping up and down, so I have to lower it to make it controllable again.

I honestly tried everything that I know how and not just sit and ask people for help. I've have been dealing with this for more than 2 years now and this is the third quad I build. I thought using better compoenent will help but I still have no luck. I checked on the gyro, baro data and they all give very clean signal. No noise nor vibration.

Please if anyone could help point me to the right material or direction. I've tried to search everywhere, youtube, google, but none give steps to fix this. They either fix hardware issues or just PID tuning quad in general. My quad fly great but when it come to navigation mode. altitude hold/GPS I just can't tune it right.

Thank you for reading my long post and I do really hope someone can help me out.
I hope this is not too much but I'm attaching logs.
[blackbox_log_2022-11-09_110757.TXT](https://github.com/iNavFlight/inav/files/9978295/blackbox_log_2022-11-09_110757.TXT)
[blackbox_log_2022-11-09_114116.TXT](https://github.com/iNavFlight/inav/files/9978296/blackbox_log_2022-11-09_114116.TXT)
[blackbox_log_2022-11-09_115420.TXT](https://github.com/iNavFlight/inav/files/9978298/blackbox_log_2022-11-09_115420.TXT)
[blackbox_log_2022-11-09_122445.TXT](https://github.com/iNavFlight/inav/files/9978299/blackbox_log_2022-11-09_122445.TXT)
[blackbox_log_2022-11-09_124111.TXT](https://github.com/iNavFlight/inav/files/9978300/blackbox_log_2022-11-09_124111.TXT)
[blackbox_log_2022-11-09_124906.TXT](https://github.com/iNavFlight/inav/files/9978302/blackbox_log_2022-11-09_124906.TXT)
[blackbox_log_2022-11-09_125641.TXT](https://github.com/iNavFlight/inav/files/9978303/blackbox_log_2022-11-09_125641.TXT)
[blackbox_log_2022-11-09_131153.TXT](https://github.com/iNavFlight/inav/files/9978304/blackbox_log_2022-11-09_131153.TXT)
[blackbox_log_2022-11-09_132800.TXT](https://github.com/iNavFlight/inav/files/9978307/blackbox_log_2022-11-09_132800.TXT)
[blackbox_log_2022-11-09_133900.TXT](https://github.com/iNavFlight/inav/files/9978308/blackbox_log_2022-11-09_133900.TXT)
[blackbox_log_2022-11-09_134326.TXT](https://github.com/iNavFlight/inav/files/9978310/blackbox_log_2022-11-09_134326.TXT)
[blackbox_log_2022-11-09_134916.TXT](https://github.com/iNavFlight/inav/files/9978311/blackbox_log_2022-11-09_134916.TXT)
[blackbox_log_2022-11-09_135347.TXT](https://github.com/iNavFlight/inav/files/9978313/blackbox_log_2022-11-09_135347.TXT)
[blackbox_log_2022-11-09_135741.TXT](https://github.com/iNavFlight/inav/files/9978316/blackbox_log_2022-11-09_135741.TXT)
[blackbox_log_2022-11-09_140125.TXT](https://github.com/iNavFlight/inav/files/9978317/blackbox_log_2022-11-09_140125.TXT)
[blackbox_log_2022-11-09_140434.TXT](https://github.com/iNavFlight/inav/files/9978318/blackbox_log_2022-11-09_140434.TXT)
[blackbox_log_2022-11-09_140814.TXT](https://github.com/iNavFlight/inav/files/9978319/blackbox_log_2022-11-09_140814.TXT)
[blackbox_log_2022-11-09_141132.TXT](https://github.com/iNavFlight/inav/files/9978321/blackbox_log_2022-11-09_141132.TXT)
[blackbox_log_2022-11-09_141503.TXT](https://github.com/iNavFlight/inav/files/9978323/blackbox_log_2022-11-09_141503.TXT)
[blackbox_log_2022-11-09_141938.TXT](https://github.com/iNavFlight/inav/files/9978324/blackbox_log_2022-11-09_141938.TXT)
[blackbox_log_2022-11-09_142310.TXT](https://github.com/iNavFlight/inav/files/9978327/blackbox_log_2022-11-09_142310.TXT)
[blackbox_log_2022-11-09_142611.TXT](https://github.com/iNavFlight/inav/files/9978328/blackbox_log_2022-11-09_142611.TXT)
[blackbox_log_2022-11-09_143019.TXT](https://github.com/iNavFlight/inav/files/9978329/blackbox_log_2022-11-09_143019.TXT)
[blackbox_log_2022-11-09_143519.TXT](https://github.com/iNavFlight/inav/files/9978330/blackbox_log_2022-11-09_143519.TXT)
[blackbox_log_2022-11-09_143846.TXT](https://github.com/iNavFlight/inav/files/9978331/blackbox_log_2022-11-09_143846.TXT)
[blackbox_log_2022-11-09_144223.TXT](https://github.com/iNavFlight/inav/files/9978332/blackbox_log_2022-11-09_144223.TXT)
[blackbox_log_2022-11-09_144702.TXT](https://github.com/iNavFlight/inav/files/9978335/blackbox_log_2022-11-09_144702.TXT)
[blackbox_log_2022-11-09_145428.TXT](https://github.com/iNavFlight/inav/files/9978338/blackbox_log_2022-11-09_145428.TXT)
[blackbox_log_2022-11-09_145814.TXT](https://github.com/iNavFlight/inav/files/9978340/blackbox_log_2022-11-09_145814.TXT)
[blackbox_log_2022-11-09_165325.TXT](https://github.com/iNavFlight/inav/files/9978341/blackbox_log_2022-11-09_165325.TXT)
[blackbox_log_2022-11-09_170618.TXT](https://github.com/iNavFlight/inav/files/9978343/blackbox_log_2022-11-09_170618.TXT)
[blackbox_log_2022-11-09_171850.TXT](https://github.com/iNavFlight/inav/files/9978344/blackbox_log_2022-11-09_171850.TXT)
[blackbox_log_2022-11-09_173202.TXT](https://github.com/iNavFlight/inav/files/9978347/blackbox_log_2022-11-09_173202.TXT)
[blackbox_log_2022-11-09_174213.TXT](https://github.com/iNavFlight/inav/files/9978348/blackbox_log_2022-11-09_174213.TXT)

Jetrell, 2022-11-11:
I see from your write up, that you have a grasp of what many of the navigation PID's do.
And as you have seen. Some don't change their specific field, in a linear fashion.

Over my years of MC setup using INAV.  I have found there is a delicate balance between _Setting changes_ and _Hardware related issues_. 
One quad can be setup very easily, and work almost first go. And another can cost you a fortune in time and new parts, to get it working better, and still not fly as perfect as the other.

When adjusting the balance between GPS altitude and Baro altitude settings. Be cautious. A very small change can have a huge effect. And you may be left chasing your tail.

Boiling it down.. Most altitude issues that occur with INAV use. Are related to the IMU experiencing random frame resonance noise. Or abnormal motor vibrations, on one or a few of the motors.
Things of this nature are hard to track down. And just as hard to fix. Due to the gyro traces often looking very clean, with respect to the raw gyro data.. 
From my past observations. I would swear that the dynamic notch filters are missing certain frame resonance spikes, that just don't show up in the gyro log traces... Even with the logs recording at 1 : 1.

z-renat, 2023-01-14:
Hello.
When switching from ANGLE to ALT HOLD, height retention is possible:

1. At the current position of the gas stick if nav_use_midthr_for_althold=0. At the same time, if the current position is low (high thrust-to-weight ratio of the copter), then the copter will unevenly change height when the position of the gas stick changes. In my case, the position of the hovering gas stick is 20%. When gas is added, the copter slowly and smoothly gains altitude. When the gas decreases, the copter decreases sharply. It is difficult to find gas positions so that the copter descends smoothly. It seems that the decline is faster than the rise, faster than set in the settings.

2. At the central position of the gas stick if nav_use_midthr_for_althold=1. In this case, the copter smoothly changes height up and down. But inevitably there are jerks when switching from ANGLE to ALT HOLD and back from ALT HOLD to ANGLE.

Is it possible to make nav_use_midthr_for_althold=1 use not the central position of the gas stick, but the position specified in thr_mid? And make a smoother descent?
Machine translation, I'm sorry.

closed 2023-10-27 by DzikuVx