iNavFlight/inav #8545: INav 6 horizon drift issue

mykls, 2022-11-12:
I was testing INAV 6 on an AR Wing Pro and late in the flight (30 mins in) I had a horizon drift issue on stabilised mode (I usually fly on Acro), which caused an uncommanded left roll and later resulted in a crash when I accidently tripped RTH too close to the ground.  I also test flew the INAV 6 build on some other planes and they performed fine.  Blackbox log available and attached (zip).

DIFF ALL config below:

# INAV/MATEKF405SE 6.0.0 Oct 20 2022 / 15:20:51 (d1f6f19a)
# GCC-10.2.1 20201103 (release)

# start the command batch
batch start

# reset configuration to default settings
defaults noreboot

# resources

# mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000

# servo mixer
smix reset

smix 0 3 0 50 0 -1
smix 1 3 1 50 0 -1
smix 2 4 0 -50 0 -1
smix 3 4 1 50 0 -1

# servo
servo 3 1000 2000 1570 -90
servo 4 1000 2000 1533 115

# safehome

# feature
feature MOTOR_STOP
feature GPS
feature BLACKBOX
feature PWM_OUTPUT_ENABLE
feature FW_LAUNCH
feature FW_AUTOTRIM

# beeper

# blackbox
blackbox -NAV_ACC
blackbox NAV_POS
blackbox NAV_PID
blackbox MAG
blackbox ACC
blackbox ATTI
blackbox RC_DATA
blackbox RC_COMMAND
blackbox MOTORS
blackbox -GYRO_RAW
blackbox -PEAKS_R
blackbox -PEAKS_P
blackbox -PEAKS_Y

# map

# serial
serial 0 64 115200 115200 0 115200
serial 1 0 115200 115200 0 115200
serial 2 2 115200 115200 0 115200
serial 5 2097152 115200 115200 0 115200
serial 30 0 115200 115200 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1200 2100
aux 1 1 2 1700 2100
aux 2 12 1 1700 2100
aux 3 53 1 1300 1700
aux 4 11 3 1700 2100
aux 5 10 3 1300 1700
aux 6 57 4 1300 2100
aux 7 13 5 1600 2100
aux 8 26 0 1200 2100
aux 9 27 2 900 1200

# adjrange

# rxrange

# temp_sensor

# wp
#wp 0 invalid

# osd_layout
osd_layout 0 0 1 14 V
osd_layout 0 1 24 12 H
osd_layout 0 7 12 15 V
osd_layout 0 8 13 14 V
osd_layout 0 11 1 11 V
osd_layout 0 12 0 12 V
osd_layout 0 13 0 7 V
osd_layout 0 14 1 1 V
osd_layout 0 15 0 8 V
osd_layout 0 20 17 0 V
osd_layout 0 21 4 0 V
osd_layout 0 22 15 0 V
osd_layout 0 23 1 9 V

# logic

# global vars

# programmable pid controllers

# master
set gyro_main_lpf_hz = 25
set dynamic_gyro_notch_q = 250
set dynamic_gyro_notch_min_hz = 30
set gyro_zero_x = -9
set gyro_zero_y = -22
set gyro_zero_z = -16
set ins_gravity_cmss =  977.286
set acc_hardware = MPU6000
set acc_lpf_hz = 102
set acczero_x = 72
set acczero_y = -20
set acczero_z = -276
set accgain_x = 4080
set accgain_y = 4094
set accgain_z = 4044
set align_mag = CW270FLIP
set mag_hardware = NONE
set baro_hardware = DPS310
set rssi_channel = 15
set serialrx_provider = CRSF
set blackbox_rate_denom = 2
set motor_pwm_protocol = STANDARD
set failsafe_delay = 10
set failsafe_procedure = RTH
set current_meter_scale = 256
set model_preview_type = 8
set small_angle = 180
set imu_acc_ignore_slope = 10
set imu_inertia_comp_method = ADAPTIVE
set applied_defaults = 3
set gps_provider = UBLOX7
set deadband = 4
set yaw_deadband = 25
set airmode_type = STICK_CENTER_ONCE
set inav_w_z_baro_p =  0.350
set nav_wp_radius = 1500
set nav_manual_climb_rate = 500
set nav_rth_climb_first_stage_mode = EXTRA
set nav_rth_climb_first_stage_altitude = 4000
set nav_rth_climb_ignore_emerg = ON
set nav_rth_allow_landing = FS_ONLY
set nav_rth_alt_control_override = ON
set nav_rth_abort_threshold = 60000
set nav_rth_altitude = 7000
set safehome_usage_mode = RTH_FS
set nav_fw_bank_angle = 45
set nav_fw_climb_angle = 25
set nav_fw_dive_angle = 20
set nav_fw_loiter_radius = 6000
set nav_fw_control_smoothness = 4
set nav_fw_launch_velocity = 150
set nav_fw_launch_max_angle = 180
set nav_fw_launch_timeout = 10000
set nav_fw_launch_max_altitude = 10000
set nav_fw_launch_climb_angle = 19
set i2c_speed = 800KHZ
set name = :WTED

# profile
profile 1

set fw_i_pitch = 12
set fw_ff_pitch = 80
set fw_i_roll = 10
set fw_ff_roll = 40
set fw_p_yaw = 0
set fw_i_yaw = 0
set fw_ff_yaw = 0
set max_angle_inclination_rll = 450
set dterm_lpf_hz = 10
set fw_turn_assist_pitch_gain =  0.200
set heading_hold_rate_limit = 101
set nav_fw_pos_z_p = 35
set nav_fw_pos_xy_p = 70
set fw_level_pitch_trim =  5.000
set rc_expo = 30
set rc_yaw_expo = 30
set roll_rate = 34
set pitch_rate = 10
set yaw_rate = 1

# profile
profile 2

set fw_p_pitch = 2
set fw_i_pitch = 12
set fw_ff_pitch = 80
set fw_p_roll = 1
set fw_i_roll = 10
set fw_ff_roll = 30
set fw_p_yaw = 0
set fw_i_yaw = 0
set fw_ff_yaw = 0
set max_angle_inclination_rll = 450
set fw_level_pitch_trim =  5.000
set roll_rate = 36
set pitch_rate = 12
set yaw_rate = 1

# profile
profile 3

set max_angle_inclination_rll = 450
set fw_level_pitch_trim =  5.000

# battery_profile
battery_profile 1

set bat_cells = 4
set vbat_min_cell_voltage = 275
set vbat_warning_cell_voltage = 325
set battery_capacity = 7000
set battery_capacity_warning = 1400
set battery_capacity_critical = 700
set throttle_idle =  5.000
set nav_fw_cruise_thr = 1550
set nav_fw_max_thr = 1800
set nav_fw_launch_thr = 1800

# battery_profile
battery_profile 2

# battery_profile
battery_profile 3

# restore original profile selection
profile 1
battery_profile 1

---
- FC Board name and vendor: MATEKF405SE
- INAV version string: INAV/MATEKF405SE 6.0.0 Oct 20 2022 / 15:20:51 (d1f6f19a)
[LOG00017.zip](https://github.com/iNavFlight/inav/files/9993963/LOG00017.zip)

breadoven, 2022-11-12:
Looks like the Baro altitude was decreasing during the final RTH Climb phase in spite of the pitch Attitude indicating nose up. Also servos 3 & 4 appear to be pitching down. This is based on their positions during the launch climb where 3 is low and 4 high. At the end of the flight during the RTH climb both servos have higher values with 3 higher than 4 indicating pitch down. So a large amount of horizon drift on the pitch axis. Is there no OSD DVR available to confirm this ?

Also why did you use Failsafe mode to trigger RTH then RTH mode itself afterwards ?

b14ckyy, 2022-11-12:
The log also shows a MASSIVE amount of vibration induced in your Flight controller. In the 15-20Hz range and also in the 45-55Hz range the noise is critical. 
Accellerometer Z axis fluctuates between -5G and +5G at high frequency. 

this could have completely thrown off the IMU. 
The Vibration amplitude raises with throttle but frequency lowers at higher throttle. I assume this is a Twin motor plane? 
In this case these are harmonic frequencies from 2 unbalanced props.

mykls, 2022-11-12:
> Looks like the Baro altitude was decreasing during the final RTH Climb phase in spite of the pitch Attitude indicating nose up. Also servos 3 & 4 appear to be pitching down. This is based on their [...]

No OSD as I was flying with the DJI OSD.  I'll extract some of the video for the last minute of the flight and upload if I can reduce the file size enough.  The triggering of FS was accidental; I have a three position switch that has FS (used for two switch arm), Acro and Angle and when I switched to Angle it pitched down and rolled, so in my haste to push it back to Acro, I pushed it too far and it went in to FS.  As the ground was approaching rapidly I was fumbling with the switches (RTH next switch over) to regain control, but got the wrong switch.

https://user-images.githubusercontent.com/58650613/201492789-82cd065e-7c1c-47b4-8709-4fef8333825e.mp4

mykls, 2022-11-12:
> The log also shows a MASSIVE amount of vibration induced in your Flight controller. In the 15-20Hz range and also in the 45-55Hz range the noise is critical. Accellerometer Z axis fluctuates between -5G and +5G at high frequency.
> 
> this could have completely thrown off the IMU. The Vibration amplitude raises with throttle but frequency lowers at higher throttle. That's a bit weird. But there is some issue with physical vibrations for sure.
> 

The video I posted in my last reply shows how bumpy the last part of the flight was.  Much more than I would expect given the conditions and at the start of the flight it was very smooth, so it's likely that the vibration started during the flight.  My low-ESR capacitor on the flight controller separated in the crash, so it may have come loose in flight, which may have made a difference.  

This was my second flight on INAV 6 for this aircraft.  Earlier flight seemed fine and no issues on the previous INAV 3 flights.

mykls, 2022-11-12:
Just checked previous logs and while previous flights showed a small peak of accelerometer noise around 170-180Hz, it was very small compared to this flight.  Not sure what changed on this flight.

b14ckyy, 2022-11-12:
@mykls one explanation could be that during the previous landing, the prop got a tip chopped off during landing and you did not notice? The vibrations are too strong and too regular to be a lose part inside the plane. But they are the most likely explanation. The gyro data look fine. so its only mechanical, not electrical.

breadoven, 2022-11-13:
Hard to tell from the video how bad the pitch "nodding" was but that has always been the main cause of horizon drift in my case where flying wings are concerned. Especially affects the pitch horizon. Did it seem worse on this particular flight compared to normal ?

b14ckyy, 2022-11-13:
You won't see the vibrations I mentioned, in the FPV fed as movement. These are vibrations, not actual plane movements. The nodding in the video is totally normal and has no effect on the AHRS at all. I flew the AR pro in very bad conditions here:  
https://youtu.be/edKjR93SS7c

the only maneuver we know of that still can induce horizon drift, is a repeated fast 360° roll with short stops in between. You can see that in the video. But it recovers quickly as soon as I level out and even RTH/Loiter is still stable.

mykls, 2022-11-13:
> Hard to tell from the video how bad the pitch "nodding" was but that has always been the main cause of horizon drift in my case where flying wings are concerned. Especially affects the pitch horizon. Did it seem worse on this particular flight compared to normal ?

Pitch nodding is not that much of an issue with my AR Wing; it's more roll oscillations that I experience in bumpy conditions.  Pitch nodding is an issue with my Heewing F-01 though, although it doesn't seem to have any issues with INAV 6.

Flight seemed normal up until the last minute.  As b14ckyy notes, I didn't notice any vibrations in the FPV feed despite the log showing that they were there for the whole flight.  I've re-built the plane now and I'll be more focused on checking for vibrations before launch now and checking stabilised mode at a safe altitude before I fly anywhere that may lead to a failsafe RTH.

closed 2022-11-13 by mykls