iNavFlight/inav #8550: Holybro Kakute H7 mini (V1.3): Feedback

BrownC0de, 2022-11-12:
Hi. I want to describe my experience with Holybro Kakute H7 mini board, board version 1.3
Changes in this version are:
EMU: BMI270
Onboard Flash - 1gb
https://ueeshop.ly200-cdn.com/u_file/UPAC/UPAC331/2208/photo/8383244e08.jpg

I used the firmware INAV/KAKUTEH7 6.0.0 Nov  7 2022 / 06:57:23 (fae79547)  GCC-10.2.1 20201103 (release) recommended here:
https://github.com/iNavFlight/inav/issues/8504

The problems I encountered: 
1. The blackbox does not work, INAV configurator does not see the flash (No dataflash
chip found)
Maybe I'm doing something wrong?

2. Default accel needs to rotate yaw 270 degrees
[image]

3. And I can not calibrate the accelerometer. When you click on the calibration - reboot.

JulioCesarMatias, 2022-11-13:
Answers:

1 - The flash chip used by the manufacturer is a W25N01GVZEIG, and iNav doesn't have the drivers to make it work.

2 - Instead of using the `KAKUTEH7` target (which you are using), use the `KAKUTEH7MINI` target.

3 - The calibration problem is associated with answer 2.

sashgorokhov, 2022-11-13:
Hey! I recently bought Kakute H7 mini v1.3 too, and can confirm that inav 6.0 target KAKUTEH7MINI indeed does not work with gyro. But inav 5.1 is just fine!

JulioCesarMatias, 2022-11-13:
> Hey! I recently bought Kakute H7 mini v1.3 too, and can confirm that inav 6.0 target KAKUTEH7MINI indeed does not work with gyro. But inav 5.1 is just fine!

What do you mean by `not work` ? Is the gyro recognized in the system?

BrownC0de, 2022-11-13:
I understand. I was misled that the link you linked didn't have HEX for mini.
https://github.com/iNavFlight/inav/issues/8550#issuecomment-1312776246
> Instead of using the `KAKUTEH7` firmware (which you are using), use the `KAKUTEH7MINI` firmware .

JulioCesarMatias, 2022-11-13:
> I understand. I was misled that the link you linked didn't have HEX for mini. http://seyrsnys.myzen.co.uk/inav_dev/
> 
> > Instead of using the `KAKUTEH7` firmware (which you are using), use the `KAKUTEH7MINI` firmware .

download the configurator here >> http://seyrsnys.myzen.co.uk/inav-configurator-next/

Go to the `Firmware Flasher` tab and use the box to search for the `KAKUTEH7MINI` firmware
[image]

alex-hsp, 2022-11-14:
> What do you mean by not work ? Is the gyro recognized in the system?
It is not. After i flashed, inav showed BAD GYRO message in OSD and configurator was failing on hardware health. in confguration tab, accelerometer was set to None. 
After setting it to correct IMU type for kakuteh7mini, issue still persisted. So flashed back to inav 5.1 and everything is fine.

DzikuVx, 2022-11-16:
This board is supported only by the latest INAV 5.1, not INAV 6.
Please flash 5.1 and all the functions should work

closed 2022-11-16 by DzikuVx