iNavFlight/inav #8551: RPM Telemetry output to transmitter (2 wire)

Outwide, 2022-11-13:
## Current Behavior
Not able to get RPM telemetry (2 wire) eg Hobbywing RPM sensor to the transmitter and OSD from ESC with no telemetry.

## Desired Behavior
To be able to use eg, Hobbywing RPM sensor 2 wire system, to get RPM telemetry from brushless motors when your ESC has no telemetry support. Telemetry should be to both transmitter and OSD. This assist in the safe development of aircraft.

## Suggested Solution
Add RMP telemetry support for the 2-wire system to Inav CLI or GUI by add or adjust information fields for correct setup and output.

## Who does this impact? Who is this for?
This impacts those using ESC's telemetry with no RPM telemetry and doing diagnostics for best performance. This would improve the programs versatility for aircraft development with ESC with no telemetry. This would be used for those new builds at first where safety is always important, and you don't need to use googles at first and then at distance as needed.    

## Additional context
This would be an excellent addition to telemetry to understand RPM and current draw and the best practices and safety of your motor and battery life, lots of good information for documentation with new laws and drone licensing rules.

Outwide, 2022-11-13:
Suggested Solution
If the 2-wire system is not possible then I2C may be better option with a similar setup as the Temperature sensor for configuration setup.

MrD-RC, 2022-11-13:
Getting data back to the transmitter is control/telemetry link dependent. An RPM sensor could be added to FrSky SmartPort, as these sensors exist and are used. But for other links, they would need to support an RPM sensor. For example, CRSF does not have RPM in its list of sensors. 

What control link are you using?

Outwide, 2022-11-13:
I am using I-Bus on one new build, and I am just getting ELRS for to other build. I need this to come through the FC with all of the out good sensors Inav pulls through.

closed 2023-10-27 by DzikuVx