iNavFlight/inav #8564: Mavlink receiver not arming

esarafis, 2022-11-19:
## Current Behavior
I am using only mavlink as receiver and I am not able to arm. No RC receiver pre-arm check.

## Desired Behavior
To bypass RC receiver not found error so I can arm sending a mavlink command

## Suggested Solution
arm sending a mavlink commandsuggested solutions -->

## Who does this impact? Who is this for?
Those who are using solely mavlink receivers

## Additional context

RomanLut, 2022-11-21:
What version of INAV are you using?

Mavlink only receiver works just fine for me with 5.1.
Make sure:
1) "Telemetry Output" is enabled in "Configuration tab
2) Mavlink telemetry with correct speed is enabled in "Ports" tab
3) Receiver type is set to "Serial", Provider "Mavlink".
You should see channel values changing in Receiver tab if everything is configured correctly.

esarafis, 2022-11-21:
Hi,

I see channel values changing in Receiver tab normally. I have setup a channel for arming. But When I send this channel value to arm, it cannot arm and there is a message "No RC receiver". I do not connect a RC receiver, of course. I suppose this is a pre-arm check. So what mavlink command gto send to arm without restrictions? Thanks

RomanLut, 2022-11-21:
There is no special mavlink command for arming.
1) What version of INAV are you using?
2) Please provide diff from CLI

closed 2024-03-20 by b14ckyy