iNavFlight/inav #8568: Mamba F722 APP MK4 RC Link disconnected once the VTX switch is on

hrnndo, 2022-11-21:
FC Mamba F722 APP MK4
RC Elrs
VTX DJI AU

## Current Behavior
When activating vtx switch using User1 mode the RC link loss 

## Steps to Reproduce

1. Connect the radio to FC
2. Switch on VTX on User1
3. RC Link disconnected

## Additional context

CLI dump

https://pastebin.com/FEHUrJPu
---

- FC Board name and vendor: MAMBAF722_2022B 5.1.0 Aug 26 2022 / 10:02:27 (b63e68a6)
- INAV version string: GCC-10.2.1 20201103 (release)

Jetrell, 2022-11-21:
I don't own one of these boards. But I do own a Mamba F405_2022A. (Mamba F405 mini MK4)

In my experience with the Mamba F405 MK4. I couldn't get the USER1 switcher to operate at all with the default PinIO box settings. But I never lost the CRSF link either.
After much testing. I found these worked. Maybe they might help in your case too.

```
set pinio_box1 = 40
set pinio_box2 = 47
```

hrnndo, 2022-11-21:
> I don't own one of these boards. But I do own a Mamba F405_2022A. (Mamba F405 mini MK4)
>
> In my experience with the Mamba F405 MK4. I couldn't get the USER1 switcher to operate at all with the [...]

@Jetrell  thanks!

I have tried those settings, but it makes the USER1 switch not work

Jetrell, 2022-11-21:
So you loaded them and it stopped it working.. No problem... It must just be related to the F405 MK4 USER1 VTX switcher.

Jetrell, 2022-11-21:
Without hardware, its hard for me to test things. But here are the target definitions for BF and INAV if you want to compare them.
Interesting that there was a Target fix in betaflight yesterday, for the F722 MK4 2022.

`INAV` _Has both A and B_  https://github.com/iNavFlight/inav/blob/master/src/main/target/MAMBAF722_2022A/target.h

`BetaFlight` https://github.com/betaflight/unified-targets/blob/master/configs/default/DIAT-MAMBAF722_2022B.config

DzikuVx, 2022-11-21:
How do you power the RC link? From which pad?

hrnndo, 2022-11-21:
RC link powered from 5V pad in line with RX1 and TX1

hrnndo, 2022-11-22:
Finally, I'm able to resolve this by moving the receiver from UART1 to UART2

closed 2022-11-22 by hrnndo